pvvx/ATC_MiThermometer #565: CGG1 - screen not refreshes at all - keeps the same values all the time

przemhb, 2024-11-05:
I have flashed v4.7 firmware to the CGG1 sensor. At first its screen was showing current measurements, but soon it stopped. Just froze at some values and kept them for days. When I reverted to the stock 2.2.1 it refreshed its screen again. So I have tried one more time, flash 4.7 (BTHome mode). Leave defaults. The same happened - it stopped updating its screen.

Interestingly, throughout the time, it sent updated readings over BLE to Home Assistant. That is why I know it works and measures, just stops refreshing its screen.

pvvx, 2024-11-05:
https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html
[image]

przemhb, 2024-11-06:
The very first thing I did before posting the issue was to check if the Screen Off option is not ticked. 
It is unticked when I connect. Stays the same when I click Get Config.
Log states:

```
21:39:06: Searching for devices
21:39:13: Get Advertising MAC (Web Experimental Features!)
21:39:13: TypeError: device.watchAdvertisements is not a function
21:39:13: Reconnect 1 from 5
21:39:15: Hardware Revision String: 2022
21:39:15: Software Revision String: V4.7
21:39:15: Firmware Revision String: github.com/pvvx
21:39:15: Detected custom Firmware
21:39:15: Hardware Version: CGG1-M 2022, Software Version: 4.7
21:39:15: Custom config HEX string: 55271500002804a9313187b4
```

pvvx, 2024-11-07:
With the CGG1-M thermometers I had, there were no such problems.

One of the thermometers has the following configuration:
```
04:36:56: Searching for devices
04:39:01: Connecting to: CGG_145087
04:39:05: Hardware Revision String: 2022
04:39:05: Software Revision String: V4.8
04:39:05: Firmware Revision String: github.com/pvvx
04:39:05: Detected custom Firmware
04:39:05: Hardware Version: CGG1-M 2022, Software Version: 4.8
04:39:05: Custom config HEX string: 55071000005004a93131075a
04:39:35: Disconnected.
```

przemhb, 2024-11-07:
Can I send configuration as a HEX string?

I've bought 2 CGG1-M's from the same batch. Both have their own peculiarities. (The second one does not send data for a few hours each night, when the temperature is changing.)

I could have tested the "Screen Off" option with the one which doesn't update screen. But I did this immediately after flashing pvxx firmware v4.7. A minute later I turned it off. Maybe this turn off was not effective? Could it be the case?

As the screen was still not refreshing I flashed v4.8, turned on the Screen off then turned it off again, but it didn't help. Next I flashed stock firmware, which in the meantime got updated to 2.2.1. At first, for a few hours, I couldn't flash pvxx again - it complained, that the firmware is unknown. When I eventually did this I flashed v4.7 again. But the issue persists.

How to move forward? Do you want me to provide some debug information?

pvvx, 2024-11-07:
> Can I send configuration as a HEX string?

Yes.
 
> How to move forward? Do you want me to provide some debug information?

At the moment I don't know what to assume... If any options arise, I will write.

a-skuba, 2025-08-07:
If anyone is looking for solution: switch from `CGG1M` firmware to `CGG1`.