pvvx/ATC_MiThermometer #566: LYWSD03MMC, FW 4.7, BTHome v2, HA: What exactly does the "Power" sensor report?

Moonbase59, 2024-11-06:
Thanks for your and atc144's fantastic work on all these temp/humidity sensors!

I have one question I couldn't exactly find out. In Home Assistant, using the BTHome integration, a "Power" sensor shows up and goes on and off for my LYWSD03MMC devices (FW 4.7, set to BTHome v2 and all defaults):

[image]

What exactly does this sensor report?

It seems to go "on" more often (or permanently) when I place the thermometer in "bad conditions", like into the fridge, or outside. Did I maybe find an "easter egg" and it actually reports being within the set "comfort zone"? Comparing temp/humidity at the times it switches on and off somehow seem to hint at that.

pvvx, 2024-11-06:
[README: Temperature or humidity trigger (GPIO PA5 LYWSD03MMC label on the "reset" pin)](https://github.com/pvvx/ATC_MiThermometer?tab=readme-ov-file#temperature-or-humidity-trigger-gpio-pa5-lywsd03mmc-label-on-the-reset-pin)

https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html
[image]

Moonbase59, 2024-11-06:
Ah, thanks for the pointer, missed that one. ;-) Let's see if I can use it for something…

Do you think we could advertise something like "within/without comfort zone" easily, as a binary sensor? Like reporting the same as the smiley face? Maybe in a future version?

This could make for a higher "partner acceptance factor"—we could set a "comfort zone" (shown by the smiley + the sensor) and report that back to HA. Thus, the girlfriend/wife/partner wouldn't have to check HA but also see the same info on the smiley, and us "techies" had the same info in HA, easily. That should allow for more possibilities than just the value+hysteresis.

It would also allow for easier automation in case of alerts, like for bedroom, plant conditions, fridge, etc. (No need to program templates for each.)

pvvx, 2024-11-06:
Transferring additional data requires more energy from the battery. 

It is more convenient to describe "comfortable" conditions in HA automation.

Moonbase59, 2024-11-06:
> The "comfort" parameters are presented […]

In the BLE advertisements? Or just in the Telink Flasher? Maybe I miss another thing.

But anyway, thanks for the response, that is certainly a valid standpoint—battery life _is_ important. It was just an idea for possible enhancement.

almirus, 2024-11-07:
@pvvx offtopic:
Is this kind of battery capacity sensor behavior normal?

[image]

The battery just won't die. 😁