pvvx/ATC_MiThermometer #569: Support for XIAOMI Mijia Bluetooth Thermometer 3? don't start after battery change

dartymath, 2024-11-20:
Dear all,
i successfully flash my XIAOMI Mijia Bluetooth Thermometer 3? but after several month my device show low battery, so i remove old one and change by new one and the screen start normally less than 1 second then nothing and it. I try with another device no issue with the new battery. i have only one ! so may be lost of the firmware ? How can i flash it with wire and usb to ttl with an  full flash and which one to take ?

thanks for your advice 

Best regards
Mathias

dartymath, 2024-11-20:
Thanks,
Yes for sure I already did it with 3 new battery that work on another device !

pvvx, 2024-11-20:
I have never seen such behavior with alternative firmware.
After flashing using a USB-COM adapter, it is always advisable to update the firmware via BLE OTA.
When flashing via BLE OTA, the firmware is checked for checksums twice.

dartymath, 2024-11-20:
Yes it was running well during several months ! Only issues after removing the battery. Now when I unplug and plug the battery I only see 1 second all digit appear and after stop.
I did same flashing in April on 2 devices only one get this issue
How to flash by usb com ?  Any tutorial for flashing it ?
Thanks

sanschaussure, 2024-11-22:
I see the same behavior. After changing battery (old battery was totally empty) the device sometimes starts sometimes not. I also try to reset the device (didn't work). Discharge the internal capacitors by connecting + and - on the battery position.

Model is LYWSD03MMC
Software version 4.6 or 4.7 

Regards

ohneschuh

pvvx, 2024-11-22:
When the voltage drops below the operating voltage (less than 2.0V), the thermometer switches on the deep-sleep mode for 2 minutes. After 2 minutes, the next check occurs, but the start will be if the voltage is more than 2.2V.
This is necessary at negative temperatures, when the battery voltage drops sharply. And also in cases of power supply from a solar battery - it is necessary to let the battery charge.
Otherwise, the device will never start ( -25..30C) or when operating from a solar battery, but will discharge the battery to zero.

sanschaussure, 2024-11-22:
What does it mean? The battery I used was fresh with 3.4 V. Do I have to put in the battery and wait for 2 Minutes?

pvvx, 2024-11-22:
If the battery connection occurred with poor contact - multiple connections/disconnections with a short pause, then most likely the thermometer will go into sleep mode for 2 minutes, waiting for normal contact with the battery.

dartymath, 2024-11-26:
dear all,
i try several time and get same issue,
i open the sensor, clean contact but no success.
i connect the sensor with stable power supply at 3 and 3.3V with same issue.
i see some times with the battery on it the screen flash again with LOW write on it but didn't understand why as the voltage is at 3V.
Any idea ?
Also how to flash the bin firmware directly with wire connection to Rx, Tx and gnd, V ? and with which tool ?
Thanks in advance

pvvx, 2024-11-26:
1. If you temporarily connect 5V to the thermometer power supply, then the chip fails - it starts to consume a lot or glitch.
2. On thermometers and other devices that have been operating for a long time in high humidity, short circuits or oxidation of the leads of chips and parts occur. This leads to the inoperability of the device. Cleaning the board and warming it up sometimes help temporarily, but not for long. I already have 3 such devices in 3 years.

dartymath, 2024-11-26:
Thanks for the read me I will try to flash it.
The sensor was only use inside the house in main room. I don’t think this was coming from humidity. The issue only occur after several months after the new firmware flash and when I see battery start to be low. I change the battery and the device only show screen test and turn off ! Switch of battery or usage of power supply at 3 or 3,3V didn’t change the state.
The other device didn’t show same issue

pvvx, 2024-11-26:
It is quite possible that the settings are incorrect - the screen is off, but the thermometer is working.
You can reset the settings in [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) or find the description in README ->
_If you have a smartphone with BT5.0+ and the nRFConnect program, then you can reset all settings to default by connecting to a thermometer and sending the 0x56 command, as shown in the screenshot:_ ....

sanschaussure, 2025-01-19:
Good Morning,

I just had a second device with same behavior. After changing battery, the device not starts, display stays off. Device is again a LYWSD03MMC, Software version 4.7.

1. I tried different new and partly used batteries. Fresh ones from the package and used from other LYWS.... Only the one I took out of working LYWS... works in the device. But after fast replacing the battery of the working one which first not working ones, the device starts smoothly.

2. In the next step I discharged the device by connecting both battery poles. The new batteries don't work. I powered it up with two AA-batteries. Direct start up.

3. In the last step I put a new battery (which formally not working) with a charged capacitor in parallel to the battery into the device. Direct start up and still running after removing the capacitor.

My interpretation: The startup of the device draws a lot of power which some coin cells cannot provide and the device stucks in and endless boot sequence. Depends on the brand of the cell. But all cells I tried, working and not working ones, are cheap Chinese ones.

My question: Is there the possibility to reduce the start-up power or to limit the power ramp-up within the software?

Regards.

Ohneschuh

pvvx, 2025-01-19:
The whole launch process:

[image]

The average consumption over 250 ms was 2.93 mA. Further typical consumption is 7..16 µA depending on settings.