pvvx/ATC_MiThermometer #571: Adding external sensor MY18B20 in devices with AAA battery (TS0201, ZTH01)

pvvx, 2024-11-24:
**Внимание:** Используются только (!) датчики **MY18B20**. Оригинальные датчики DS18B20 не работают от напряжения двух AAA батареек, имеют большее потребление и цену!

Распиновка MY18B20:

[image]

_Вариант с одним датчиком MY18B20:_
* Датчик припаивается к выводам модуля ZTU GND, VCC, PD2.
* Используется прошивка TS0201S1 или ZHT01S1.

_Вариант с двумя датчиками MY18B20:_
* Первый датчик припаивается к выводам модуля ZTU GND, VCC, PD2.
* Второй датчик припаивается к выводам модуля ZTU GND, VCC, PD4.
* Используется прошивка TS0201S2 или ZHT01S2.

---

[TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) пока не поддерживает интерфейс установки коэффициентов для данных датчиков, но fw термометра уже имеет данные функции. Будет дополнено в будущих версиях...

pav-tmn, 2024-12-25:
Такой upgrade возможен с BTH01 или любым другим работающим по BLE ?

pvvx, 2024-12-25:
Такая функция для чипсета на PHY62x2 пока не реализована.

dkbiz83, 2025-04-03:
> Такая функция для чипсета на PHY62x2 пока не реализована.

Добрый день, Виктор!
Подскажите, пожалуйста, могли бы вы мне помочь собрать прошивку для подключения двух внешних датчиков MY18B20 к ZG-227ZS v1.2 (ZHA опознает его как TS0601 by _TZE200_a8sdabtg)? Я понимаю, что это непростая задача, однако я уже пару недель пытаюсь найти в точности TS0201/ZTH01 на Aliexpress и, судя по описаниям и скриншотам покупателей там, везде уже поменялась компонентная база, даже если в описании продавец пишет по прежнему ZTH01. По сути я делал заказ ZG-227ZS v1.2 только ради вашего проекта и понимая, что по описанию он в точности соответсвует TS0601, а по факту он оказался ZG-227Z еще и на новой плате. Я был бы вам очень благодарен за помощь!

[image]
[image]
[image]
[image]

dkbiz83, 2025-04-03:
Сейчас, например, много лотов из тех, что доступны на Али, они продают ZTH02, называя его на коробке и на корпусе ZTH01. Скриншот одного из покупателей:

[image]

pvvx, 2025-04-04:
> Сейчас, например, много лотов из тех, что доступны на Али, они продают ZTH02, называя его на коробке и на корпусе 

[ZY-ZTH02](https://pvvx.github.io/ZY-ZTH02) -> https://github.com/pvvx/ATC_MiThermometer/issues/571#issue-2688370353

dkbiz83, 2025-04-04:
> > Сейчас, например, много лотов из тех, что доступны на Али, они продают ZTH02, называя его на коробке и на корпусе
> 
> [ZY-ZTH02](https://pvvx.github.io/ZY-ZTH02) -> [#571 (comment)](https://github.com/pvvx/ATC_MiThermometer/issues/571#issue-2688370353)

Благодарю вас! Заказал два ZY-ZTH02.
Подскажите, пожалуйста:
1. ZY-ZTH02 использует термодатчик Clone SHT30 - CHT832x. В отличие от CHT8305, CHT832x работает при более низких напряжениях как и датчики АНТХХ?
2. Выше вы также обновили описание, добавив ZTH02. Я не увидел прошивки для него с внешними термодатчиками, или он использует ту же самую прошивку, что и ZTH01? Также из ваших комментариев ранее я понимал, что такой мод возможен для устройств на двух ААА в виду повышенного энергопотребления. При этом, ZTH02 питается от таблетки.

pvvx, 2025-04-04:
> 1. ZY-ZTH02 использует термодатчик Clone SHT30 - CHT832x. В отличие от CHT8305, CHT832x работает при более низких напряжениях как и датчики АНТХХ?

Да - напряжение работы у других датчиков шире.

> При этом, ZTH02 питается от таблетки.

ZTH01-ZTH02 - У этих устройств одинаковая схема и прошивка идентична. Отличия только в названии. Без перепайки и выкидывания TuyaMCU они нормально не работают и прошивки только для перепаянного устройства и там все выводы соединяются одинаково.

> Я не увидел прошивки для него с внешними термодатчиками, или он использует ту же самую прошивку, что и ZTH01?

Прошивки с дополнением в имени "S1" или "S2".

pvvx, 2025-04-04:
> А прошить данные термометры из штатной прошивки в вашу зигби или блутус возможно только через программатор? 

В основном да. У проданных устройств какие-то ошибки в Zigbee OTA. Это обычно для Tuya - все они выпускаются чтобы продать и забыть. Купили и ладно - никаких гарантий у Tuya не предусмотрено.

dkbiz83, 2025-04-04:
> Обновление на альтернативные прошивки в Z2M и ZHA производятся путем копирования требуемого indexNNN.json из директория ZigbeeTLc/ ATC_MiThermometer прошивок. Или указания ссылки в конфигурации HA/ZHA/Z2M. Далее будьте внимательны, на какие устройства включать обновление. После обновления желательно удалить то, что было произведено для появления обновлений.

Да, я так и делал. Но проблема у меня - на странице ОТА в Z2M даже нет моего устройства, чтобы я мог для него проверить потом обновление или же начать обновление нажанием кнопки. Устройства просто нет в списке.

dkbiz83, 2025-04-13:
Виктор, добрый день! Хотел поделиться результатом - все получилось с  ZY-ZTH02, успешно подключил к нему два MY18B20 и прошил вашей прошивкой v5.1_2 через программатор. По ОТА мне это сделать не удалось. Вот несколько фото в добавок к вашему руководству выше.
Подскажите, пожалуйста, по двум вопросам:
1. А можно ли подключить теперь к ZY-ZTH02 управление реле? И возможно ли управлять реле от какого-то из этих внешних датчиков? Куда необходимо подключаться к ZY-ZTH02?
2. А для моего ZG-227ZS v1.2 подойдет ваша прошивка с индексом ZG227Z?

pvvx, 2025-04-13:
Реле подключается через MOSFET, т.к. другие типы коммутаторов требуют больших токов для включения.
Для ZTH02 выход "триггера" это GPIO_PB1 (обычно маркирован "TX", или смотрите PB1 на схеме модуля ZTU).

Описание установок дано тут:
https://github.com/pvvx/ATC_MiThermometer?tab=readme-ov-file#temperature-or-humidity-trigger-gpio-pa5-lywsd03mmc-label-on-the-reset-pin
Но, для версии с датчиком MY18B20, установки для температуры и влажности заменены на установки температуры для первого и второго MY18B20.

dkbiz83, 2025-04-13:
> Но, для версии с датчиком MY18B20, установки для температуры и влажности заменены на установки температуры для первого и второго MY18B20.

В настройках я увидел только настройку температуры и влажности. Я правильно вас понял, что несмотря на то, что они подписаны в TelinkMiFlasher.html как температура и влажность встроенного датчика, это по факту настройка температуры первого и второго внешнего датчика соответственно?

pvvx, 2025-04-13:
> В настройках я увидел только настройку температуры и влажности. Я правильно вас понял, что несмотря на то, что они подписаны в TelinkMiFlasher.html как температура и влажность встроенного датчика, это по факту настройка температуры первого и второго внешнего датчика соответственно?

Типа того. Я досконально не проверял и сменилось много версий... Проверял только один MY18B20.
Проверить быстро и не сложно... пробуйте, т.к. у меня все датчики настроены давно и я не хочу менять из установки... да и некогда...

pvvx, 2025-04-13:
@dkbiz83 - посмотрел исходные коды... к сожалению замена при наличии основного датчика не делается.
А я проверял на своем устройстве, без основного встроенного датчика...

Там возникают множественные варианты и они пока не решены.
Делать ещё дополнительные ветвления и сохранения ещё настроек увеличит всем код и ветвления в разных программах, включая TelinkMiFlasher...

А есть всего 2 настройки для "триггера". И пусть остаются 2.

Распределить можно так:

**Вариант со встроенным датчиком и 1 или 2 MY18B20:**
"Температура" - это Температура от внутреннего датчика,
"Влажность" - это Температура от первого внешнего датчика

Так выполняется больше обычно требуемых комбинаций, таких как поддержание температуры по показаниям внутреннего датчика и/или одного внешнего. И раз уж поставили доп. датчики температуры, то влажность явно не нужна.

"Температура" и "Влажность" аналогично меняются в других комбинациях датчиков на другие измеряемые типы.

А всего, в текущей выложенной версии, поддерживаемых "типов" датчиков уже за 15...
```
	TH_SENSOR_SHTC3,   // 1
	TH_SENSOR_SHT4x,   // 2
	TH_SENSOR_SHT30,	// 3
	TH_SENSOR_CHT8305,	// 4
	TH_SENSOR_AHT2x,	// 5
	TH_SENSOR_CHT8215,	// 6
	IU_SENSOR_INA226,	// 7
	IU_SENSOR_MY18B20,	// 8
	IU_SENSOR_MY18B20x2,	// 9
	IU_SENSOR_HX71X,	// 10
	IU_SENSOR_PWMRH,	// 11
	IU_SENSOR_NTC,		// 12
	IU_SENSOR_INA3221,	// 13
	IU_SENSOR_SCD41,	// 14
	IU_SENSOR_BME280,	// 15
```
и не все со всеми совместимы...

Сейчас пересоберу прошивки с новым вариантом...

dkbiz83, 2025-04-13:
Абсолютно с вами согласен по поводу достаточности 2х настроек и комбинации из двух термосенсоров. Я правильно понял, что вы сможете собрать прошивку, где настройки в TelinkMiFlasher смогут настраивать два датчика - или 1 встроенный и один внешний, или 2 внешних? И я, когда прошьюсь в эту прошивку, смогу это настроить и использовать?

pvvx, 2025-04-13:
**Вариант со встроенным датчиком и 1 или 2 MY18B20:**
"Температура" - это Температура от внутреннего датчика,
"Влажность" - это Температура от первого внешнего датчика

[image]

"Триггер" (включение реле или ещё чего, а так-же показание "Power") работает с двумя настройками: 
Если задан гистерезис у обоих - значит будет работать сразу с двумя. 
Если гистерезис задан у одного - значит будет работать только с ним.
Если гистерезис не задан ни у одного - значит отключен и управление выходом "Триггер" осуществляется в TelinkMiFlasher при соединении...

dkbiz83, 2025-04-13:
> 1. Если задан гистерезис у обоих - значит будет работать сразу с двумя - "включает" выход любой из двух показаний датчиков. "Отключает" выход триггера только состояние - у обоих отключено.

Вы имеете в виду, что это работа с встроенным и первым внешним датчиком?

> "Температура" - это Температура от внутреннего датчика,
> "Влажность" - это Температура от первого внешнего датчика

А это соответственно настройка внутреннего и первого внешнего датчика в TelinkMiFlasher?

pvvx, 2025-04-13:
> А это соответственно настройка внутреннего и первого внешнего датчика в TelinkMiFlasher?

Это настройки выхода "триггера".

Настройки коэффициентов поправок для двух MY18B20 не вписаны в TelinkMiFlasher.
Слишком много комбинаций накопилось от старых и новых вариантов и версий прошивок за 4-ре года и ныне произвожу переконфигурацию всего в новый общий вариант с распределением по флагам имеющихся сервисов в прошивке и т.д.
Для этого будет нужен переход на новую версию TelinkMiFlasher и многое другое. Но это ещё в работе...
Будет новая версия, отбросившая все старые версии. Произошло зарастание кода TelinkMiFlasher всяким устаревшим хламом и там уже не разобраться, не поломав поддержку древних версий.
Старые останутся в поддержке старого (текущего TelinkMiFlasher) .

dkbiz83, 2025-04-13:
> Это настройки выхода "триггера".

Да, я понял. Я имел в виду, что для "триггера" работают или внутренний, или внутренний+первый внешний датчики, даже если подключены 2 внешних. Это верно? 2й внешний датчик в этих комбинациях нигде не задействован, верно?

pvvx, 2025-04-13:
> 2й внешний датчик в этих комбинациях нигде не задействован, верно?

Да. Только на передачу.

dkbiz83, 2025-04-13:
Ок, т.е. чтобы мне настроить срабатывание реле только по 1му внешнему датчику, я настраиваю температуру и гистерезис для влажности в TelinkMiFlasher, верно?  И RS mode держу в режиме Switch?

[image]

А в чем смысл значения Power теперь? Ранее оно имело статус Off

dkbiz83, 2025-04-13:
> Все условия и переключения по значениям от датчиков проще написать в HomeAssistan.

Это вы имеете в виду создать в HA автоматизацию уже? Но это мне не поможет в управлении подключенном к термометру реле, ведь так?

pvvx, 2025-04-13:
> А в чем смысл значения Power теперь? Ранее оно имело статус Off

Показывает состояние вывода "GPIO_TRG" (Триггер).

dkbiz83, 2025-04-13:
> "GPIO_RS" - это геркон или дополнительно припаянная кнопка.
> Может считать импульсы от счетчиков или работать как открытие-закрытие двери, окна, или нажатия припаянной кнопки...

Теперь понял... Эта настройка к подключению реле не имеет отношения...

Виктор, не знаю как выразить вам свою благодарность! Вы лучший! Искал возможность как-то написать вам благодарность через приватные сообщения, но не нашел... Я вам очень благодарен!

pvvx, 2025-04-13:
> Это вы имеете в виду создать в HA автоматизацию уже? Но это мне не поможет в управлении подключенном к термометру реле, ведь так?

Если есть программа, которая будет передавать на термометр переключение выхода "триггера", то можно написать и такой сценарий в HA... Всё для этого в прошивке есть...

dkbiz83, 2025-04-13:
> Если есть программа, которая будет передавать на термометр переключение выхода "триггера", то можно написать и такой сценарий в HA... Всё для этого в прошивке есть...

Теперь понял! Но я пока не знаю как этого сделать. Ваше же руководства и настройки в прошивки достаточно просты для понимания...

dkbiz83, 2025-04-13:
> Показывает состояние вывода "GPIO_TRG" (Триггер).

Только что перепрошил термометр и проверил срабатывание и статус Power по первому внешнему датчику - все работает!

pvvx, 2025-04-13:
Если в настройках гистерезисов оба нуля, то работают:
 
[image]

dkbiz83, 2025-04-13:
> Если в настройках гистерезисов оба нуля, то работают:

Да, я это понял. И тогда можно управлять через НА?

pvvx, 2025-04-13:
Да, но никто за более 3-года не написал этого управления из HA.

dkbiz83, 2025-04-13:
Т.к., видимо, настроек внутри прошивки итак достаточно)

pvvx, 2025-04-13:
И состояние обоих триггеров (по температуре и влажности) отображается в TelinkMiFlasher и передается в маяке, но столько в формате cystom.

dkbiz83, 2025-04-13:
> И состояние обоих триггеров (по температуре и влажности) отображается в TelinkMiFlasher и передается в маяке, но столько в формате cystom. Замусоривать ещё и BTHome не стал.

Я понял. Теперь, благодаря вам, получилось сделать из этого китайского термометра крутой элемент автоматизации с управлением нагрузкой и всей инфой в НА. Низкий вам поклон!

pvvx, 2025-04-13:
В итоге в TelinkMiFlasher уже такие “наросты”, что вставить что-то новое уже очень трудно.
А у меня даже сам HA запитан от ИБП с BLE, и там такой бардак:

[image]

[image]

Т.е. все состояния токи в ИБП. 
А заряд АКБ включается тоже по триггеру но уже в единицах напряжения и т.д. :)

А так-же вентиляция по CO2 и т.д., набор воды в резервный бак, мотор скважины, кондиционер, .... И всё по BLE...

dkbiz83, 2025-04-13:
Я даже не догадывался, что возможно по блютус подключить такое большое количество устройств, и чтобы они стабильно работали...

Подскажите, я понял, что в НА реализовали даже все ваши настройки, если термометр прошив в зигби. А такие же настройки в самом НА возможно реализовать при блутус прошивке?

pvvx, 2025-04-13:
Примерно более сотни BLE устройств. Адаптер USB-BT максимально может принимать до 300 маяков в секунду.
У меня ещё столько устройств нет, чтобы полностью забить эфир основных каналов BLE.

dkbiz83, 2025-04-13:
> Примерно более сотни BLE устройств. Адаптер USB-BT максимально может принимать до 300 маяков в секунду. У меня ещё столько устройств нет, чтобы полностью забить эфир основных каналов BLE. Это не Zigbee - там всё печально... типа до 4-х в секунду.

Т.е. BLE лучше зигби? Я считал наоборот...
А у вас нет описания вашего проекта по переделке УПСа для такого мониторинга, как вы показали выше?

dkbiz83, 2025-04-13:
Я использую встроенный в Распбери 5 блютус. Его же тоже достаточно для всех этих целей? Или лучше внешний USB-BT?

pvvx, 2025-04-13:
> Т.е. BLE лучше зигби? Я считал наоборот...

И в режиме "Le Long Range" дальность связи примерно в 2 раза дальше чем у Zigbee и WiFi.

> А у вас нет описания вашего проекта по переделке УПСа для такого мониторинга, как вы показали выше?

[Тут что-то писал](https://esp8266.ru/forum/threads/rezervnoe-pitanie-usb-c-pd-qc3-0.6986/) 

> Я использую встроенный в Распбери 5 блютус. Его же тоже достаточно для всех этих целей? Или лучше внешний USB-BT?

Внешний USB-BT свисток примерно в 100 раз лучше...  По уровню сигналов приема и количеству принятых сообщений. Но надо выбирать правильный USB-BT свисток.

dkbiz83, 2025-04-13:
> Внешний USB-BT свисток примерно в 100 раз лучше... По уровню сигналов приема и количеству принятых сообщений. Но надо выбирать правильный USB-BT свисток.

Можете посоветовать? Я бы хотел заказать что-то стоящее, как у вас.

> Тут что-то писал

Я давно хотел сделать что-то подобное. Но я так понимаю, в той ветке вы выложили только базовую информацию...

pvvx, 2025-04-13:
> Можете посоветовать? Я бы хотел заказать что-то стоящее, как у вас.

У меня сейчас в HA воткнуто 3 USB-BT. Один на работу с "Le Long Range", другой для интеграции "bluetooth", третий для интеграции "Passive BLE monitor". Чтобы они не сорились и каждой интеграции досталось :) 

На сайте [Home Assistant](https://www.home-assistant.io/integrations/bluetooth/) правильно указаны востребованные BT адаптеры...

CSR8510A10 - для ближней связи и самый шустрый, но BT4.2.
RTL8761BU - обеспечивает прием от самых дальних устройств (поддержка уровня приема до -115 дБм) и работу в "Le Long Rane"
Третий у меня какой-то неизвестный китайский...
Все со встроенными антеннами, но RTL8761BU спозиционирован - ориентирован так, чтобы принимал через стены большинство уличных и BLE устройств из моего второго дома...

> Я давно хотел сделать что-то подобное. Но я так понимаю, в той ветке вы выложили только базовую информацию...

Её достаточно. Там всё из готовых модулей с али. Докинуто на макетке пара деталей, указанных на блочной схеме. 

PS: И совсем отклонились от темы. Всё уже неоднократно описано в issue и т.д. На этом дублирование инфы заканчиваю...

dkbiz83, 2025-04-13:
> PS: И совсем отклонились от темы. Всё уже неоднократно описано в issue и т.д. На этом дублирование инфы заканчиваю...

Благодарю вас!

dkbiz83, 2025-04-14:
> Для ZTH02 выход "триггера" это **GPIO_PB1** (обычно маркирован "TX", или смотрите PB1 на схеме модуля ZTU).
> 

Виктор, померял напряжение при сработавшем "триггере" (когда в НА статус Power = On и температура выше заданной через настройки) - около 0.1-0.2 В, при неработающем то же самое... Как определить, что триггер работает? Не могу понять

dkbiz83, 2025-04-14:
> Проверял напряжение между PB1 (=TX1) и землей

Методом тыка выяснил, что управление происходит пином PA1 - не наго подключается/отключается напряжение VBAT+

pvvx, 2025-04-14:
Я смотрел для TH03 -> [DEVICE_TH03Z](https://github.com/pvvx/ATC_MiThermometer/blob/master/src/app_config.h#L1047)

dkbiz83, 2025-04-14:
> Я смотрел для TH03 -> [DEVICE_TH03Z](https://github.com/pvvx/ATC_MiThermometer/blob/master/src/app_config.h#L1047)

ну я измерял напряжение на выводах PB1 и PA1. На PB1 постоянно 0.2В, на PA1 подается VBAT при срабатывании триггера

pvvx, 2025-04-14:
А надо было смотреть для [DEVICE_ZYZTH02](https://github.com/pvvx/ATC_MiThermometer/blob/master/src/app_config.h#L1924-L2150). Т.е. в итоге перепутал какой у вас термометр...

dkbiz83, 2025-04-14:
> А надо было смотреть для [DEVICE_ZYZTH02](https://github.com/pvvx/ATC_MiThermometer/blob/master/src/app_config.h#L1924-L2150). Т.е. в итоге перепутал какой у вас термометр...

Ну все ок, я выяснил) Спасибо! Вы же не будете в последующих прошивках менять этот пин тут?) Я спрашиваю, т.к. уже вывел проводок и хочу в термоусадку все запечатать, без доступа к модулю ZTU...

pvvx, 2025-04-14:
Не собираюсь, т.к. обычно ставлю на выведенные пины с маркировкой "TX", "TX1" или "TX2". А геркон на "RX", "RX1", ...
Исключение только Xiaomi LYWSD03MMC, так как там выведен пин с маркировкой "reset" и не надо разбирать. А это "триггер" появился как первое дополнение прошивки года наверно уже 4 назад...

dkbiz83, 2025-04-14:
РА1 вроде не TX1/TX2 на этом термометре

pvvx, 2025-04-14:
https://pvvx.github.io/ZY-ZTH02Pro/ = https://pvvx.github.io/ZY-ZTH02/
Pro - версия с LCD...

dkbiz83, 2025-04-14:
> https://pvvx.github.io/ZY-ZTH02/ :

Да, проверил у своего ZTH01-ZTH02 - действительно РА1 = TX2, сорри.

pvvx, 2025-04-15:
@dkbiz83 - скопировал для вас https://github.com/pvvx/ATC_MiThermometer/issues/638

dkbiz83, 2025-04-15:
> [@dkbiz83](https://github.com/dkbiz83) - скопировал для вас [#638](https://github.com/pvvx/ATC_MiThermometer/issues/638)

Ох, спасибо! Огромнейшее! Возможно ли с вами через вашу группу в телеграм связаться для личного контакта? Я думаю, вы там меня по моим вопросам для прошивки под ZG-227ZS видите) Но я вам там написать не могу...

dkbiz83, 2025-04-18:
> У меня сейчас в HA воткнуто 3 USB-BT. Один на работу с "Le Long Range", другой для интеграции "bluetooth", третий для интеграции "Passive BLE monitor". Чтобы они не сорились и каждой интеграции досталось :)

Виктор, подскажите, пожалуйста, а возможно ли настроить работу адаптера в "Le Long Range" на Home Assistant Operating System, установленной на Rasberry Pi5? Я видел вашу инструкцию по настройке в Линукс, но понимаю, что она мне не поможет?

pvvx, 2025-04-18:
На "Rasberry Pi5" используется очень плохой вариант встроенного BT адаптера. 
Если установите внешний BT адаптер на USB - всё будет работать.

dkbiz83, 2025-04-18:
Сорри, я не упомянул - я купил внешние USB-BT из рекоммендованных - один на CSR8510A10, второй на RTL8761BU. Но я не понимаю как на HA Operating System перевести RTL8761BU в режим Long Range. Видел вашу инструкцию по переводу его в этот режим на Линукс. Возможно ли это на HA Operating System?

pvvx, 2025-04-18:
Я не использую "HA Operating System". Не вижу смысла в этом...
И тут "issue" про MY18B20 - не замусоривайте ветки по нетематическим вопросам...

pvvx, 2026-02-15:
> Any suggestions for alternative sources for the MY18B20, or alternatives to the MY18B20?

DS18B20 have not been sold on Aliexpress for many years. They only sell clones = MY18B20.
And if you find a DS18B20, then its price will be astronomical.
MY18B20 is determined by a price ten times lower than that of DS18B20.

pvvx, 2026-02-15:
Сan try measuring the current consumption to find out how long the measurement takes and what currents are involved.
The documentation from www.maximintegrated.com for the DS18B20 doesn't match.
It lists a measurement time of 750 ms and a current of 1 mA.

[image]

[image]

But the sensors from AliExpress have:

[image]

Measurement time is 32 ms, and the current is 36 µA on the power line (3.3V).
This is more suitable for the MY18E20:

[image]

[image]