pvvx/ATC_MiThermometer #573: LYWSD03MMC freezes after random time

rk125, 2024-12-05:
I have some 12 LYWSD03MMC sensors around the house. 2 of them do freeze at random intervals, it might be week working or just some hours. Display is set to show humidity and battery, and while frozen values do not alternate any more. I take out battery, put it back and sensor works again. Battery is drained (level depends how fast I notice outage) and slowly recovers (can see on the graph). If I flash sensor with atc1441 image, it works for years without problem (setup and sensors working since 2020, wanted to move to more advanced image). I have v1.4 and v1.5 hw, both show the same behaviour. It seems location related, I would expect drops in readings, but cannot understand why sensor should freeze. Gateway is openmqttgateway if that matters, both active and passive modes are same. May be some ideas what could be the issue?
[image]

pvvx, 2024-12-05:
It looks like a connection is being made and something is being entered into the thermometer configuration.

rk125, 2024-12-06:
Oki, I will try pin code idea, will report back. Though I would expect that small bluetooth sign on frozen screen, if something would make connection before freeze, so frozen screen would have that sign. But I do not have it, screen shows just temperature and humidity or battery reading. And I would expect atc1441 freeze the same then.
By gateway I mean device between BT and ethernet (device sending mqtt messages) to get data into HASS or something, just like esphome BT proxy. For me HASS is on separate server in basement where there would be bad signal for BT network. In beginning I had OMG on ESP32 board (even old wroom with wireless IP connection), batteries on sensors worked fine for year and more with my network size (despite your warning that ESP32 is not good). Moved sensors to your firmware v4.3 june 2023, all except of those 2 sensors work flawlessly, batteries still holding. Then thought for bigger range and got rtl8761 dongle january 2024, OMG running on container on orangepi. Sensors work fine since then, batteries good, freezes are with those 2 sensors the same.
Area similar as yours, but wireless is everywhere (2.4GHz).
Thank you!

pvvx, 2024-12-14:
When the voltage drops below the operating level, the thermometer enters sleep mode for 3 minutes.

rk125, 2024-12-14:
Well, voltage was "normal" on this long working sensor, see in graph. Of course with next transmit battery + capacitor charge was not enough any more.
[image]
If I take sensor which freezes with still good battery, I see that voltage "restores" over the time - so seems it was not in "into sleep mode for minimum consumption (the best examples - less than 1 μA)", but rather actively using battery as if it were active connection to the sensor, but there is no connection.
[image]

pvvx, 2024-12-19:
The capacitor capacity is not enough to maintain power when starting a BLE connection.
The voltage quickly drops below the operating voltage, the processor does not have time to measure the voltage. Nothing is turned off, but on the contrary, it begins to consume more.
Then the battery is no longer able to charge the capacitor and  the spontaneously turned off chip...

The normal process looks like this:
When the temperature drops, the voltage of CR2032 drops.
The voltage is checked after periods set in the settings as the measurement period.

[image]

The same applies when powered by a solar battery...