pvvx/ATC_MiThermometer #575: "NotSupportedError: GATT operation failed for unknown reason" for MJWSD05MMC

changty6, 2024-12-06:
Hi, I just flashed the custom firmware BTH_v48.bin, rebooted device and tried to flash the Original_OTA_Xiaomi_MJWSD05MMC_v2.0.0_0026.bin, then I got the NorSupportError right after it cleaned the flash OTA region:

3:03:51 PM: Hardware Revision String: V2.3
3:03:51 PM: Software Revision String: V4.8
3:03:51 PM: Firmware Revision String: github.com/pvvx
3:03:51 PM: Detected custom Firmware
3:03:51 PM: Hardware Version: MJWSD05MMC V2.3, Software Version: 4.8
3:03:51 PM: Custom config HEX string: 55870000005004a93100095a
3:04:21 PM: Send Ext data: 223930000058026F ok
3:04:21 PM: ExtShow: big_number: 12345, small_number: 0, ext.vtime: 600 s, flg: 6F
3:04:27 PM: Send Ext data: 2239300000580267 ok
3:04:27 PM: ExtShow: big_number: 12345, small_number: 0, ext.vtime: 600 s, flg: 67
3:04:29 PM: Send Ext data: 223930000058026F ok
3:04:29 PM: ExtShow: big_number: 12345, small_number: 0, ext.vtime: 600 s, flg: 6F
3:04:35 PM: Send Ext data: 2239300000580267 ok
3:04:35 PM: ExtShow: big_number: 12345, small_number: 0, ext.vtime: 600 s, flg: 67
3:05:32 PM: Send Ext data: 223930000058026F ok
3:05:32 PM: ExtShow: big_number: 12345, small_number: 0, ext.vtime: 600 s, flg: 6F
3:05:43 PM: Send Ext data: 2239300000580267 ok
3:05:43 PM: ExtShow: big_number: 12345, small_number: 0, ext.vtime: 600 s, flg: 67
3:06:15 PM: Start DFU
3:06:17 PM: Clear Ext.OTA region (132kib)...
3:06:18 PM: Clean Flash OTA region: 0x040000, size: 132 kbytes. Go OTA...
3:06:23 PM: Update error: NotSupportedError: GATT operation failed for unknown reason.
3:06:23 PM: Disconnected.

Now it displayed normally, and the battery is pretty new (just bought it last week).

pvvx, 2024-12-06:
Should work on the second try.
Erasing large volumes of Flash and simultaneously running code from it is difficult :)

changty6, 2024-12-07:
Solved. Thank you!!

closed 2024-12-07 by changty6