pvvx/ATC_MiThermometer #576: CGG1 don`t connect to flash custom

xmanslava, 2024-12-11:
Hello. After reading the thread to the end, I could not figure it out. Please tell me if this sensor supports firmware updates. Sensor CGG1 M. I just have 2 more sensors of an older revision, I updated them 3 years ago, then flashed them to version 4.8, I am happy, I do not know grief. Firmware is a thing! But I can not flash this sensor, it does not connect. As I understand it, there is a non-Tellink chip there, right?
[image]

pvvx, 2024-12-11:
This thermometer has a different main chip.

xmanslava, 2024-12-11:
I understand you. It's a pity that I won't be able to flash it with your firmware. Thank you. I just didn't know, I thought they could all be flashed.
But overall, respect for the excellent firmware and your efforts!

closed 2024-12-11 by pvvx