pvvx/ATC_MiThermometer #584: Strange measurements reported at low (-18°C) temps.

recursinging, 2024-12-20:
Hi, 

first off, huge thank you for your work.  I have a deployment of over 100 LYWSD03MMC (SHT4x) flashed with ATC_v48.bin around our clean-rooms and building complex, with the measurement data being aggregated by several ESP32 gateways.  It works very well, and being able to gather so much data gives tremendous insight into the ambient gradients across rooms and buildings.

I received a request from our kitchen staff that they would like to have the temperatures of their chest freezers recorded over time and be alerted if there is a failure.  So, I soldered a coin cell battery holder and wire to the terminals of a couple LYWSD03MMC units, so that the battery can hang outside the freezer (the cold would dramatically shorten the life span). and started monitoring the temperatures via one of my gateways.  The v48 firmware configuration was left at default with the exception of the BLE advertising type which is set to "PVVX (Custom)". At first the temps were reporting normal, and sinking fast as the units cooled down, but 30 minutes later, as I went to integrate their data into a Grafana dashboard, I was met with some interesting numbers.

The sensors are advertising temperature values of around 630°C in freezers that are set to -18°C (the relative humidity values are being reported in normal ranges).  Obviously, this is some sort of scaling, or slope/offset issue related to the low temps, but I wasn't able to quickly find in the firmware source, where these calculations are being made, so I thought I'd ask.

Any ideas why this might be happening, and how I might mitigate it?

Cheers

pvvx, 2024-12-20:
The graph in the [README-eading-measurements-from-flash](https://github.com/pvvx/ATC_MiThermometer?tab=readme-ov-file#reading-measurements-from-flash)shows the operation of the LYWSD03MMC outdoors.
I have similarly been running several LYWSD03MMC outdoors and in a cold attic for several years now. No problems have been observed at temperatures below -30C.

LYWSD03MMC on the external unit of the air conditioner:
[image]

recursinging, 2024-12-20:
I see...  Which hardware version are you using in that case?   Are you convinced that this behavior could not have been introduced more recently? For example in 1825177def3ee65be8ea3c1f9fb4afe14ce64044?  

Can you point me to where the temperature value might be manipulated in the code base so I can debug it?

pvvx, 2024-12-20:
The value 630C cannot be transmitted. Maximum values ​​of int 16 bits = from -32786 to 32767. Temperature is transmitted in 0.01C. Total: -327.86..327.67C are maximum values.

recursinging, 2024-12-20:
I see, then perhaps there is a bug in my gateway implementation. I'll look there.  Thanks for your help!

closed 2024-12-20 by recursinging

pvvx, 2024-12-20:
Programs on ESP32 can throw out even more bugs. Programs are written in C++, and C++ requires a CPU with an MMU. Otherwise, there will be bugs and crashes due to inevitable memory defragmentation.

recursinging, 2024-12-20:
Yeah, I'm supplementing the measurement data at the gateway with some psychrometric calculations (dew point, absolute humidity, etc.) It is very possible that I've overseen something there.  Your int16 hint was key.   Thanks again for your work!

pvvx, 2024-12-20:
> Which hardware version are you using in that case? 

Home Assistant, 3 x USB-BT adapters.

recursinging, 2024-12-20:
Just a quick follow up.  The bug was in my gateway implementation. In parsing the temperature from your advertising packet I hadn't casted the result of the bitwise operation to int16_t.  With the temperature in going below 0, the signed two's complement format was wrongly being handled as unsigned, hence the high temperature.  All of the sensors I have deployed have been indoors, so I never ran across this bug until sticking one in a freezer. 😂