pvvx/ATC_MiThermometer #586: New revision (?) of MHO-C401 not recognized correctly

Mortim-zz, 2024-12-21:
I bought a few Mija MiaoMiaoCe on Aliexpress.
TelinkMiFlasher.html recognized them as MHO-C401N (see attached screenshot). After flashing MHO_C401N_v48.bin the screen is frozen.
I opened one, the PCB and screen ribbon look exactly like the non-N variant (see picture below).
After flashing MHO_C401_v48.bin everything works fine, so it seems this is just recognized incorrectly.
Do you need any more information to update the flasher or anything else?

[image]

[image]

pvvx, 2024-12-21:
Probably a new display...

https://github.com/pvvx/ATC_MiThermometer/issues/114#issuecomment-957880386

Mortim-zz, 2024-12-21:
It does have the lightning bolt in the battery symbol, so its C401 right?
[image]

pvvx, 2024-12-22:
They cannot be recognized programmatically using data from the original firmware.

Mortim-zz, 2024-12-22:
That's too bad. Here is the flash dump (ff.bin) if you want to have a closer look and/or extract the ota file: https://file.io/jlL9fJtYotIH

Before realizing this problem I flashed one with the Zigbee firmware, is there a way to undo that without soldering? Zigbee is working on this device, but the screen is not.

pvvx, 2024-12-22:
[Revert firmware from Zigbee to BLE version.](https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota)
https://github.com/pvvx/ATC_MiThermometer/blob/master/zigbee_ota/1141-0208-99993001-MHO_C401N_v49.zigbee

Mortim-zz, 2024-12-22:
Thanks a lot, now everything works.
If you want me to do/check something just tell me, I still have one with stock firmware.

pvvx, 2024-12-22:
I can flash any firmware or original FullFlash by myself, I have both MHO-C401/C401N.
Have to check out the ZigbeeTLc MHO-C401 variant. I'll post it in about 30 minutes.

changty6, 2024-12-26:
> That's too bad. Here is the flash dump (ff.bin) if you want to have a closer look and/or extract the ota file: https://file.io/jlL9fJtYotIH
>
> Before realizing this problem I flashed one with the [...]

Hi, I have the same problem here. Could you tell me how to undo the zigbee firmware? I couldn't find the device list of the flasher. Thanks!

Mortim-zz, 2024-12-28:
@pvvx It seems you fixed this in the last update, at least it was recognizing my last device correctly. Thanks a lot, then this can be closed.

@changty6 Copy the firmware for your device (according to the flashed firmware if it is the wrong version) and create the fitting index.json in the data directory of your z2m install, then add 
```
ota:
  zigbee_ota_override_index_location: index.json
```
to your configuration.yaml
Then you can flash the firmware in the z2m OTA tab.

closed 2024-12-28 by pvvx

changty6, 2025-01-10:
@Mortim-zz Thank you!