pvvx/ATC_MiThermometer #587: MJWSD05MMC Character data for Time and date wrong format

kezzkezzkezz, 2024-12-26:
[image]
[image]
trying both 12 hour and 24 Hour time formats show incorrectly, pressing top button and cycling through to to the time does show correctly

kezzkezzkezz, 2024-12-26:
I have an English version with the same model number

kezzkezzkezz, 2024-12-26:
[image]

pvvx, 2024-12-26:
> I have an English version with the same model number

I and other users have only a Chinese version.

kezzkezzkezz, 2024-12-26:
What do i need to do to either Dump the English firmware or change to correct it? i just picked up two of them today from Boxing day sales. i still have one at stock firmware

pvvx, 2024-12-26:
You need to read the firmware.
There are several options.
[Programmer ](https://github.com/pvvx/TlsrComSwireWriter) - [TLSR825xComFlasher.py](https://github.com/pvvx/TlsrComSwireWriter/blob/master/TLSR825xComFlasher.py)
or https://github.com/pvvx/TLSR825x_OTA_Flash_Hacker

---

Another option:
Publish a map of the segments of the new LCD.
1. Connect to MJWSD05MMC
2. Enter the command: "60N0N1..N17", where N0 - N17 bytes of the LCD buffer (HEX).

Enter one active data bit in the LCD-buffer and draw a card...
"60010000000000000000000000000000000000"
"60020000000000000000000000000000000000"
"60040000000000000000000000000000000000"
...
"60000000000000000000000000000000000080"

[Sample map.](https://github.com/pvvx/ATC_MiThermometer/blob/master/src/lcd_mjwsd05mmc.c#L25-L79)

huncrys, 2024-12-26:
I got the same SKU, I created the LCD segment mapping for it:

```c
/* MJWSD05MMC SKU: BHR9041GL LCD buffer:  byte.bit
                                                              __ __
  	 AM  PM 	  BLE         BAT  				  17.5( ^__^    ^   )17.5
  	 3.0 2.0	  0.0       [|17.4]#                17.6  17.7

         --3.4--         --2.4--            --1.4--         -17.0--      o 0.4
  |    |         |     |         |  1.0   |         |     |         |      +--- 0.4
  |   3.1       3.5   2.1       2.5  o   1.1       1.5   0.1      17.1     |
  |    |         |     |         |        |         |     |         |      ---- 0.5
 4.4     --3.2--         --2.2--    1.0     --1.2--         --0.2--        |
  |    |         |     |         |   o    |         |     |         |      |
  |   3.3       3.6   2.3       2.6      1.3       1.6   0.3      17.2    0.6
  |    |         |     |         |        |         |     |         |         o/ 0.7
         --3.7--         --2.7--     *      --1.7--         -17.3--           /o
                                   13.0

 Воскресенье  Понедельник  Вторник    Среда    Четверг   Пятница  Суббота
   Sunday       Monday     Tuesday  Wednesday  Thursday  Friday   Saturday
    4.5          4.6         4.7       4.3       4.2      4.1       4.0

      --5.0--         --6.0--               --7.0--         --8.0--
    |         |     |         |           |         |     |         |
   5.5       5.1   6.5       6.1      /  7.5       7.1   8.5       8.1
    |         |     |         |      /    |         |     |         |
      --5.6--         --6.6--      6.4      --7.6--         --8.6--
    |         |     |         |    /      |         |     |         |
   5.7       5.2   6.7       6.2  /      7.7       7.2   8.7       8.2
    |         |     |         |           |         |     |         |
      --5.3--         --6.3--               --7.3--         --8.3--

      --9.4--         -10.4--      10.0      -11.4--           -12.4--      o 9.0
    |         |     |         |     o      |         |       |         |      +--- 9.0
   8.4       9.5   9.1      10.5         10.1      11.5    11.1      12.5     |
    |         |     |         |    10.0    |         |       |         |      ---- 7.4
      --9.6--         --9.2--       o        -10.2--           -11.2--        |
    |         |     |         |            |         |       |         |      |
   9.4       9.7   9.3      10.6         10.3      11.6    11.3      12.6    5.4
    |         |     |         |            |         |   *   |         |
      --9.4--         -10.7--                -11.7--    11.0   -12.7--

          -13.4--         -14.4--            -15.4--       o 15.0
  |     |         |     |         |        |         |       +--- 15.0
  |   12.1      13.5  13.1      14.5     14.1      15.5      |
  |     |         |     |         |        |         |       ---- 15.1
 12.0     -12.2--         -13.2--            -14.2--         |
  |     |         |     |         |        |         |       |
  |   12.3      13.6  13.3      14.6     14.3      15.6     15.2
  |     |         |     |         |   *    |         |          o/ 15.3
          -13.7--         -14.7--    14.0    -15.7--            /o

none: 16.0..16.7
*/
```

Main difference is that `s3` can't show all symbols in the first digit, as bit `9.4` is shared (like in original version bit `6.0` in `s2`), so I lowered the threshold to `2999` and relocated `Hi`/`Lo` to the right.

Differences: https://github.com/huncrys/ATC_MiThermometer/commit/1023f0a5b110833ed28743bdb1776c14d08daad7

Full flash obtained with https://github.com/pvvx/TLSR825x_OTA_Flash_Hacker:
[MJWSD05MMC_BHR9041GL_ff.zip](https://github.com/user-attachments/files/18255970/MJWSD05MMC_BHR9041GL_ff.zip)

Custom firmware: https://github.com/huncrys/ATC_MiThermometer/raw/1023f0a5b110833ed28743bdb1776c14d08daad7/BTH_v48.bin

Tested it on my devices, LCD renders correctly:
  [image]

Additional images:
[image]
[image]
[image]
[image]

kezzkezzkezz, 2024-12-26:
> I got the same SKU, I created the LCD segment mapping for it:
> 
> ```c
> /* MJWSD05MMC SKU: BHR9041GL LCD buffer:  byte.bit
>                                                               __ __
>   	 AM  PM 	  BLE         BAT  				  17.5( ^__^    ^   )17.5
>   	 3.0 2.0	  0.0       [|17.4]#                17.6  17.7
> 
> 
>          --3.4--         --2.4--            --1.4--         -17.0--      o 0.4
>   |    |         |     |         |  1.0   |         |     |         |      +--- 0.4
>   |   3.1       3.5   2.1       2.5  o   1.1       1.5   0.1      17.1     |
>   |    |         |     |         |        |         |     |         |      ---- 0.5
>  4.4     --3.2--         --2.2--    1.0     --1.2--         --0.2--        |
>   |    |         |     |         |   o    |         |     |         |      |
>   |   3.3       3.6   2.3       2.6      1.3       1.6   0.3      17.2    0.6
>   |    |         |     |         |        |         |     |         |         o/ 0.7
>          --3.7--         --2.7--     *      --1.7--         -17.3--           /o
>                                    13.0
> 
>  Воскресенье  Понедельник  Вторник    Среда    Четверг   Пятница  Суббота
>    Sunday       Monday     Tuesday  Wednesday  Thursday  Friday   Saturday
>     4.5          4.6         4.7       4.3       4.2      4.1       4.0
> 
>       --5.0--         --6.0--               --7.0--         --8.0--
>     |         |     |         |           |         |     |         |
>    5.5       5.1   6.5       6.1      /  7.5       7.1   8.5       8.1
>     |         |     |         |      /    |         |     |         |
>       --5.6--         --6.6--      6.4      --7.6--         --8.6--
>     |         |     |         |    /      |         |     |         |
>    5.7       5.2   6.7       6.2  /      7.7       7.2   8.7       8.2
>     |         |     |         |           |         |     |         |
>       --5.3--         --6.3--               --7.3--         --8.3--
> 
> 
>       --9.4--         -10.4--      10.0      -11.4--           -12.4--      o 9.0
>     |         |     |         |     o      |         |       |         |      +--- 9.0
>    8.4       9.5   9.1      10.5         10.1      11.5    11.1      12.5     |
>     |         |     |         |    10.0    |         |       |         |      ---- 7.4
>       --9.6--         --9.2--       o        -10.2--           -11.2--        |
>     |         |     |         |            |         |       |         |      |
>    9.4       9.7   9.3      10.6         10.3      11.6    11.3      12.6    5.4
>     |         |     |         |            |         |   *   |         |
>       --9.4--         -10.7--                -11.7--    11.0   -12.7--
> 
> 
>           -13.4--         -14.4--            -15.4--       o 15.0
>   |     |         |     |         |        |         |       +--- 15.0
>   |   12.1      13.5  13.1      14.5     14.1      15.5      |
>   |     |         |     |         |        |         |       ---- 15.1
>  12.0     -12.2--         -13.2--            -14.2--         |
>   |     |         |     |         |        |         |       |
>   |   12.3      13.6  13.3      14.6     14.3      15.6     15.2
>   |     |         |     |         |   *    |         |          o/ 15.3
>           -13.7--         -14.7--    14.0    -15.7--            /o
> 
> none: 16.0..16.7
> */
> ```
> 
> Main difference is that `s3` can't show all symbols in the first digit, as bit `9.4` is shared (like in original version bit `6.0` in `s2`), so I lowered the threshold to `2999` and relocated `Hi`/`Lo` to the right.
> 
> Differences: [huncrys@1023f0a](https://github.com/huncrys/ATC_MiThermometer/commit/1023f0a5b110833ed28743bdb1776c14d08daad7)
> 
> Full flash obtained with https://github.com/pvvx/TLSR825x_OTA_Flash_Hacker: [MJWSD05MMC_BHR9041GL_ff.zip](https://github.com/user-attachments/files/18255970/MJWSD05MMC_BHR9041GL_ff.zip)
> 
> Custom firmware: https://github.com/huncrys/ATC_MiThermometer/raw/1023f0a5b110833ed28743bdb1776c14d08daad7/BTH_v48.bin
> 
> Tested it on my devices, LCD renders correctly: ![MJWSD05MMC_BHR9041GL_custom_firmware](https://private-user-images.githubusercontent.com/1208344/398811874-08d81e82-fe59-4b4a-8b81-e7d0545d899b.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUyNTU3NjgsIm5iZiI6MTczNTI1NTQ2OCwicGF0aCI6Ii8xMjA4MzQ0LzM5ODgxMTg3NC0wOGQ4MWU4Mi1mZTU5LTRiNGEtOGI4MS1lN2QwNTQ1ZDg5OWIuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIyNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMjZUMjMyNDI4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MDJhZDAwYTE1ZDMzNjQyYTJjNDIxZTQ4ZjI1NmY5MGNmMGUwMGFjMzg0ZGZlOThhODI3YmI0NDA2MGQ5ZTQyYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.FLiMfB3a6HIWCRSaUoJX-eT22m1V6IWFfmUbd2Pd-WA)
> 
> Additional images: ![MJWSD05MMC_BHR9041GL_pcb](https://private-user-images.githubusercontent.com/1208344/398812525-5f1f1fd8-ba94-46d4-a7a8-0269adbe8825.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUyNTU3NjgsIm5iZiI6MTczNTI1NTQ2OCwicGF0aCI6Ii8xMjA4MzQ0LzM5ODgxMjUyNS01ZjFmMWZkOC1iYTk0LTQ2ZDQtYTdhOC0wMjY5YWRiZTg4MjUuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIyNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1 [... cut ...] anBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIyNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMjZUMjMyNDI4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDU2ZmQ4ZmY0NTc5MzI0YzExOWY3ZTIxYmYxMGI3NGI4NzA3MDNmYmM1ZmIwMTk3MDAwZGEwZWY3ODMxNWQ2MiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.A5U2ql42uFoKFWpp1qlGtLn_egoUKhUBFOiU_Nq-irE)

Absolute legend, i was going to do a firmware dump today, Can confirm works on both of mine

pvvx, 2024-12-27:
How to distinguish MJWSD05MMC CH or EN version during initial firmware update?

ddijkstra, 2024-12-27:
I also have this global version. 
I can also confirm this adapted version is working. 

Perhaps add both screen versions to the same firmware and make it selectable via config?  
Defaulting in global version (just my preference :) ) and make it selectable via BT on the flasher page after login?
Advantage : only one firmware to manage. If more display versions appear, they can be easily added. Firmware update will use the old settings. 
Disadvantage : firmware will increase in size and complexity a bit due to new display maps, and initial firmware creation will take a little more time to incorporate the display selection.  

Or a maybe simpler option: have 2 different versions on the flasher page, but ask if the uses wants the global(en) or the chinese(cn) version flashed.
Advantage : Less work on creating the first firmwares, correct language is detected during upgrade by version using flasher. 
Disadvantage : If firmware changes happen, both need to be adapted/kept in sync.

huncrys, 2024-12-27:
> How to distinguish MJWSD05MMC CH or EN version during initial firmware update?

Flashed the CH version original OTA from https://github.com/pvvx/ATC_MiThermometer/raw/refs/heads/master/Original_OTA_Xiaomi_MJWSD05MMC_v2.0.0_0026.bin and the only difference I could find, apart from serial number, is that firmware revision string (`00002a26-0000-1000-8000-00805f9b34fb` ) and software revision string (`00002a28-0000-1000-8000-00805f9b34fb`) are both `0005` on the EN version, opposed to `0026`. So if the CH model always comes with `0026` revision, this could be used to determine which custom firmware to offer initially.

nRF Connect configuration: https://gist.github.com/huncrys/510694c3a6098df9191b2dcef115cf52

pvvx, 2024-12-27:
Neither version is suitable.
If you are asked a question and you do not know that there is another option, then the answer may be wrong.

The number for MJWSD05MMC(en) () version is already assigned. And support for https://github.com/pvvx/ATC_MiThermometer/blob/master/BTE_v49.bin is already written into TelinkMiFlasher.html.

And: ZigBee OTA is selected by the ID of the transmitting device, not by user selection....

pvvx, 2024-12-27:
@huncrys
Changes have been made but need to be tested on a real device.
BTE_v49.bin, TelinkMiFlasher.html ...

huncrys, 2024-12-27:
@pvvx 

Thanks, checked it:

- Correct firmware is offered when on stock:
  [image]
- OTA works, device is operational afterwards:
	```
    14:23:38: Searching for devices
	14:23:44: Connecting to: BTE_8FD8A5
	14:23:49: Hardware Revision String: V2.3
	14:23:49: Software Revision String: V4.9
	14:23:49: Firmware Revision String: github.com/pvvx
	14:23:49: Detected custom Firmware
	14:23:49: Hardware Version: MJWSD05MMC(en) V2.3, Software Version: 4.9
	14:23:49: Custom config HEX string: 55870000005004a931000c5a
  	```
- Correct firmware is offered when on BTE:
  [image]

Found one issue with configuring though, `Set Config` triggers this error:
```
TelinkMiFlasher.html:3077  Uncaught TypeError: Cannot read properties of null (reading 'checked')
    at SendCustomConfig (TelinkMiFlasher.html:3077:26)
    at HTMLButtonElement.onclick (TelinkMiFlasher.html:1:1)
```

PR for fix: https://github.com/pvvx/pvvx.github.io/pull/27

With this applied, everything seems to work.

ddijkstra, 2024-12-27:
> If you are asked a question and you do not know that there is another option, then the answer may be wrong.
True, however in my opinion there are options that can cause more user issues than asking the user the question : is the display displaying the day in chinese (周一、周二、周三、周四、周五、周六、周日), or English (Mon, Tue, Wed, Thu, Fri, Sat, Sun) 

That being said, as huncrys mentioned that there is a way to detect between the 2 initial versions, this would be the best to eliminate to ask the user this question.

I have tested (the display part of) the BTE_v49.bin version as well and see it is working fine.

pvvx, 2024-12-27:
> is the display displaying the day in chinese (周一、周二、周三、周四、周五、周六、周日), or English (Mon, Tue, Wed, Thu, Fri, Sat, Sun)

In TelinkMiFlasher.html there is an option to disable the display of days of the week.

huncrys, 2024-12-27:
@ddijkstra 
> That being said, as huncrys mentioned that there is a way to detect between the 2 initial versions, this would be the best to eliminate to ask the user this question.

pvvx already implemented the detection in https://github.com/pvvx/pvvx.github.io/commit/faec1df5c953146048613c4690079bd719af1e46

kezzkezzkezz, 2024-12-30:
Awesome work guys, you've given me a reason to go buy two more of these lol

closed 2025-02-11 by pvvx