pvvx/ATC_MiThermometer #589: ZTH05 v48 and v49 dont work with aht20 sensor on U4 position

Nikpesu, 2024-12-27:
ZTH05 [v48](https://github.com/pvvx/ATC_MiThermometer/blob/master/ZTH05_v48.bin) and [v49](https://github.com/pvvx/ATC_MiThermometer/blob/master/ZTH05_v49.bin) don't work with "S8310" (found it in Tuya TH01 [1](https://github.com/esphome/issues/issues/5289)/[2](https://www.elektroda.com/rtvforum/topic4019974.html)) sensor on U4 position. You can see in picture.

log for config:
4:37:35 AM: Hardware Version: ZTH05 V1.1, Software Version: 4.8
4:37:35 AM: Custom config HEX string: 55871000005004a93131215a00

[image]
[image]
[image]
[image]

pvvx, 2024-12-27:
[CHT8310](https://github.com/pvvx/pvvx.github.io/blob/master/THB2/CHT8310%20Advanced%20Datasheet%20Ver1.1%2020230927.pdf).

The ZTH05Z fw does not include the CHT8310 sensor.
It will be in the next version, in a few days...

Nikpesu, 2024-12-27:
Still nothing
[image]

pvvx, 2024-12-27:
ANT30 - ok
```
17:46:50: Connecting to: TH5_F8CA3B
17:46:54: Hardware Revision String: V1.1
17:46:54: Software Revision String: V4.9
17:46:54: Firmware Revision String: github.com/pvvx
17:46:54: Detected custom Firmware
17:46:54: Hardware Version: ZTH05 V1.1, Software Version: 4.9
17:46:54: Custom config HEX string: 55871000005004a93131215a
17:46:58: Get Sensor Settings...
17:46:59: Sensor ANT20/30 id: 00200018, I2C address: 0x70, Kt: 1250, Kh: 625, Zt: -5000, Zh: 0
```

[image]

[image]

[image]

Nikpesu, 2025-01-18:
Any new changes? v49 doesn't work for me