pvvx/ATC_MiThermometer #591: Xiaomi Mijia (LYWSD03MMC) - Inconsistent battery readings

EddieTH, 2025-01-01:
For some time now I have had a low battery message (less than 2V) on the display of one of my Xiaomi sensors. At the same time, the sensor normally reports temperature and quite a different voltage.
The soft is version 4.8, I'm using BTHome integration, all on standard settings. 
[image]
[image]
[image]
[image]

pvvx, 2025-01-01:
Clean the battery contacts.

EddieTH, 2025-01-01:
The battery contacts are clean. Anyway, as you can see in the graph the voltage jumps by only a few mV. I also have the capacitors on the board doped. The battery is indeed weak and that is not in dispute. I wonder if it would be useful to be able to disable the mV indication on the LCD and leave only the weak battery logo. For me that would be enough information, and let the thermometer run until the last mAh in the battery.

pvvx, 2025-01-01:
The battery voltage varies with the load current during and before the measurement.
The average current during activity varies with the program branching and the type of scanning by the external adapter. During active scanning, the thermometer transmits an additional packet upon request. This increases the load on the battery.
The request during active scanning does not occur every time. As a result, we get different load on the battery.
After averaging several measurements of the battery voltage (under a load of about 3..5 mA), we get a curve. Without averaging, the graph will show noise with a large amplitude.

pvvx, 2025-01-01:
> run until the last mAh in the battery.

Below 2.0 V operation is not recommended. The chip has a working minimum of 1.8 V. When writing Flash and other actions, the 
current increases and the battery voltage drops below...

If you do not monitor the 2.0 V level, the chip will freeze and the current from the battery will increase sharply. As a result, after the first drop below 2.0 V, the battery will discharge.
When registering the threshold of 2.0 V, the program switches the chip to deep sleep mode for several minutes with minimal consumption. The next start will be possible if the battery voltage under the measuring load is more than 2.2 V.

EddieTH, 2025-01-01:
Of course, I understand that with a weak battery, miracles can happen. It is possible that the addition of capacitors even made things worse. The charge in them allows the thermometer to revive for a while and then everything dies again. 
In any case, thanks for your work and best wishes for the new year!

closed 2025-01-01 by EddieTH

pvvx, 2025-01-01:
When the temperature drops, the battery voltage drops. The algorithm used allows for recovery when the temperature rises. The same applies to the solar panel. Otherwise, after the first voltage drop, everything will be discharged to zero and will not allow recovery. But the current algorithm cannot provide all options.