pvvx/ATC_MiThermometer #593: Xiomi LYNWSD03MMC new HW revision won't flash OTA

elf55, 2025-01-03:
Telink OTA flasher fails to flash HW rev B1.5 (08.2024)

2:48:19 PM: Hardware Revision String: B1.5
2:48:19 PM: Firmware Revision String: 2.1.1_0159
2:48:19 PM: Detected Mi device
2:48:19 PM: (!) Version: 2.1.1_0159 is temporarily not supported (!)

VivantSenior, 2025-01-04:
It's a known limitation, please see [comment](https://github.com/pvvx/ATC_MiThermometer/issues/378#issuecomment-1756740533). It's really easy to flash it via UART and [USBCOMFlashTx](https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html). I was able to convert 6 LYWSD03MMCs using UART without issues.

pvvx, 2025-01-04:
[README](https://github.com/pvvx/ATC_MiThermometer): "Warning: https://github.com/pvvx/ATC_MiThermometer/issues/378#issuecomment-1756740533"

closed 2025-01-04 by pvvx