pvvx/ATC_MiThermometer #594: LYWSD03MMC only  reporting signal strcngt missing temperature and humidity

Bach-Danielsson, 2025-01-04:
My device only reporting signal strength so i miss temperature and humidity data.
Flashed two days ago for HomeAssistant BLE transfer

huncrys, 2025-01-05:
@Bach-Danielsson try enabling AdFlags in config with https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html

pvvx, 2025-01-05:
And the transmission format: "BTHome v2"

Bach-Danielsson, 2025-01-06:
[image]
[image]
[logs_outdoor-unit_logs.txt](https://github.com/user-attachments/files/18315650/logs_outdoor-unit_logs.txt)
[image]
[image]

babelhoo, 2025-01-09:
Had the same issue, and doing as pvvx said was the solution!
I changed Advertising type to PVVX (custom), cleared AdFlags and set GPIO_RS to None, and now it works on ESPHome.
Thanks!

pvvx, 2025-01-09:
ESPHome uses outdated formats.
In future versions, all firmware will switch to the BTHome format.
For some options, the Mijia (MI-Home) format will remain.
ESPHome should introduce support for new formats.