pvvx/ATC_MiThermometer #595: Remove humidity "smiley" from display

josephernest, 2025-01-06:
Thank you for this project.

When using the Xiaomi or Tuya, can we remove the annoying humidity smiley from the display, with this firmware?

    (^_^) 50%
    (‾^‾)  70%  <- this is anxiogen and I don't want to see this everywhere (in certain places 70% is ok: outside, etc.)

[image]

closed 2025-01-20 by pvvx

josephernest, 2025-01-20:
How to do it @pvvx ? Where are the settings to disable these smiley displays ?

pvvx, 2025-01-20:
[image]