pvvx/ATC_MiThermometer #600: frozen screen after using LCD segment mapper

freebase1ca, 2025-01-13:
Hello,  I used the Windows web version of Telink Flasher for Mi Thermostat to flash the custom firmware BLE MHO_C401_v49.bin onto three MHO-C401 e-paper devices I just bought.  They are all currently connecting to my Home Assistant and updating temps and humidity as they should.

However, on one device I chose to play with the LCD segment mapper.  Ever since then, my device has a frozen display.  Everything else still works.  I've found no documentation on the segment mapper and have not figured out how to undo my changes.

I've tried resetting the device, removing the battery, reverting to the original firmware and back to v49, flashing settings, setting to default, setting various segments.  Nothing helps.  The screen remains frozen.

Any thoughts on how to get this screen updating?  Cheers!

pvvx, 2025-01-20:
MHO-C401 is supported by two versions. They have different displays.
MHO-C401 and MHO-C401N

freebase1ca, 2025-01-20:
Yes, but I know I have the correct firmware because I had all three of my
devices working. Only one of them froze when I played with the segments.

chrisjantzen, 2025-02-04:
I've ran into the same issue. I only have 1 device, it is also a MHO-C401, I also tried the same thing. I flashed it and played with all the settings, including the segment mapper. After doing that the screen froze. It still reports data to home assistant, but the screen won't update. I've tried replacing the battery, flashing the old software, reflashing the new software, and using the reset button. Nothing has got the screen working again. So far it seems like the screen was bricked by these settings.

chrisjantzen, 2025-02-05:
I was able to get my device working again by downloading and flashing the "Original_OTA_Xiaomi_MHO_C401_v1.0.0_0016.bin" version of the original firmware. After that, the screen updates and starts working again. Unfortunately, when re-flashing the custom firmware the screen immediately stops updating again. I tried playing around with the custom firmware settings but none of the changes seemed to get the screen working again sadly. For now it's back to the original software I guess.

chrisjantzen, 2025-02-05:
I opened it up and took a picture of the board. Most of the pictures I'm seeing in other Git Issues are the 2020 revision of this device, whereas this board is 2021. Perhaps there is some difference?

[image]

Toniob, 2025-03-28:
I had the same issue as you. Did you try to flash the C401N version manually ? By default, the web flasher propose you the non-N version, but you can download and flash the other one. I have the same PCB as you and it did work for me.

pvvx, 2025-03-28:
> I opened it up and took a picture of the board. Most of the pictures I'm seeing in other Git Issues are the 2020 revision of this device, whereas this board is 2021. Perhaps there is some difference?

The printed circuit boards are identical. The only difference is in the display and its built-in controller on the cable.

chrisjantzen, 2025-04-25:
Thanks guys! I gave this another shot and flashed the latest version of the C401N firmware (MHO_C401N_v52.bin). That seems to have done the trick. I was able to get it working this time with the new firmware. Appreciate the guidance!