pvvx/ATC_MiThermometer #602: Firmware via OTA for LYWSD03MMC version 2.1.1_0159

pvvx, 2025-01-14:
**Обновление LYWSD03MMC ver2.1.1_0159**

В текущем варианте требуется:

1. Зарегистрировать термометр в MiHome.
2. С помощью альтернативного ПО получить все ключи для данного термометра.
3. Запустить [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) и соединиться с термометром.
4. Ввести "Device known id", "Mi Token", "Mi Bind Key" и выбрать "Login" 
5. Выбрать прошивку "ATC_vNN" и запустить OTA.

[Альтернативные варианты от "atc1441"](https://github.com/atc1441/ATC_MiThermometer/issues/298):

> Video Demo of Full OTA update of 2.1.1_0159 aka no wires needed. https://youtu.be/NfZHh6wmTp8

pvvx, 2025-01-14:
For unknown, different from the latest version FW 2.1.1_0159, wait for the new version (v9.1) TelinkMiFlasher.html.

BelgarionNL, 2025-01-14:
yep works!

fl0rian089, 2025-01-14:
just tried this and it worked, thank you very much. 

which zigbee firmware is possible after the atc exploit?

itsmepete, 2025-01-15:
Yesterday I followed your work-around with v9.0 to flash to custom ATC. That worked fine as described.
In the next step I flashed with the following file to get it zigbee ready. https://github.com/devbis/z03mmc/releases/download/1.1.0/z03mmc.bin

The log file said it was disconnected due to an unknown error and since then the screen stays off. I have tried to reset it by taking out the battery and also by shorting the RESET & GROUND without success. 

But as of the latest post I saw that there is actually a newer file for the zigbee binary. I was not aware of that. 

So I repeated everything following your guide above with another sensor that I have. Works for me.

Any chance I can revive the first sensor that apparently was flashed with an older .bin file?

Thanks, really appreciate the guide above with step-by-step instruction.

adnn-alc, 2025-01-15:
Any idea what do do if device is kind of bricked? Tried UART flashing but screen is blank, only slightly visible if I look at it from the side. Help is appreciated.

pvvx, 2025-01-16:
> Yesterday I followed your work-around with v9.0 to flash to custom ATC. That worked fine as described. In the next step I flashed with the following file to get it zigbee ready. https://github.com/devbis/z03mmc/releases/download/1.1.0/z03mmc.bin

This firmware is from devbis and it is not adapted for loading via standard BLE OTA.
You will have to act as indicated on https://github.com/devbis/z03mmc/blob/master/README.md.

mfatgt, 2025-01-16:
Hello @pvvx,

Thanks a lot for your valuable work. 

on SKU: NUN4106CN

I have first flashed the original stock, then the exploit. The display showed all letters afterwards. After removing battery, it reverts back to original stock.

mfatgt, 2025-01-16:
Sorry, I meant the custom firmware.

pvvx, 2025-01-16:
Original_OTA_Xiaomi_LYWSD03MMC_v2.1.1_0159c.bin ?

pvvx, 2025-01-16:
On some BT adapters, a reset is required after each reflashing. Otherwise, the adapter remembers the old version of the previous firmware headers.
If it is USB-BT, then reconnect the adapter to the USB port.
In other cases, you need to restart your computer or Explorer. Reset cache.

dasrohr, 2025-01-17:
Thank you, guys.
Thank you for your persistence and all the work you put into this project 

I just bought a bunch of these thermometers which came with the new firmware from the factory. 
Luckily, you already worked out how to flash those without UART.

Can confirm that the new method is indeed working. Flashed 10 Decices with Custom Firmware and converted them to Zigbee.

thanks again, and keep up this incredible work.

Andurilll, 2025-01-17:
amazing how you fixed this problem with the newest firmware! Thank you!

mackva, 2025-01-31:
Привет @pvvx , спасибо за работу! 

Я обновил прошивку на ATC_v49. Как написано в первом сообщение. На днях заметил что дисплей аюпоказывет старые данные, по блютузу шлются актуальные. Если вытащить батарейку, то дисплей тухнет, но датчик работает и шлёт данные. 

Если перепрошить ещё раз дисплей начинает работать, но через 1-2 дня зависает.

pvvx, 2025-01-31:
Кто-то включает опцию "отключение экрана"?
На B1.5 именно так это работает.

griga123456, 2025-02-05:
Hi! Zigbee'd mine, but now I have a question...
Why is the smiley face always sad? My room temp is 24C, humidity is 34%. Well within the limits that are set in the manual. Is it because of the hack?

pvvx, 2025-02-06:
> My room temp is 24C, humidity is 34%. 

Default: 20..25C, 40..60%.

[Google:](https://www.google.com/search?q=INDOOR+AIR+TEMPERATURE+AND+HUMIDITY+PARAMETERS) The ideal average (office) room temperature varies between 20 and 23 °C and from 40 and 60 % relative humidity.

Depending on the level of humidity, the following room modes are distinguished: dry (less than 40%), normal (40÷50%), humid (50÷60%) or wet (over 60%).

griga123456, 2025-02-06:
> Depending on the level of humidity, the following room modes are distinguished: dry (less than 40%), normal (40÷50%), humid (50÷60%) or wet (over 60%).
Ok, i gotcha on this.
> Replace the hub with a normal one that supports Zigbee 3.0. A normal modern hub has a better reception level, which affects the communication distance.
Yandex Station Midi supports Zigbee 3.0, it's in the documentation. I don't want to use a standalone router, that defies the reason why i bought the little dingus. Plus the Speaker-hub is literally less than a meter away from each other. 
The most frequently-polled data is the humidity, which, according to the graphs, was polled almost every minute, but for some reason the entire night was practically empty. Temperature is bad: no updates for over 6 hours at this point. Battery reports have the same time stamps as temperature. What could be wrong?

pvvx, 2025-02-06:
> Yandex Station Midi supports Zigbee 3.0, it's in the documentation. 

Learn Zigbee 3.0 and set the correct Binding Clusters and Reporting parameters in Yandex Station Midi.

[README](https://github.com/pvvx/ATC_MiThermometer/blob/master/README.md): _Support for Zigbee firmware is in the [ZigbeeTLc](https://github.com/pvvx/ZigbeeTLc) repository._

mackva, 2025-02-09:
> Кто-то включает опцию "отключение экрана"? На B1.5 именно так это работает.

Hello, @pvvx ! 

i don't understand my case. i have 3 sensors (two with B1.5 hardware revision  and one 1.2 hardware revision).

One senser with 1.5 has problem described above. After your answer i flashed different firmwares.  

it broke. i took out the battery for a week. on the weekend i decided to connect it and it came back to life and now it works well.

it looks like some kind of hardware error

Thank you!

Neverhood2005, 2025-02-13:
у меня есть термометр  LYWSD03MMC. На экране температура была в градуса Цельсия.Оригинальная прошивка была "Firmware Revision String: 2.1.1_0159",  "Hardware Revision String: B1.5", "Software Revision String: 0159". Обновил прошивку сначала на "ATC_v50.bin", затем на прошивку "Z03MMC_v0124.bin" для использования ZigBee в умной колонке Алиса Миди.
Прибор успешно прошился, подключился к Алиса Миди, но теперь на экране вместо градусов по Цельсию (o C) информация о температуре в градусах по Фаренгейту (o F)! Это очень не удобно для меня, я понимаю температуру по Цельсию, не понимаю температуру по Фаренгейту. В Приложении "Умный дом" с Алисой данные с термометра транслируются в Цельсиях. 
Как можно изменить на экране LYWSD03MMC с oF на oC?

raspberryswirl, 2025-02-23:
Hi, as I bricked one LYWSD03MMC - here is another way to "free" them - I only do this very brief and once :
connect them via phone to bluethooth and add them to mi.home
use then the tokenextractor.
after that use the credentials here
https://atc1441.github.io/TelinkFlasher.html
do connect, then login, then flash with ATC_Exploit.bin
`09:05:47: Connecting to: LYWSD03MMC
09:05:58: Detected Mi Thermometer
09:05:58: Fw_version: 2.1.1_0159
09:05:58: Connected
09:06:05: Received device infos are correct
09:06:05: Login successfull
09:06:12: File was selected, size: 69680 bytes
09:06:12: Count: 4355
09:06:17: Start DFU
09:06:53: Update done after 35.481 seconds
09:06:59: Disconnected.
09:15:24: File was selected, size: 69680 bytes
09:15:24: Count: 4355`

after that use:
https://pvvx.github.io/ATC_MiThermometer/TelinkOTA.html
do connect, you find allready an ATC - falsh it with the Z003MMC for the zigbee 
`9:10:08 AM: Connecting to: ATC_ACE6AF
9:10:13 AM: Connected.
9:10:14 AM: Start DFU
9:11:05 AM: Update done after 50.38 seconds
9:11:09 AM: Disconnected.`

then go to your zigbee portal and permit the join
after a minute you should see your thermo

HTH

pvvx, 2025-02-23:
>after that use the credentials here
https://atc1441.github.io/TelinkFlasher.html
do connect, then login, then flash with ATC_Exploit.bin

Note: FW **ATC_Exploit.bin** may not support flashing Zigbee firmware via BLE OTA.

xumxum, 2025-02-25:
[image]

[image]

I flashed the atc custom firmware but there is no more flash button to flash the zigbee..what am I missing? 
where is the flash button after the custom firmware was flashed?

xumxum, 2025-02-25:
I managed to connect from mobile & then the flash options appeared, screenshot was from laptop..not sure why it gets stuck there, even though i flashed the atc_050.bin from there...

pvvx, 2025-02-25:
You have a thermometer with firmware from [atc1441](https://github.com/atc1441/ATC_MiThermometer).
This firmware does not support ZigBee download.

xumxum, 2025-02-25:
nope, I had original and flashed yours! Also check the log, it says pvxx firmware !!

pvvx, 2025-02-25:
In some equipment variants, after flashing the BLE device, it is necessary to reset the explorer cache (you can use "incognito mode") or restart the Bluetooth adapter (restart the computer).

MartijnPol95, 2025-03-04:
I followed the guide, found my devices but am now stuck as the Do the Activation button doesn't appear for me so I cannot get the values. Any ideas? 

[image]

pvvx, 2025-03-04:
> Any ideas?

There are ideas. Read the first message https://github.com/pvvx/ATC_MiThermometer/issues/602

mastersf, 2025-03-18:
so the only way that this works is OTA with the app and token extractor? Can't get the physical flashing working, and I've spent almost an entire day on it

ozanpak7, 2025-03-25:
i think i change to zigbee my LYWSD03MMC. but now i ll try to come back BLE ver2.1.1_0159
my only way is usb-to-uart with dongle eyc. or do i have any other way to come back BLE
?

i've no zigbee gateway :(

pvvx, 2025-03-25:
> my only way is usb-to-uart with dongle eyc. or do i have any other way to come back BLE ?

It depends on what firmware you flashed.
BZdevice and ZBdevice has BLE OTA and Zigbee OTA.
ZigbeeTlc only has Zigbee OTA.

Gruzin0066, 2025-04-08:
Здравствуйте, у меня при подключении датчика все коды id и токены определяются, устройство прошивается промежуточной прошивкой ATC_ota_40000 но далее не получается прошить z03mmc. Может подскажете, что я делаю не так?

Hello, when I connect the sensor, all the id codes and tokens, the device is flashed with the intermediate firmware ATC_ota_40000, but then I can't flash z03mmc. Can you tell me what I'm doing wrong?

slo-ko, 2025-04-11:
I'm having a bit of a struggle with getting the bind keys etc. from MiHome with the extractor software/script, hence I'm unable to flash the device over OTA. Is there some other ways to fetch the Mi token and bind key, either from the MiHome app or some other method? 

Is it possible to flash the FW 2.1.1_0159 version with USB-UART direclty without the bind keys etc, or are those required for that too?

javipt96, 2025-04-11:
> Tengo problemas para obtener las claves de acceso, etc., de MiHome con el software/script de extracción, por lo que no puedo flashear el dispositivo vía OTA. ¿Hay alguna otra forma de obtener el token Mi y la clave de acceso, ya sea desde la app de MiHome o con algún otro método?
> 
> ¿Es posible flashear la versión FW 2.1.1_0159 con USB-UART directamente sin las teclas de enlace, etc., o también se requieren para eso?

https://github.com/PiotrMachowski/Xiaomi-cloud-tokens-extractor?tab=readme-ov-file desde aqui puedes obtener la información que buscas

pvvx, 2025-04-11:
If the device is registered in MI-Home, the keys are written to Flash at address 0x78000 in open HEX format.
Without keys from MI-Home, via UART (USB-COM) you can always reflash these thermometers.

pvvx, 2025-04-11:
> Здравствуйте, у меня при подключении датчика все коды id и токены определяются, устройство прошивается промежуточной прошивкой ATC_ota_40000 но далее не получается прошить z03mmc. Может подскажете, [...]

Zigbee firmware size has become larger than BLE. Typical BLE OTA can no longer load Zigbee firmware. The probability of getting a brick is exactly 50%.

 Typical BLE OTA loads firmware into flash memory area at 0x00000 or 0x20000 and OTA size should not exceed 128 KB - otherwise it will overwrite the current running firmware. Result = a brick and a UART programmer.

The current running BLE firmware is at one of 0x00000 or 0x20000 address and loads OTA into free one. 
Zigbee firmware does not support such addressing. Zigbee uses addresses for OTA 0x00000 or 0x40000, or if it's Tuya with bootloader, then Zigbee OTA will eventually be moved to 0x8000 by Tuya bootloader.

### No other FW and third-party BLE OTA programmer, except [TelinkMiFlasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html), has special functions for loading current Zigbee firmware. Only [BLE firmware ATC_xxxx.bin](https://github.com/pvvx/ATC_MiThermometer/tree/master/bin) specified in [TelinkMiFlasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) can do this and only when loading the "*.zigbee" file, not "*.bin"!

pvvx, 2025-04-11:
> Parece que el problema es "Clear Ext.OTA region (129kib)..." se puede quedar ahi durante minutos...

If the cleaning failed the first time, then you need to repeat all the steps.
Before starting the OTA firmware, you can use the button to delete the entire history of records in TelinkMiFlasher. This can help, since it is difficult for a thermometer with a dead battery to clear half of the Flash for Zigbee OTA, while maintaining the BLE connection...

If the Flash was not cleared when writing via UART ("Erase All Flash" was not used), then problems may arise, since the device may have several partially working FW and the program may get confused about where there is a free place for OTA.

### With the release of new original FW versions, Xiaomi has even more dependencies with the addresses where OTA will be recorded. For these reasons, use only what is indicated in the previous message.

alexbarmin, 2025-04-18:
Привет!

Обновилcя по твоей инструкции до ATC_51.bin далее при обновлении на zigbee словил "Clear Ext.OTA region (129kib)..."

Решил попробовать через промежуточную прошивку по этой инструкции [https://github.com/devbis/z03mmc?tab=readme-ov-file#flashing-over-the-air-easy-way](https://github.com/devbis/z03mmc?tab=readme-ov-file#flashing-over-the-air-easy-way)

Все прошло без ошибок, дисплей отображает температуру/влажность, но zigbee hub (в моем случае Яндекс) не видит устройство. Подскажи в чем может быть проблема?

pvvx, 2025-04-18:
> Все прошло без ошибок, дисплей отображает температуру/влажность, но zigbee hub (в моем случае Яндекс) не видит устройство. Подскажи в чем может быть проблема?

Слишком мало информации и у меня нет "zigbee hub (Яндекс)". 
Обычно надо как-то включить "спаривание" или обнаружение новых устройств, а на термометре перед включением обнаружение (за несколько секунд) лучше замкнуть контакты сброса на 15 сек. При этом термометр минуту будет чаще искать Zigbee сеть. Потом это действие будет раз в 1..3 минуты - иначе батарея сразу сядет, т.к. поиск сети Zigbee требует много энергии...

joker-fu, 2025-04-28:
[image]

Greetings, young man. Assist me, please, for there appears to be an absence of firmware.

bearli97, 2025-05-01:
@pvvx I followed your perfect ZigBee installation guide, but I am stuck. 

**Custom Firmware: ATC_v51.bin** 
Flashed successfully. I can change config and everything. 

**Zigbee Firmware: 1141-020a-01273001-Z03MMC.zigbee**
After pushing the Start Flashing button:
Status: Waiting Start ext.OTA...

And that's it. Display goes into "---"-mode. But no flashing. 

```
22:14:17: Hardware Revision String: B1.5
22:14:17: Software Revision String: V5.1
22:14:17: Firmware Revision String: github.com/pvvx
22:14:17: Detected custom Firmware
22:14:18: Hardware Version: LYWSD03MMC B1.5, Software Version: 5.1
22:14:18: Custom config HEX string: 5507110a062804a931310ab4
22:14:32: Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/1141-020a-01273001-Z03MMC.zigbee'...
22:14:33: File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/1141-020a-01273001-Z03MMC.zigbee
22:14:33: Zigbee OTA Info: ManCode: 1141, ImageType: 020A, FwVer: 01273001
22:14:33: APP Release: 0.1 Build: 2.7, Stack Version: 3.0 Release: 0.1
22:14:33: File size: 131428 bytes
22:14:33: Count: 8215
22:14:34: Start DFU
22:14:36: Clear Ext.OTA region (129kib)...
22:14:36: Waiting Start ext.OTA...
```

pvvx, 2025-05-01:
>And that's it. Display goes into "---"-mode. But no flashing.

Try to execute the command "!Delete all records!" before performing OTA.

[image]

If it does not help, change the adapter.
On Apple, the Flash region clearing function for Zigbee often does not work. The reasons are that Apple does not comply with Bluetooth SiG standards. Switching to a different connection timing occurs, which may not work with adapters that do not comply with BLE standards. Cannot be corrected.

VicCoca, 2025-05-05:
I'm a bit stuck, what do I do here? I want to flash the zigbee firmware, I'm on ATC's custom firmware.
[image]

bearli97, 2025-05-06:
> If it does not help, change the adapter. 

That helped! Changed from MacBook to a different Windows laptop and it worked.
Thank you sir. :-)

nilsbyte, 2025-06-20:
@pvvx I am trying to flash my LYWSD03MMC, it came with firmware 2.1.1_0159. I hope you can clarify...

- I added the device to MiHome and retrieved all keys with the [Xiaomi Cloud Tokens Extractor] (https://github.com/PiotrMachowski/Xiaomi-cloud-tokens-extractor), [fixed version here] (https://github.com/PiotrMachowski/Xiaomi-cloud-tokens-extractor/issues/129#issuecomment-2985505332)
- I opened [TelinkMiFlasher] (https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html). 
- I entered ID, Token and Key and clicked Login > Success
- I flashed [ATC_v52.bin] (https://github.com/pvvx/ATC_MiThermometer/blob/master/bin/ATC_v52.bin) to my LYWSD03MMC
- I reconnected to the device, the ATC flash was successful (many settings).
- I deleted all records
- I restarted the device (remove and reinsert battery)
- I reconnected the device again
- I flashed the suggested Zigbee firmware: Zigbee `Firmware: 1141-020a-01283001-Z03MMC.zigbee`
- After the flash, the device was still available via Bluetooth and the ATC firmware is still there.
- I also tried flashing [db15-0203-79993001-Z03MMC_v0128.zigbee] (https://github.com/pvvx/ZigbeeTLc/blob/master/bin/devbis/db15-0203-79993001-Z03MMC_v0128.zigbee), no luck.

I have tried this on my Apple MacBook, on a Linux Mint installation on a PC, and with Windows 10. No luck.

[image]

device info:

```
22:50:21: Hardware Revision String: B1.4
22:50:21: Software Revision String: V5.2
22:50:21: Firmware Revision String: github.com/pvvx
22:50:21: Detected custom Firmware
22:50:21: Hardware Version: LYWSD03MMC B1.4, Software Version: 5.2
```

I don't know what I am doing wrong...

da0mao, 2025-06-21:
Same here, I also tried a different FW [here ](https://github.com/pvvx/ZigbeeTLc/tree/master/bin) (e.g., 1141-020a-01273001-Z03MMC) and still got no luck. And I run it in Chrome Incognito mode.

```
10:49:03 PM: Searching for devices
10:50:51 PM: Connecting to: ATC_CC6441
10:50:54 PM: Hardware Revision String: B1.5
10:50:54 PM: Software Revision String: V5.2
10:50:54 PM: Firmware Revision String: github.com/pvvx
10:50:54 PM: Detected custom Firmware
10:50:54 PM: Hardware Version: LYWSD03MMC B1.5, Software Version: 5.2
10:50:54 PM: Custom config HEX string: 5507100a062804a931310ab4
10:51:10 PM: File: 1141-020a-01273001-Z03MMC.zigbee
10:51:10 PM: Zigbee OTA Info: ManCode: 1141, ImageType: 020A, FwVer: 01273001
10:51:10 PM: APP Release: 0.1 Build: 2.7, Stack Version: 3.0 Release: 0.1
10:51:10 PM: File size: 131380 bytes
10:51:10 PM: Count: 8212
10:51:12 PM: Start DFU
10:51:13 PM: Clear Ext.OTA region (129kib)...
10:51:14 PM: Waiting Start ext.OTA...
10:51:21 PM: Clean Flash OTA region: 0x040000, size: 132 kbytes. Go OTA...
10:52:48 PM: Update done after 86.136 seconds
10:52:52 PM: Disconnected.
10:56:02 PM: Searching for devices
10:56:37 PM: Connecting to: ATC_CC6441
10:56:41 PM: Hardware Revision String: B1.5
10:56:41 PM: Software Revision String: V5.2
10:56:41 PM: Firmware Revision String: github.com/pvvx
10:56:41 PM: Detected custom Firmware
10:56:41 PM: Hardware Version: LYWSD03MMC B1.5, Software Version: 5.2
10:56:41 PM: Custom config HEX string: 5507100a062804a931310ab4
```

da0mao, 2025-06-21:
Same here, I also tried a different FW [here ](https://github.com/pvvx/ZigbeeTLc/tree/master/bin) (e.g., 1141-020a-01273001-Z03MMC) and still got no luck. And I run it in Chrome Incognito mode.

```
10:49:03 PM: Searching for devices
10:50:51 PM: Connecting to: ATC_CC6441
10:50:54 PM: Hardware Revision String: B1.5
10:50:54 PM: Software Revision String: V5.2
10:50:54 PM: Firmware Revision String: github.com/pvvx
10:50:54 PM: Detected custom Firmware
10:50:54 PM: Hardware Version: LYWSD03MMC B1.5, Software Version: 5.2
10:50:54 PM: Custom config HEX string: 5507100abb2804a931321ab4
10:51:10 PM: File: 1141-020a-01273001-Z03MMC.zigbee
10:51:10 PM: Zigbee OTA Info: ManCode: 1141, ImageType: 020A, FwVer: 01273001
10:51:10 PM: APP Release: 0.1 Build: 2.7, Stack Version: 3.0 Release: 0.1
10:51:10 PM: File size: 131380 bytes
10:51:10 PM: Count: 8212
10:51:12 PM: Start DFU
10:51:13 PM: Clear Ext.OTA region (129kib)...
10:51:14 PM: Waiting Start ext.OTA...
10:51:21 PM: Clean Flash OTA region: 0x040000, size: 132 kbytes. Go OTA...
10:52:48 PM: Update done after 86.136 seconds
10:52:52 PM: Disconnected.
10:56:02 PM: Searching for devices
10:56:37 PM: Connecting to: ATC_CC6441
10:56:41 PM: Hardware Revision String: B1.5
10:56:41 PM: Software Revision String: V5.2
10:56:41 PM: Firmware Revision String: github.com/pvvx
10:56:41 PM: Detected custom Firmware
10:56:41 PM: Hardware Version: LYWSD03MMC B1.5, Software Version: 5.2
10:56:41 PM: Custom config HEX string: 5507100abb2804a931321ab4
```

pvvx, 2025-06-22:
> The same for me:
> 
> Hardware Revision String: B1.6 Software Revision String: 0159 Firmware Revision String: 2.1.1_0159
> 
> Flashed to 2.1.1_0159 and then V5.2, works in HA BLE but display is blank.

[New version Xiaomi LYWSD03MMC B1.6 with different LCD controller.](https://github.com/pvvx/ATC_MiThermometer/issues/664) ?

kayamuskas, 2025-06-22:
> [New version Xiaomi LYWSD03MMC B1.6 with different LCD controller.](https://github.com/pvvx/ATC_MiThermometer/issues/664) ?

yeah, looks like

Nozletriangle, 2025-06-26:
Hello, so I have a Hardware Version: LYWSD03MMC B1.5, I updated it via OTA (I tried several firmwares) the issue is that the thermometer works (BLE) but nothing is displayed on the screen, it is completely blank or very light, what should I do to solve this and make the screen work as expected?
thanks advanced.

pvvx, 2025-06-26:
>  (I tried several firmwares)

Try further if you can't describe what you flashed. The message "it doesn't work for me" doesn't mean anything. :)

apppie123, 2025-06-26:
> > > (I tried several firmwares)
> > 
> > 
> > Try further if you can't describe what you flashed. The message "it doesn't work for me" doesn't mean anything. :)
> 
> Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin Stock_fw_2.1.1_0159.bin ATC_Thermometer.bin (LAST VERSION)
> 
> As I said, the thermometer works (sends data) but the screen shows nothing, I tried updating it via OTA and TTL with the same result.

https://github.com/pvvx/ATC_MiThermometer/issues/664

Nozletriangle, 2025-06-26:
> > > > > > (I tried several firmwares)
> > > > > 
> > > > > 
> > > > > Try further if you can't describe what you flashed. The message "it doesn't work for me" doesn't mean anything. :)
> > > > 
> > > > 
> > > > Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin Stock_fw_2.1.1_0159.bin ATC_Thermometer.bin (LAST VERSION)
> > > > As I said, the thermometer works (sends data) but the screen shows nothing, I tried updating it via OTA and TTL with the same result.
> > > 
> > > 
> > > [#664](https://github.com/pvvx/ATC_MiThermometer/issues/664)
> > > Just read the exact post above you...
> > 
> > 
> > I don't have B1.6, as I indicated it is B1.5
> 
> B1.5 and B1.6 both have a UART chip for the display. Only difference is the temperature/humidity sensor. What is the production date of this B1.5?

TLW_V1.4 CHMLmini
202102
HW: B1.5

apppie123, 2025-06-26:
> TLW_V1.4 CHMLmini 202102 HW: B1.5

Production date on the back of the package it came with, not on the PCB.
B1.6 manufactured in 03-2025 also has the 202103 date coding on the PCB.

Nozletriangle, 2025-06-26:
> Production date on the back of the package it came with, not on the PCB. B1.6 manufactured in 03-2025 also has the 202103 date coding on the PCB.

02-2025 in the package

Nozletriangle, 2025-06-26:
> Yeah, so that makes it very likely they use the same new SPI display driver.
> Did you already try to flash the custom 5.3 firmware from the other issue?

so i just tried 5.3 beta and the screen came back again!

Dimu1234, 2025-06-28:
I could not login with token extractor.

Although the correct MI Home user & password given, it says

Logging in...
Invalid login or password.
Unable to log in.

Press ENTER to finish

Please help

eingeryn, 2025-07-04:
> I could not login with token extractor.
> 
> Although the correct MI Home user & password given, it says
> 
> Logging in... Invalid login or password. Unable to log in.
> 
> Press ENTER to finish
> 
> Please help

It has been solved few days ago. Use newest release.

Bjk8kds, 2025-07-17:
@pvvx 
I also couldn't activate MJWSD05MMC (en) via the website (both atc1441 or pvvx), it just showed the activation process (forgot exact text) in the log without any further progress.
I don't remember the default fw version but I remember seeing numbers (more or less) 2.0 and 05 in the log, probably still the same, sorry for not paying attention. Hw version 2.3.
Then I saw your warning for LYWSD03MMC, so I tried using the token_extractor application on Windows, and it worked perfectly. Now I'm using custom firmware.

[10 more replies]