pvvx/ATC_MiThermometer #606: (Zigbee firmware) MHO-C401N update via OTA Z2M

im0t0, 2025-01-24:
[image]

У меня есть 6 датчиков MHO-C401N, у всех версия железа 2021.03
Версия Z2M: 2.0.0-2

Все они прошивались в сентябре 2024 на ZigBee [тут](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html)
Все успешно прошились, работают, в т.ч. экран.

Часть из них в Z2M определяется так:

[image]

и они без проблем прошиваются через OTA на [1141-0208-01233001-ZMHOC401N](https://github.com/pvvx/ZigbeeTLc/raw/refs/heads/master/bin/1141-0208-01233001-ZMHOC401N.zigbee)

а часть так:

[image]

и вот они не прошиваются ни на [1141-0208-01233001-ZMHOC401N](https://github.com/pvvx/ZigbeeTLc/raw/refs/heads/master/bin/1141-0208-01233001-ZMHOC401N.zigbee )

[image]

ни на [1141-0201-01233001-ZMHOC401 ](https://github.com/pvvx/ZigbeeTLc/raw/refs/heads/master/bin/1141-0201-01233001-ZMHOC401.zigbee)
OTA просто не видит её.

[image]

На какую прошивку прошивать проблемные?

pvvx, 2025-01-24:
Z2M загружает прошивки с https://github.com/Koenkk/zigbee-OTA
Всё остальное конфигурируется для Zigbee OTA вручную, путем создания файлов по описанию для Z2M.

Для прошивки должны совпадать номера "Manufacturer Code", "Image Type", "Stack Version" в header файла "*.zigbee".
Для автоматического  Zigbee OTA номер "File Version" должен иметь номер больше, чем текущий.
Иначе будет возможна только принудительная Zigbee OTA - описывается по другому или с какими-то дополнительными флагами.

Вам нужно получить "Manufacturer Code", "Image Type", "Stack Version" текущей прошивки.
Как это сделать в Z2M я не знаю. Я не использую Z2M и уточнить не могу.

im0t0, 2025-01-24:
>Всё остальное конфигурируется для Zigbee OTA вручную, путем создания файлов по описанию для Z2M.

Это делаю, другие же датчики обновляются.

>Иначе будет возможна только принудительная Zigbee OTA - описывается по другому или с какими-то дополнительными флагами.

Да, это делается так (в файле my_index.json):

`[
    {
        "url": "1141-0201-01233001-ZMHOC401.zigbee",
        "force": true
    }
]`

> Вам нужно получить "Manufacturer Code", "Image Type", "Stack Version" текущей прошивки. Как это сделать в Z2M я не знаю. Я не использую Z2M и уточнить не могу, т.к. описаний по этому у Z2M нет и очень путанный help по OTA.

А как это сделать не через Z2M? Какие есть еще способы?

@pvvx еще прошу вас, пожалуйста посмотрите вот [этот](https://github.com/pvvx/ATC_MiThermometer/issues/519#issuecomment-2611730575) комментарий по другому issue. Там про датчики с более старой версией железа.

pvvx, 2025-01-24:
Я не использую Z2M и не знаю, что там и какие есть обновления и скрипты для каких устройств.
[Тут я вижу только старые прошивки](https://github.com/Koenkk/zigbee-OTA/tree/master/images/Telink) всего на 2 типа устройства. Наверно только к ним и что-то было написано в Z2M.

im0t0, 2025-01-24:
Как переименовывание может повлиять на картинку в Z2M? Ответ: Никак. 
Ок, я понял, зря я написал про тот комментарий. Про него забыли, игнорируем.

Прошивки [отсюда](https://github.com/Koenkk/zigbee-OTA/tree/master/images/Telink) не причем, я беру [ваши](https://github.com/pvvx/ZigbeeTLc/tree/refs/heads/master/bin) и подсовываю вручную. Этот issue об этом. Часть устройств с версией железа 2021.03, без проблем видит их, и обновляется, часть с тем же железом их не видит. Предположим:

> Вам нужно получить "Manufacturer Code", "Image Type", "Stack Version" текущей прошивки. Как это сделать в Z2M я не знаю. Я не использую Z2M и уточнить не могу, т.к. описаний по этому у Z2M нет и очень путанный help по OTA.

[Тут](https://github.com/pvvx/ZigbeeTLc?tab=readme-ov-file#zigbee-ota) вы пишите об этом же:

> Before flashing the firmware, check the compliance of the “Manufacturer Code” and “Image Type” between the OTA file and those issued by the Zigbee device.

Но как это можно сделать? Как получить "Manufacturer Code", "Image Type", "Stack Version" текущей прошивки?
Только в ZHA?

pvvx, 2025-01-24:
> Только в ZHA? 

В ZHA так же - никак. Но если сильно покапаться в логах при подсовывании прошивки или старта обновлений, то можно с трудом найти среди тысяч строк лога...
Ни Z2M, ни ZHA не хотят делать Zigbee OTA дружественным для пользователей.
Делают только наоборот - усложняют, требуют вписывать [всякие соглашения](https://github.com/pvvx/ZigbeeTLc/issues/7#issue-2000695703): 
```
      allow_advanced_ota_dir: I understand I can *destroy* my devices by enabling OTA
        updates from files. Some OTA updates can be mistakenly applied to the
        wrong device, breaking it. I am consciously using this at my own risk.
```

pvvx, 2025-01-24:
"force": true - не пойдет. У работающей прошивки другие ранее указанные ID.
В инет есть скрипты меняющие ID в заголовке файла ZigbeeOTA.

В данном репозитории вы не получите нормального ответа, т.к. это репо про BLE устройства.

im0t0, 2025-01-24:
Решил проблему так.

Прошил через Z2M local OTA BLE прошивку [1141-0201-99993001-MHO_C401_v50.zigbee](https://github.com/pvvx/ATC_MiThermometer/raw/refs/heads/master/zigbee_ota/1141-0201-99993001-MHO_C401_v50.zigbee)

Потом на портале прошил Zigbee [1141-0208-01233001-ZMHOC401N.zigbee](https://github.com/pvvx/ZigbeeTLc/raw/refs/heads/master/bin/1141-0208-01233001-ZMHOC401N.zigbee)

При добавлении в Z2M немного ругнулся:

[image]

Но в итоге добавил. Правда всё равно определяется криво, но хоть прошивка последняя:

[image]

closed 2025-01-25 by im0t0

pvvx, 2025-01-28:
Вопрос к DeepSeek:  "_Как получить "Manufacturer Code", "Image Type", "Stack Version" устройства Zigbee в ZHA?_"

Ответ DeepSeek:

Чтобы получить Manufacturer Code, Image Type и Stack Version для устройства Zigbee в ZHA (Zigbee Home Automation), выполните следующие шаги:

1. Подключите устройство к ZHA:
Убедитесь, что ваше Zigbee-устройство подключено к ZHA через совместимый Zigbee-координатор (например, CC2652P, EFR32MG и т.д.).

Устройство должно быть успешно добавлено в сеть ZHA.

2. Используйте инструменты ZHA:
В интерфейсе ZHA (например, через Home Assistant) найдите ваше устройство в списке подключенных устройств.

Откройте страницу с подробной информацией о устройстве.

3. Просмотр информации о устройстве:
В разделе "Device Information" или "Zigbee Device Signature" вы найдете следующие данные:

Manufacturer Code: Уникальный идентификатор производителя устройства (обычно в формате HEX, например, 0x1234).

Image Type: Тип прошивки устройства (также в формате HEX).

Stack Version: Версия Zigbee-стека, используемого устройством.

4. Использование логов ZHA:
Если информация недоступна в интерфейсе, вы можете проверить логи ZHA:

Включите отладку для ZHA в Home Assistant:

```yaml
logger:
  default: info
  logs:
    homeassistant.components.zha: debug
    zigpy: debug
```
Перезагрузите Home Assistant и проверьте логи. В них будет подробная информация о устройстве, включая Manufacturer Code, Image Type и Stack Version.

5. Использование Zigpy:
Если вы знакомы с Python, вы можете использовать библиотеку zigpy для получения информации о устройстве программно:

```python
import zigpy.zcl
from zigpy.zcl.foundation import ZCLHeader

# Пример получения информации о устройстве
device = zha_device  # Ваше устройство
print(f"Manufacturer Code: {device.manufacturer_code}")
print(f"Image Type: {device.image_type}")
print(f"Stack Version: {device.stack_version}")
```
6. Пример вывода:
```
Manufacturer Code: 0x1234 (например, для Xiaomi Aqara).
Image Type: 0x1234 (зависит от прошивки устройства).
Stack Version: 0x1234 (версия Zigbee-стека).
```