pvvx/ATC_MiThermometer #612: [Documentation] LYWSD03MMC RS Mode options without external GPIO

wta12, 2025-02-06:
I wonder what exactly the RS Mode  "Connect" option in configuration setup - I flashed LYWSD03MMC B1.6 to v4.9

[image]
After **Set TRG** with Connect option as above, the RS Mode becomes **Switch** with *Invert RS event* checked
Send Trg data: 4434086419C9FFF4012C0111 -ok
Threshold Temp/Humi: 21.00°C/65.00%, Hysteresis T/H: -0.55°/5.00%, Reed switch mode: 17, Rs rep.interval: 300 sec, flg: 0x04, RS input inversion
[image]
**Status**: _Vbat: 3015 mV , Temp: 23.95°C, Humi: 57.81%, ID: 160, Switch: Close, Counter: 0, TRG Out: 0, flg: 0x04_ 
I aslo checked BTHOME in HA and see that _Power_ is logged as detected but Count and Opening(Switch??) stills 0 and closed respectively. How those data are reported to HA ? For MJWSD05MMC, the bottom button will activate **Opening** to closed and add one to **Count**. 
**Question**: how "Count/Opening/Power" change values in HA with different RS Mode (Counter/Switch/Connect) ??

[image]

Thanks,

pvvx, 2025-02-06:
I have the second button on the MJWSD05MMC set to "Switch" mode and use it to turn the floodlight on and off outside.
"Switch" is shown in the "Passive BLE monitor" integration
It is also "Opening" in the "BTHome" integration.
As for the confusion in the names, ask the author of these two integrations. This question has already been asked to the author, but there is no answer.

pvvx, 2025-02-06:
> From your link, I could not convert oF to oC (second bullet) by toggling upper (connect) button for a few seconds. 

This does not apply to the MJWSD05MMC, but applies to other thermometers with small screens.

pvvx, 2025-02-06:
> The "Passive BLE monitor" integration is being incorporated into HA core (HA Bluetooth has passive scan option now) so I want to move forward with BTHOME integration

The "Passive BLE monitor" integration has not been updated by the author for a long time to correct inaccuracies and for correct operation in the BTHome format. Also, many things are not supported there, which are declared in the description of the BTHome v2 format. This is reported in its README repository.

wta12, 2025-02-06:
> > From your link, I could not convert oF to oC (second bullet) by toggling upper (connect) button for a few seconds.
> 
> This does not apply to the MJWSD05MMC, but applies to other thermometers with small screens.
> 
> I don't know English and don't want to learn it. Part of the README translation was not written by me. And other parts were added via Google Translate. If you want to make a correct and complete description - send what needs to be changed to https://github.com/pvvx/ATC_MiThermometer/pulls.

I can read in russian well enough to see conflict between RU and EN comments but I am not a native speaker for both. Will update a more descriptive docs on trigger option once I fully understand the HA (receiving) workflow.

wta12, 2025-02-06:
> > The "Passive BLE monitor" integration is being incorporated into HA core (HA Bluetooth has passive scan option now) so I want to move forward with BTHOME integration
> 
> The "Passive BLE monitor" integration has not been updated by the [author](https://github.com/ernst79) for a long time to correct inaccuracies and for correct operation in the BTHome format. Also, many things are not supported there, which are declared in the description of the BTHome v2 format. This is reported in its repository - https://github.com/custom-components/ble_monitor?tab=readme-ov-file#important-announcement-about-the-future-of-ble-monitor .
> 
HA core has dedicated components for each BLE interfaces. [BTHOME ha_ble](https://github.com/Bluetooth-Devices/bthome-ble/tree/main) is the BLE parser module for BTHOME message  so as long adv message is formatted correctly according to BTHOMEv2 schema - I can make a pull request to address any bug from HA component side.
Can you point me to the test unit with a sample of adv message in BTHOMEv2 contains  Count/Switch/TRCout output as seen in FlashWeb so I can test with HA component first ? 
for example: "Vbat: 3095 mV , Temp: 23.40°C, Humi: 49.09%, ID: 42115, Switch: Open, Counter: 28, TRG Out: 0, flg: 0x25" 
> > Also Switch/Count in WebFlash - Status are not same as reported value in HA so I will check more on HA receiver side to see how BTHOME convert Adv message into data.
> 
> Inconsistencies or reception delays greater than 7ms after the "Switch" event and switch count (Open/Close) indicate gaps in reception by your BT adapter. Then the transmission of the event and count is duplicated several times with a period of 50..70 ms. Additional, duplicate sending of the “Open/Closed” states and their number is performed after the event and periodically at the time interval specified in the TelinkMiFlasher.html menu. But during BLE connection events are not processed. Only partially. This is due to bugs in the Telink BLE SDK - Enabling event processing during a BLE connection causes the entire program to crash spontaneously. That's why the processing during a BLE connection is very different.
I don't work at ms interval just want to get correct info parsed from adv message then let HA processes those data.

pvvx, 2025-02-07:
> I don't work at ms interval just want to get correct info parsed from adv message then let HA processes those data.

The initial BTHome format was developed by [ernst79](https://github.com/ernst79) together with me.
And I don't understand your question.

pvvx, 2025-02-07:
> [BTHOME ha_ble](https://github.com/Bluetooth-Devices/bthome-ble/tree/main) is the BLE parser module for BTHOME message so as long adv message is formatted correctly according to BTHOMEv2 schema - I can make a pull request to address any bug from HA component side.

This has been done a long time ago, but there are no answers.
The authors of the entire bluetooth chain in HA have a poor understanding of BLE. That's why they use the outdated Bluez API, which does not support BT5.0+. There is one option - to give them the entire full application code. But then the point is lost. It is easier to create your own integration.
Also, don't forget about compatibility with older versions for users. If combine the event names in the "BT Home" and "Passive BLE monitor" integration, users may have problems...

wta12, 2025-02-07:
> > I don't work at ms interval just want to get correct info parsed from adv message then let HA processes those data.
> 
> The initial BTHome format was developed by [ernst79](https://github.com/ernst79) together with me. As well as the algorithm for parsing the package structure for all variants accepted advertisements. And I don't understand your question.
I wonder if there is any sample of datastring in byte format similar to this [line](https://github.com/pvvx/ATC_MiThermometer/blob/70cc0caaf9e82ffeaac8264f13fc997059f3cfe0/python-interface/atc_mi_interface/atc_mi_format_test.py#L151) with Switch/Count/TRGout specified - I can log adv data myself but can't confirm my decoding is correct since you said WebFlash status are different than adv message ??

pvvx, 2025-02-07:
The BTHOME data structure for transmitting events depends on the version and functions of the device.
https://github.com/pvvx/ATC_MiThermometer/blob/master/src/bthome_beacon.h#L226-L235
https://github.com/pvvx/ATC_MiThermometer/blob/master/src/bthome_beacon.h#L255-L262
https://github.com/pvvx/THB2/blob/master/bthome_phy6222/source/bthome_beacon.h#L170-L178
https://github.com/pvvx/THB2/blob/master/bthome_phy6222/source/bthome_beacon.h#L202-L210

wta12, 2025-02-07:
> > Can you point me to the test unit with a sample of adv message in BTHOMEv2 contains Count/Switch/TRCout output as seen in FlashWeb so I can test with HA component first ?
> 
> What will this fix? I think you're looking in the wrong place. Events are handled differently in HA than other data. And it's not clear what you want to find? And why do you direct questions related to processing the BTHome format in Home Assistant integrations here? :)

I am using collecting data  in BTHOMEv2 format from thermostat  flashed with your awesome firmware for testing purpose. 
At the moment, I just want to correctly decode your adv message (with switch/count/TRGout info if available) myself then test if HA can transform it into entity state properly within my acceptable timeframe and efforts. Otherwise, I just record raw string then processing outside of HA. Btw, I am fully aware that HA stack, which is Python based, won't sustain at large scale and microsecond :P . 

About software dev stuff, I am not young but still loving tinkering. I don't plan any fancies but want to have some features which are useful for our own productivity so let see if it's worth to spend time or not. The (duplicated/wrong/mess-up) naming schema in HA is smth I want to address even if it's only for in-house dev purpose.

While I might not be at best position to comment about software dev, I am really appreciate opensource projects and contributors like yours. However, from my observation on software dev process  for both my works and hobbies , most of user don't care about 80% features except UI/UX so I am skeptical if BLE 5.0 can be speeded up. Currently, the home automation application just is not critical enough to use more responsive or longer battery devices so on HA unless we have more demanding scenarios of operation (scenarios etc..) - That's not a problem of software dev can solve.

pvvx, 2025-02-07:
The main reason why there is no point in changing anything in HA is that it uses the Bluez API which is 10 years out of date.

wta12, 2025-02-07:
> The BTHOME data structure for transmitting events depends on the version and functions of the device. https://github.com/pvvx/ATC_MiThermometer/blob/master/src/bthome_beacon.h#L226-L235 https://github.com/pvvx/THB2/blob/master/bthome_phy6222/source/bthome_beacon.c#L122 https://github.com/pvvx/ATC_MiThermometer/blob/master/src/bthome_beacon.h#L255-L262 https://github.com/pvvx/ATC_MiThermometer/blob/master/src/bthome_beacon.c#L133
> 
> https://github.com/pvvx/THB2/blob/master/bthome_phy6222/source/bthome_beacon.h#L170-L178 https://github.com/pvvx/THB2/blob/master/bthome_phy6222/source/bthome_beacon.h#L202-L210 https://github.com/pvvx/THB2/blob/master/bthome_phy6222/source/bthome_beacon.c#L122
> 
> Example of decoding the BTHome structure: https://github.com/pvvx/ATC_MiThermometer/blob/master/src/scanning.c#L86 However, HA cannot yet transmit a response to a request from a device. For example, to synchronize the clock or display user data on the screen without a BLE connection. This is yet to be implemented in future BTHome integrations when BT5.4 support is added... And this implementation works with BT4.2, replacing the "PAwR" function from BT5.4 on older hardware... :) That is, for lazy authors of BLE integrations, in addition to the functions already implemented in the firmware, I have already prepared several more options that they cannot yet implement. It will be years before this is implemented in the HA...

Thanks for detail reference. I am pretty sure a lot of in-house dev version of HA or its alternative will be beneficial from your development. Otherwise, popular stuff should be KISS(Keep it stup*d simple) regardless it's clever or ugly.

pvvx, 2025-02-07:
> About software dev stuff, I am not young but still loving tinkering.

That's the point. I'm also approaching the end of my sixth decade. That's why I recommend - leave it to the young. Examples for them for many years to come have already been created.