pvvx/ATC_MiThermometer #617: Questions and WiKi

milandzuris, 2025-02-18:
Hi i have some questions, there is a chance to create a WiKi?

What does: Low Power Mode, Tx Measures, BT5+ PHY, AdFlags.
What consumes less Battery: BTHome V2, Mijia etc.

Thank you.

pvvx, 2025-02-18:
Editing the [Wiki page](https://github.com/pvvx/ATC_MiThermometer/wiki) for this repository is available to anyone with an account on GitHub.com.

I don't speak English and I don't want to learn it.

milandzuris, 2025-02-18:
> Editing the [Wiki page](https://github.com/pvvx/ATC_MiThermometer/wiki) for this repository is available to anyone with an account on GitHub.com.
> 
> I don't speak English and I don't want to learn it.

Okay, but where do I get the information about what means what?

nepozs, 2025-02-19:
> Okay, but where do I get the information about what means what?

Just search interesting informations in existing issues (also in closed)
e.g. 
#596 #204 
and so on

BTW create Wiki articles only if you have enough knowledge…