pvvx/ATC_MiThermometer #618: Error during firmware assembly

butaikis, 2025-02-24:
Hello. 
1. I installed it according to the instructions:
- downloaded the IDE from [here](http://wiki.telink-semi.cn/wiki/IDE-and-Tools/IDE-for-TLSR8-Chips/)
- imported the **ATC_MiThermometer** project
- build, and error assembly

Last lines:
```
Building Standard target: out/ATC_Thermometer.elf
Building Reduced target: out/ATC_Thermometer.elf
 - Cannot openC:\TelinkSDK\opt\tc32\bin\tc32-elf-ld.exe: invalid hex number `-L'
make: *** [out/ATC_Thermometer.elf] Error 1
```

2. How to change the assembly in the firmware, for example, for **TH03Z** or **ZTH01** devices ?

pvvx, 2025-02-24:
https://github.com/pvvx/ATC_MiThermometer/blob/master/src/app_config.h#L62

```C
#define DEVICE_TYPE	DEVICE_TH03Z
```

```C
#define DEVICE_TYPE	DEVICE_ZTH01
```

or
 
```
make -s -j PROJECT_NAME=ZTH01%SWVER% POJECT_DEF="-DDEVICE_TYPE=DEVICE_ZTH01"
```
```
make -s -j PROJECT_NAME=TH03Z%SWVER% POJECT_DEF="-DDEVICE_TYPE=DEVICE_TH03Z"
```

butaikis, 2025-02-24:
@pvvx Thanks! How can I solve the build problem?

pvvx, 2025-02-24:
> How can I solve the build problem?

https://wiki.telink-semi.cn/wiki/IDE-and-Tools/IDE-for-TLSR8-Chips/

Set: "copy progect into workspace"

butaikis, 2025-02-24:
@pvvx 
That's what I did. Maybe your SDK path is different from mine. I have it installed on the way C:\TelinkSDK

Other warnings
[image]

pvvx, 2025-02-24:
Or without Eclipse - open `MakeAll.cmd` and change `PATH` in the first line.

butaikis, 2025-02-24:
@pvvx Yes, I've already noticed this file, changed the paths, and the project is ready, thank you. It's strange that Eclipse doesn't want to..

pvvx, 2025-02-24:
[image]

butaikis, 2025-02-24:
@pvvx 
I have other ways, but the right ones)

[image]

pvvx, 2025-02-24:
copy python.exe -> python3.exe
or edit makefile 
```
PYTHON ?= python
```
```
	@$(TC32_PATH)tc32-elf-ld --gc-sections -Ttext `$(PYTHON) $(PROJECT_PATH)/TlsrRetMemAddr.py -e $(ELF_FILE) -t $(TC32_PATH)tc32-elf-nm` -L $(TEL_PATH)/components/proj_lib -L $(OUT_PATH) -T $(LS_FLAGS) -o $(ELF_FILE) $(OBJS) $(USER_OBJS) $(LIBS)

```

butaikis, 2025-02-24:
@pvvx Great, it helped. Thanks!

closed 2025-02-25 by butaikis