pvvx/ATC_MiThermometer #621: MJWSD05MMC(ch): Show WeekDay bug

ufoczek, 2025-02-27:
Since I bought the Chinese version, I uploaded version 0.51. I noticed that in TelinkFlasher, there is an option: "Show WeekDay." Regardless of whether this option is checked or not, it always shows the day of the week. As I have an LCD with a Chinese display, I can't change the display to American, I just wanted to turn it off.

`Status: Hardware Version: MJWSD05MMC(ch) V2.3, Software Version: 5.1`

Show WeekDay: checked:
`New custom config: [143, 1, 10, 6, 80, 4, 169, 49, 0, 9, 90]
TelinkMiFlasher.html:1903 Send: 558F010A065004A93100095A
TelinkMiFlasher.html:1903 Status: Hardware Version: MJWSD05MMC(ch) V2.3, Software Version: 5.1
TelinkMiFlasher.html:1907 Hardware Version: MJWSD05MMC(ch) V2.3, Software Version: 5.1
TelinkMiFlasher.html:1907 Custom config HEX string: 558f010a065004a93100095a00`

Show WeekDay: unchecked:
`New custom config: [143, 1, 138, 6, 80, 4, 169, 49, 0, 9, 90]
TelinkMiFlasher.html:1903 Send: 558F018A065004A93100095A
TelinkMiFlasher.html:1903 Status: Hardware Version: MJWSD05MMC(ch) V2.3, Software Version: 5.1
TelinkMiFlasher.html:1907 Hardware Version: MJWSD05MMC(ch) V2.3, Software Version: 5.1
TelinkMiFlasher.html:1907 Custom config HEX string: 558f018a065004a93100095a00`

pvvx, 2025-02-27:
Updated, check it out.

ufoczek, 2025-02-27:
Thanks for quick response.

> 18:14:58: New custom config: [135, 1, 138, 6, 80, 4, 169, 49, 0, 9, 90]
> 18:14:58: Settings 5587018A065004A93100095A was sent successfully
> 18:14:59: Hardware Version: MJWSD05MMC(ch) V2.3, Software Version: 5.1
> 18:14:59: Custom config HEX string: 5587018a065004a93100095a00

The day of the week is still visible. No changes from the previous version.
If you need any additional data, debug messages etc., let me know.

ufoczek, 2025-02-27:
After uploading the updated version 51, it works fine.

ufoczek, 2025-02-27:
After resetting the thermometer (by pressing two buttons), the date shows up again. Can the value be stored so the date doesn't show after a reset? Thanks.

pvvx, 2025-02-27:
No. Reset is a complete reset of all settings to default.

Build a custom personal version from source, replacing the default settings.

closed 2025-02-27 by ufoczek