pvvx/ATC_MiThermometer #627: Unknown Qingping Firmware

dedelastuce, 2025-03-19:
Hi all, I have few Qingping thermometer & humidity sensors (model CGDK2)
I have flashed some of them but 1 doesn't want

When I try to connect via my windows PC (bluetooth) I have this message and I cannot flash the firmware nothing else appears on the screen 

> 00:04:24: Searching for devices
> 00:04:36: Connecting to: Qingping Temp & RH Lite
> 00:04:40: Hardware Revision String: 0001
> 00:04:40: Firmware Revision String: 1.0.2
> 00:04:40: Unknown Qingping Firmware

I have already tried to factory reset the sensor, to change the battery for a new one but the issue is the same

Any one has an idea of what is the matter wit this one?

Thanks

pvvx, 2025-03-20:
Your model name is different from other CGDK2.
CGDK2 = "Qingping Temp RH Lite" 
"Qingping Temp & RH Lite" = ???

dedelastuce, 2025-03-20:
Thx works like a charm
You are wonderfull

closed 2025-03-20 by dedelastuce