pvvx/ATC_MiThermometer #629: Adding the ability to send a timestamp

cRazyK67, 2025-03-29:
I have several LYWSD03MMCS stitched with your firmware and connected to the home assistant via BTHome integration.

Like many users, I experience problems with the wrong passage of time on my devices.

Synchronize time using TelinkMiFlasher.html it's very inconvenient, so I want to adjust the time on the devices using HA and ESP32.

The BT Home v2 protocol allows you to transmit a timestamp (Object id 0x50 timestamp uint48 (4 byte)) in BLU advertising, among other things. 
Could you please add settings to the firmware for transmitting tags along with transmitting other events?

pvvx, 2025-04-16:
### Синхронизация времени по BTHome “TimeStamp”.

Включение передачи термометром “TimeStamp” предполагается только в варианте, описанном далее:
1.	Прошивка собирается с опцией “DEV_SERVICES | SERVICE_SCANTIM”.
2.	В TelinkMiFlasher будет дополнено  меню с настройкой интервала запроса установки  времени и MAC адреса передающего устройства  (Ещё не реализовано  в текущей версии TelinkMiFlasher) 

### Принцип работы:

По истечению установленного интервала в секундах (от 1 часа до столетия)  термометр передает маяк с  “TimeStamp”. 
Сразу  после передачи маяка  включается сканирование на 9 ms. 
Если за это время (9 ms) не прият ответ от внешнего устройства с назначенным MAC и  новым “TimeStamp”, тогда  через 500 ms производится повтор цикла до 3 раз. 
Если ответ принят, тогда синхронизация  “TimeStamp” завершается. 
Всего производится 3 итерации синхронизации. 
Внешнее приемо-передающее устройство, после приема маяка с “TimeStamp” от термометра  должно в течении до 2 ms начать передавать маяк с  текущим “TimeStamp”.

---
Google translate:

### Time synchronization by BTHome “TimeStamp”.

Enabling the transmission of “TimeStamp” by the thermometer is assumed only in the variant described below:
1. The firmware is built with the “DEV_SERVICES | SERVICE_SCANTIM” option.
2. In TelinkMiFlasher, a menu will be supplemented with settings for the interval for requesting the time setting and the MAC address of the transmitting device (Not yet implemented in the current version of TelinkMiFlasher)

### Operating principle:

After the set interval in seconds (from 1 hour to a century) the thermometer transmits a beacon with “TimeStamp”.

Immediately after transmitting the beacon, scanning is enabled for 9 ms.

If during this time (9 ms) a response from an external device with an assigned MAC and a new “TimeStamp” is not received, then after 500 ms the cycle is repeated up to 3 times.

If the response is received, then the “TimeStamp” synchronization is completed.

A total of 3 synchronization iterations are performed.
The external transceiver, after receiving a beacon with the “TimeStamp” from the thermometer, must start transmitting a beacon with the current “TimeStamp” within 2 ms.

pvvx, 2025-04-16:
In the currently released source codes, this option is only provided in the debug version.
Not released into production, as it will be later converted to the BT5.4 standard with the "PAwR" function.
ESP32 supports only the outdated Bluetooth standard version 4.2! Also, ESP32 has a lot of reception gaps when working with BLE together with WiFi. Therefore, work with ESP32 in future versions of thermometers is not provided.

pvvx, 2025-04-16:
1.	У пользователей в наличии ещё множество полностью устаревших BT адаптеров и чипов поддерживающих только древний стандарт BT4.2 (ему более 10 лет).
2.	Затраты энергии от батареи  на создание BLE соединения у чипов ESP32-xx и прочих, не поддерживающих новые стандарты с функцией “быстрого соединения”  очень большие.
3.    Аналогично и Linux на сегодня  поддерживает только полностью устаревший BT4.2.

По этим причинам приходится создавать дополнительный протокол, замещающий “PAwR” для BT4.2.
[3 рекламы с 9ms сканированием](https://github.com/pvvx/ATC_MiThermometer/issues/629#issuecomment-2810525754) раз в сутки для синхронизации времени создают значительно меньшее увеличение потребления, чем одно соединение на хорошем BT адаптере не имеющем функции “быстрого соединения” из последних стандартов Bluetooth SiG (ver 5.4..6.0).
Описанный вариант возможен для реализации на типовом адаптере USB-BT с описанием на скрипте Python, а так же на любом современном SoC с BLE (исключая ESP32 и ESP32-xxx с работой в BLE и WiFi одновременно). 

---

Google translate:

1. Users still have many completely outdated BT adapters and chips that support only the ancient BT4.2 standard (it is more than 10 years old).
2. The battery power consumption for creating a BLE connection for ESP32-xx chips and others that do not support new standards with the “fast connection” function is very high.
3. Similarly, Linux today only supports the completely outdated BT4.2.
 
For these reasons, it is necessary to create an additional protocol replacing “PAwR” for BT4.2.

[3 advertisements with 9ms scanning](https://github.com/pvvx/ATC_MiThermometer/issues/629#issuecomment-2810525754) once a day for time synchronization create a significantly smaller increase in consumption than one connection on a good BT adapter that does not have the “fast connection” function from the latest Bluetooth SiG standards (ver 5.4..6.0).
The described option can be implemented on a typical USB-BT adapter with a description in a Python script, as well as on any modern SoC with BLE (excluding ESP32 and ESP32-xxx with operation in BLE and WiFi simultaneously).