pvvx/ATC_MiThermometer #631: 3x LYWSD03MMC dead after flashing Zigbee firmware

CZonin, 2025-04-02:
I've previously flashed about 10 LYWSD03MMC with the ZigbeeTLc firmware via the Telink Flasher without any issues. I just tried flashing 3 other LYWSD03MMC that I had lying around but each displayed "---" indefinitely after flashing. I eventually tried removing the battery and putting it back in but after that they are just blank. Any idea what's going wrong?

pvvx, 2025-04-02:
It is not specified with what type and version of firmware you used OTA for ZigbeeTLc.

CZonin, 2025-04-02:
> It is not specified with what type and version of firmware you used OTA for ZigbeeTLc.

I used what Telink pulls which seems to be `https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/Z03MMC_v0125.bin`

pvvx, 2025-04-02:
1.It is not specified what type and version of firmware was used before performing the OTA with ZigbeeTLc. Not an OTA file, but what worked in the device before the OTA ZigbeeTLc firmware.
2. HW version LYWSD03MMC is not specified.

CZonin, 2025-04-02:
> 1.It is not specified what type and version of firmware was used before performing the OTA with ZigbeeTLc. Not an OTA file, but what worked in the device before the OTA ZigbeeTLc firmware. 2. HW version LYWSD03MMC is not specified.

They are all the original hardware, not the revision. One of the sensors was on the stock firmware and the other 2 were on an older version of your custom firmware. I'm not exactly sure which version though.

pvvx, 2025-04-02:
New ZigbeeTLc version can't be flashed with simple BLE OTA. Extended BLE OTA is required.
Original firmware and very old BLE firmwares don't support ext.OTA.

CZonin, 2025-04-03:
Ah I didn't realize that. So I should flash the latest version of your firmware first before the ZigbeeTLc firmware?

pvvx, 2025-04-03:
Yes.  https://github.com/pvvx/ATC_MiThermometer/issues/602#issuecomment-2590913442

CZonin, 2025-04-03:
Makes sense, thanks for the help @pvvx 

A bit off topic, but what's the best way to update my other sensors with an older version of the ZigbeeTLc firmware (3001-0123)? I don't believe OTA in Z2M works.

pvvx, 2025-04-03:
Zigbee OTA in Z2M works well.

CZonin, 2025-04-03:
Hmm, when I check for updates it says that there are no updates available.

CZonin, 2025-04-03:
Didn't know that. Thanks again for all the help @pvvx !

closed 2025-04-03 by CZonin