pvvx/ATC_MiThermometer #632: LYWSD03MMC read out the Red Switch state

Chilliriha, 2025-04-06:
Hello,
Can you tell me some  information about how to read the Red Conact status or the Counter status in Home Assistant? In my integration I am using Custom Component Passive Ble. It works, but nowhere I found information on how to do it. I have flashed the LYWSD03MMC and I would like to connect the Impuls counter to read out water consumption.
Best regard Jiri

pvvx, 2025-04-07:
1.  https://github.com/pvvx/ATC_MiThermometer?tab=readme-ov-file#reed-switch-or-button-functions
 
2. https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html

[image]

[image]

3. Home Assistant

 [image]

Chilliriha, 2025-04-07:
Thank you for qiuck reply. 
All you have described is clear. I just don't know how I get Sensor "Counter" in my yaml file. The only types of sensor I found these:
sensor:
  - platform: atc_mithermometer
    mac_address: xx : xx : xx : xx 
    temperature:
      name: "ATC Temperature"
    humidity:
      name: "ATC Humidity"
    battery_level:
      name: "ATC Battery-Level"
    battery_voltage:
      name: "ATC Battery-Voltage"
    signal_strength:
      name: "ATC Signal"

pvvx, 2025-04-07:
[image]

---

[image]

---

[image]

pvvx, 2025-04-07:
How does working with Home Assistant relate to thermometer firmware?
Why don't you write questions in forums and other resources for Home Assistant?

Chilliriha, 2025-04-08:
I just wanted to read out a certain byte recieved form the thermometer and use to show its value

ngxson, 2025-10-24:
In esphome, you can use this snippet to receive the counter and switch data. Remember to set beacon format to ATC

```yaml

esp32_ble_tracker:
  scan_parameters:
    active: false  # can be true or false depending on your needs
  on_ble_advertise:
    then:
      - lambda: |-
          using namespace esphome::esp32_ble;
          using namespace esphome::esp32_ble_tracker;

          std::string target = "XX:XX:XX:XX:XX:XX"; // replace with your device's MAC address
          auto e_counter = id(my_counter);
          auto e_switch = id(my_reed_sw);

          if (x.address_str() != target) return;

          const BLEScanResult & res = x.get_scan_result();
          if (res.adv_data_len > 0) {
            const uint8_t * ble_adv = res.ble_adv;
            const size_t adv_data_len = res.adv_data_len;
            if (adv_data_len != 15) return;

            const bool valid = ble_adv[0] == 0x02
                            && ble_adv[1] == 0x01
                            && ble_adv[2] == 0x06
                            && ble_adv[4] == 0x16
                            && ble_adv[10] == 0xEB;
            unsigned int cnt_val = ble_adv[14] | (ble_adv[13] << 8) | (ble_adv[12] << 16);
            e_counter->make_call().set_value((float)cnt_val).perform();

            bool switch_val = ble_adv[7] == 0x0F;
            e_switch->publish_state(switch_val);
          }

number:
  - platform: template
    id: my_counter
    name: My Counter
    optimistic: true
    min_value: 0
    max_value: 4294967295
    step: 1

binary_sensor:
  - platform: template
    id: my_reed_sw
    name: My Reed Switch

# the rest is the same as normal
sensor:
  - platform: atc_mithermometer
    mac_address: XX:XX:XX:XX:XX:XX
    temperature:
      name: "Kitchen Temperature"
    humidity:
      name: "Kitchen Humidity"
    battery_level:
      name: "Kitchen ATC Battery-Level"
```

ngxson, 2026-09-21:
@kermit943 I would not recommend though, as the chip will detect the switch as "contact" even with a very low voltage (high resistance), so it will be too sensitive. A proper rain sensor gives out exact voltage reading, not just on/off

@Ulf349 not sure, maybe you should add a small pull down resistor. Or easier, just connect via an optocoupler: PIR sensor --> optocoupler --> thermometer