pvvx/ATC_MiThermometer #634: LYWSD03MMC-z disconnected to Zigbee2MQTT coordinator after OTA update

xitters, 2025-04-07:
New firmaware via OTA:
Fecha del firmware
    20240904
Versión del firmware
    3001-0123
Yesterday I updated, and then I realized the device wasn't paired to any coordinator (it wasn't sending updates to HA). I ​​rebooted the device and paired it again. It was fixed for a while. Today I see that it's not paired to any coordinator (as seen in the image).

What can I do? It worked fine with the previous version of FW.
[image]

pvvx, 2025-04-07:
Install ver0.1.2.6.

xitters, 2025-04-07:
> Install ver0.1.2.6.

Hi, ty for your work! where is the ver0.1.2.6. ?... sorry

pvvx, 2025-04-07:
https://github.com/pvvx/ZigbeeTLc/tree/master/bin

yongliqi, 2025-04-08:
how to manually flash this version firmware? In zigbee2mqtt, creating an override index with index json, restarting, the device doesn't respond to ota check.

xitters, 2025-04-08:
> how to manually flash this version firmware? In zigbee2mqtt, creating an override index with index json, restarting, the device doesn't respond to ota check.

https://smarthomescene.com/guides/how-to-setup-zigbee2mqtt-local-ota-index/

holden72, 2025-04-14:
Same problem here. I have LYWSD03MMC-z device.
Which firmware should I have to download for it, to flash ver0.1.2.6?
from here: [https://github.com/pvvx/ZigbeeTLc/tree/master/bin](url)

pvvx, 2025-04-14:
https://github.com/pvvx/ZigbeeTLc/raw/refs/heads/master/bin/1141-020a-01263001-Z03MMC.zigbee
or
https://github.com/pvvx/ZigbeeTLc/blob/master/bin/index0126.json

```
[
    {
        "fileVersion": 19279873,
        "fileSize": 131474,
        "manufacturerCode": 4417,
        "imageType": 522,
        "sha512": "d4f829dc1bfd201e30ef1899b4ad4064adf94cfcbd2556c3bdd798408412394091b51a52bf2a5662f9e1bd8b9167b1f8b822ab0e96231d1944a73f9fa3c45b54",
        "url": "https://github.com/pvvx/ZigbeeTLc/raw/refs/heads/master/bin/1141-020a-01263001-Z03MMC.zigbee"
    }
]
```