pvvx/ATC_MiThermometer #636: OTA on original MJWSD06MMC version 2.1.1.0009

pvvx, 2025-04-10:
**OTA on original MJWSD06MMC version 2.1.1.0009.**

### The current version requires:

1. Register the thermometer in MiHome.
2. Using alternative software, get all the keys for this thermometer. Find the MiHome Token Extractor here:  https://github.com/PiotrMachowski/Xiaomi-cloud-tokens-extractor
3. Run TelinkMiFlasher.html and connect to the thermometer.
4. Enter "Mi Token", "Mi Bind Key" and select "Login".
5. Select the firmware "MJ6_vNN" and run OTA.

[image]

[image]

[image]

[image]

[image]

### Zigbee

1. Update the firmware to version "MJ6_vNN".
2. Run TelinkMiFlasher.html and connect to the thermometer named "MJ6_XXXXX".
3. Select the firmware "MJ6_vNN" and run OTA.

RobWilks, 2025-04-15:
Thanks for looking how to integrate the new Mijia thermometer with Home Assistant
I flashed my Mijia thermometer 3 ([MJWSD06MMC) using Telink following the instructions above
The device had Xiaomi firmware version 2.1.1.0009
New firmware bin/MJ6_v51.bin uploaded successfully
In Home Assistant BTHome" finds the new device

But . . . 

The entities are incorrect - see below:
MJ6 FC99 Opening
binary_sensor.mj6_fc99_opening
MJ6 FC99 Power
binary_sensor.mj6_fc99_power
MJ6 FC99 Voltage
sensor.mj6_fc99_voltage
Packet Id
sensor.mj6_fc99_packet_id
Signal Strength
sensor.mj6_fc99_signal_strength

And they do not update.  Any ways to fix the problem?

Thanks again for all your work on this

pvvx, 2025-04-15:
Try the following:
Update the BLE firmware. I rebuilt the firmware for the MJ6 and tested it on my MJ6s, and updated the files in the repository.
If it doesn't start, disconnect and reinsert the battery.
I don't see any other options.

RobWilks, 2025-04-16:
pvvx, 
I am much impressed by the speedy response
The rebuilt firmware (new version of MJ6_v51.bin] is working well - previously I could only receive the battery voltage data
What do you think the problem might have been? 
Thankyou
RobWilks

pvvx, 2025-04-16:
Perhaps the MJ6_v51.bin file was not updated. (After the first test version.)

zzzealed, 2026-09-23:
Having the "Not registered!" issue too with my MJWSD06MMC. \
Firmware version 2.1.1_0009.

Couldn't get login to work no matter what I tried. Got the keys with [Xiaomi-cloud-tokens-extractor](https://github.com/PiotrMachowski/Xiaomi-cloud-tokens-extractor).

siradlib, 2026-09-24:
Instructions given in the first post worked perfect for me! After flashing `MJ6_v59.bin` and setting the firmware to "BTHome v2", the device was discovered by Home Assistant automatically. Very happy so far- thank you for the excellent how to guide!