pvvx/ATC_MiThermometer #637: Battery percentage wild fluctuation

apppie123, 2025-04-13:
I can't reopen https://github.com/pvvx/ATC_MiThermometer/issues/591 but I wanted to add some info here.

I'm having a similar issue as described with multiple sensors running fw 4.9.

Here's a screenshot of one of the sensors:
[image]

In #591 it was discussed that this is due to low battery voltage, but there are also fluctuations when the battery is full:
[image]

Is it possible to for example take a longer average? Or record the battery level at another time so the sensor is not transmitting and measuring the battery voltage at the same time?

pvvx, 2025-04-13:
The cause of the oscillations (beats) during voltage averaging has not yet been established.
"Averaging" is performed by a filter over 512 measurements, but using a shortened algorithm.
This does not appear on all thermometers and devices and so far there is no way to track down the cause...

> I've been looking at balancing the battery consumption. I need relative fast updates so I set advertising interval to 8000ms. 

Different HW thermometers use different sensors. That is, without specifying the type of thermometer and its specific HW (or sensor type), I cannot specify what the delay will be. And also whether the "Sensor Low Power" option works for it or not. On the SHTv3 (SHTС3) sensor, the program (FW) will not let you turn on "Sensor Low Power", since it has its own, different from others and the only operating algorithm.

On some, the algorithm works as follows:

If the "Sensor Low Power" mode is set:
After counting the specified number of beacon intervals for measurements, a command is sent to the sensor - start measuring.
After the next interval, a beacon with the previous values ​​is transmitted. The data from the sensor is then read and the information is updated for transmission in the next beacon. (This data will only start to be transmitted from the next beacon.)
As a result, we have a delay of 2 beacon intervals.

If the "Sensor Low Power" mode is not set, then the delay will be 1 interval.

But there are also types of sensors for which other algorithms are used, because they work differently.

pvvx, 2025-04-13:
> I set advertising interval to 8000ms.

If the "Sensor Low Power" mode is set -> The delay in data transmission from the sensor will be at least 8*2 seconds.

apppie123, 2025-04-22:
Thank you for the extensive answer!
I have a mix of different versions of hardware and also different sensors, SHTC3 but also 4.

What I noticed after running about 20 of these for the past year is that they sometimes completely forget their settings when the battery is empty. I set them to ATC format and some other custom settings, so they stop working when this happens since the format is different.

I was thinking about automating checking and writing back the same settings to all the different sensors when this happens, but I'm not sure if this is a good idea, since you mention each hardware needs different settings.

pvvx, 2025-04-22:
> What I noticed after running about 20 of these for the past year is that they sometimes completely forget their settings when the battery is empty

This is the first time I've heard of this.
In testing and using hundreds of different devices for 4 years, this has never happened.
Look for what or who is resetting your settings.

> I was thinking about automating checking and writing back the same settings to all the different sensors when this happens, but I'm not sure if this is a good idea, since you mention each hardware needs different settings.

Settings for the same FW version are fully compatible. If the thermometer is with SHTV3, the "low consumption" mode will automatically turn off.
Upon receipt, all settings are checked in the thermometer.

apppie123, 2025-05-05:
> This is the first time I've heard of this.
> In testing and using hundreds of different devices for 4 years, this has never happened.
> The settings are saved in Flash.
> Look for what or who is resetting your settings.

Today it happened again to other two devices where the battery was empty.

Settings used: 55202000002805bf313c04b400
Hardware: LYWSD03MMC B1.5
One is running 4.8, the other is running 4.9

I have initally flashed them using a serial converter(not via web).

No device is connecting to these two sensors so nothing is resetting setting.
They are simply broadcasting in ATC1441 format until the battery is empty. Then on some occasions the settings seem lost.

Perhaps a glitch, modifying memory due to an unstable battery? These sensors are also writing history data I believe, so perhaps if the battery is bad they start to write at the wrong place due to a voltage glitch?

pvvx, 2025-05-07:
> Perhaps a glitch, modifying memory due to an unstable battery? These sensors are also writing history data I believe, so perhaps if the battery is bad they start to write at the wrong place due to a voltage glitch? 

If the Flash recording erased other addresses, then in 60% of cases the firmware would be lost, and not the saved settings.

apppie123, 2025-05-07:
Then I'm really wondering where this comes from. It happened on multiple devices in the past months all flashed with 4.8 and 4.9. It is slowly getting hard to believe it's just bad luck.

Since yours is 1.5 years old, perhaps has to do with the hardware? Do you have extra capacitors? Mine have unpopulated footprints. I got these devices end of last year.

> No device is connecting to these two sensors so nothing is resetting setting.
> 
> To prevent anyone from changing your settings from outside, enable the pin-code.
Yes, that's active.

pvvx, 2025-05-07:
The thermometer shown on the chart hasn't even been opened. I threw it into the attic through the window and it's lost somewhere there. I don't want to look for it and change the battery.

pvvx, 2025-05-07:
Here is the full lifespan of this thermometer, from the moment the battery is installed and it is thrown into the attic:

[image]

Just connected with him.
His last update was: _Hardware Version: LYWSD03MMC B1.9, Software Version: 4.8_

apppie123, 2025-05-07:
Yes, pincode is set.

OK, so yours is version B1.9.
These were all version B1.5( LYWSD03MMC B1.5, Software Version: 4.9 ).
They have the locked OTA firmware and so I flashed them with a serial cable(since your great fix wasn't ready at the time).

Now I'm trying to find out what the difference is between these revisions, but I can't find a B1.9 version here.

apppie123, 2025-06-02:
> Table HW version: https://github.com/pvvx/ATC_MiThermometer?tab=readme-ov-file#chipset-lywsd03mmc-hwb14
> 
> Different Display controllers and different sensor.
> 
> Does the PIN code also disappear? All settings records are saved in one Flash sector and processed by one function. Records are added, and if there was a power failure during recording, then the previous recorded settings will be read next time. When the sector is full, the settings are transferred to a new sector, deleting old duplicates. If there was a power failure during the transfer, the previous settings will be read in the next start. That is, nothing depends on the interruption of the recording. A failure can lead to old settings, but the recording during the connection cannot be interrupted, then the connection will be interrupted too...

Sorry, missed this edit.
It appears as if the device is completely reset to the default settings. I have to go to your web interface and send the configuration string again to set it up with the right settings. It happened again last week with another one.