pvvx/ATC_MiThermometer #638: UPS USB-C PD/QC 12V

pvvx, 2025-04-15:
# Блок бесперебойного питания для устройств требующих 5-9-12 В 5..10Вт от USB-C.
 
Применяются готовые модули с aliexpress c примерными параметрами:
•	Выход модуля DC-DC c USB-C c PD/QC3.0 на 5/9/12B 10 Bт, ток нагрузки 420..650 мА 12В (5..8 Вт) и КПД от 92.5% от АКБ в 18..21В
•	Потребление блока зарядяки от сети 220..240В до 16 Вт в режиме заряда АКБ  и питания блока (КПД от 82%)
•	Емкость АКБ  80 Вт (20В 4Aч).

### Блок схема устройства:
[image]

### Макет:

[image]

[image]

Корпус был напечатан из доработанного этого проекта.
https://makerworld.com/en/models/548389-makita-power-bank-box-creative-channel
В итоге такой:

[image]
В него вставляется клеммник https://aliexpress.ru/item/1005006097000119.html

## Использованные модули:

### Модуль DC-DC c USB-С QC/PD 5-9-12V:

[image]
 
От АКБ 18950 из 5x3 банок и работе на среднюю нагрузку 5..8 Вт 12В КПД составляет 92.5%, имеет холостой ход около 2 мА (у него работает светодиод, это много, но на фоне нагрузки это ерунда).
 
### Модуль DC-DC для питание BLE модуля TB-03F и INA3221 на 3.3В:

[image]
 
Выход обязательно шунтировать танталовым конденсатором от 22 мкФ! Иначе на выходе шум более 280 мВ p-p. 
От АКБ в 20.5В потребляет  менее 300 мкА вместе с подключенным модулем TB-03F имеющим среднее потребление менее 10 мкА по 3.3В и INA3221 со средним потреблением 480..600 мкА по 3.3В.
 
### Модуль измерителя  с INA3221:

[image] 

Светодиоды желательно отключить путем отпайки резисторов.
На блок схеме данный модуль нарисован как раздельные элементы.
 
 
### Модуль TB-03F:
 
[image]

Подключение:
GND и 3.3В в питание от модуля DC-DC на 3.3В.
I2C_SCL  - PC0
I2C_SDA - PC1
Выход управления зарядом - PA1
К PA7(SWS) может быть включена кнопка. Будет на 80 секунд включать малый интервал маяка для соединения, если в TelinkMiFlasher установлен интервал основного маяка на большое значение.
К PD3 и PD4 могут быть подключены дополнительные кнопки с контактом к GND. Будут передавать события нажатия и отпускания, или счетчика импульсов. Включение этих кнопок конфигурируется в "app_config.h" при сборке FW.
 
### АКБ и зарядное устройство из набора:

[image]
 
В наборе идет зарядка выдающая до 21В с ограничением по току примерно до 600 мА. КПД более 82%.
Все части возможно купить и по раздельности.
АКБ можно всегда выдернуть на ходу из данного UPS (ИБП) и  использовать в инструменте.
Или, если другой используемый АКБ в инструменте сел, можно заменить севшим для зарядки, а заряженным продолжить работу с инструментом...
 
---

### Использование с NanoPi-R5S с SSD/NWMe работающим c системой Home Assistant (с неё и показаны все замеры и графики - интеграция BTHome). 

[image]

Систему с HA (NanoPi-R5S c SSD) и 3-мя USB-BT адаптерами и одним USB-Zigbee включенными через USB switch держит на АКБ с 4.2Ач более 14 часов.

Измерение во время работы системы и заряда АКБ:

[image] 
 
Пороги включения заряда АКБ выставлены на включение при менее 20.5В, выключение при более 20.75 В.
 
При кратковременном отключении внешнего питания (менее 2-х секуд) в момент включения срабатывает подзарядка до пары секунд. Это связано с усреднением значений  у INA3221 и включенной цепь R2  и R4, включающей Q1 при отключении внешнего питания для минимизации потерь - шунтирует встроенный диод в Q1.
Можно исправить программно, но такое поведение делает всё правильно - немного подзаряжает АКБ после короткого разряда.
 
Измерение во время работы системы, заряд АКБ давно закончен (25 дней назад):

[image]
 
Пример разряда и заряда АКБ:

[image]
 
Разряд с показаний в -72.131Вт до -14.835Вт (разница 57.296 Вт) за 10.83 часа (5.289 Вт в час)  
Старт: 20.655В (263mA), стоп: 19.023В (285mA) 
И заряд до -72.139Вт (разница 57.304 Вт) за 8.733 часа  (6.879 Вт в час) 
Старт: 19.247В (-339mA), стоп: 20.8В (-332mA)  
Т.е. потери в разнице по энергии заряда-разряда у данного АКБ минимальны.
 
Длительность саморазряда до 20.50В с момента зарядки до 20.80В у испытуемого АКБ длится более 25 дней.
 

### Сборка проекта:

В файле "app_config.h" изменить:
#define DEVICE_TYPE  DEVICE_TB03F
В конфигурации для TB03F, после строки "#elif DEVICE_TYPE == DEVICE_TB03F" изменить:
#define USE_SENSOR_INA3221        1 
Остальные "USE_SENSOR_xxx" установить в 0.

pvvx, 2025-04-25:
## Вариант для 21V АКБ с разъемом (выводом) зарядки и дополненной платой балансировки.

[image]

* Для предотвращения перезаряда разбалансированного АКБ используется вывод зарядки на АКБ.
* Дополнено ключом балансировки, т.к. поставляемые зарядные устройства для таких АКБ не имеют режима балансировки.

Внутренняя схема таких АКБ отключает вход заряда при достижении на одной из банок более 4.2В. При этом остаются недозаряженные банки. Со временем такой дисбаланс усугубляется.

Данная модификация использует АКБ c добавленной платой балансировки. Такие платы обычно работают по принципу: если на банке более 4.2В, тогда она шунтируется  резистором в 62 Ом, создавая ток разряда банки в 68 мА. Если ток заряда АКБ превышает эти 68 мА, то возможен перезаряд этой банки и прочие последствия...

Для исключения любого перезаряда на таких АКБ необходимо использовать два типа заряда – заряд через разъем заряда и последующая балансировка с ограничением  напряжения до 21В и тока до 60 мА но уже на входные-выходные клеммы АКБ. Поставляемые зарядные устройства для таких АКБ не имеют режима балансировки.

Типичные схемы китайских АКБ используют чип CM1051-DS. Данный чип имеет пороги отключения входа заряда при перезаряде любой банки более 4.25В и повторным  включением при 4.15B. Итоговое напряжения на 5 банках будет соответственно: 21.25В и 20.75В. Платы балансировки срабатывают примерно аналогичны – подключают резистор разряда банки при 4.15..4.2В. В итоге порог отключения  ключа балансировки настраивается на напряжение не более 20.75В. При саморазряде АКБ до порога в 20.5..20.6В так-же использует включение ключа балансировки. 

Напряжения срабатывания обоих ключей - "заряда" и "балансировки" устанавливаются пользователем в TelinkMiFlasher.

pvvx, 2025-10-25:
У меня такие настройки для [этого варианта](https://github.com/pvvx/ATC_MiThermometer/issues/638#issuecomment-2830057712) :
[image]
TelinkMiFlasher сделан для термометров и отображение триггеров не менял.

По этому показания напряжений переключения в x100 мВ (одна целая единица = 100 мВ)

Temperature threshold: 205.60, Temperature hysteresis: -0.50 это ключ заряда:
включение при менее 20.56-0.05 = 20.51 В 
отключение при более 20.56+0.05 = 20.61 В
Включает ток заряда до напряжения 20.61. Далее работает ключ малого тока (менее 60 mA) для подзарядки и балансировки.  

Humidity threshold: 206.20, Humidity hysteresis: -0.80 это ключ слабого тока для балансировки:
включение при менее 20.62-0.08 = 20.54 В 
отключение при более 20.62+0.08 = 20.7 В
Он в основном обеспечивает подзарядку по мере саморазряда АКБ и балансировку.

[image]

dkbiz83, 2025-11-13:
Подскажите, пожалуйста, резисторы R1, R5, R6 по 100 Ом ставятся дополнительно (параллельно) к таким же резисторам, которые уже впаяны на INA3221? Это не искажает измерения? Спасибо!

pvvx, 2025-11-14:
> Подскажите, пожалуйста, резисторы R1, R5, R6 по 100 Ом ставятся дополнительно (параллельно) к таким же резисторам, которые уже впаяны на INA3221? Это не искажает измерения? Спасибо!

Нет, внешние не ставятся. Это показаны резисторы на модуле. При 0.1 Ом у INA3221 ограничение измерения до 0.8A.
Но есть установка значений калибровки и для других номиналов шунтов...