pvvx/ATC_MiThermometer #639: Just display humidity in the larger digits.

elkarto, 2025-04-16:
Not really an issue, more of a question or idea for future features. I don't know if this is possible with the current version, I haven't found a way.

Is there a way to display only the humidity using the large portion of the display?

I plan to use the devices in dry boxes for 3d printer filament. There are hygrometers with a larger display but without bluetooth/zigbee.

I would like to use these to be able to send the humidity value to home assistant, but the humidity display is too small to be visible clearly.

If only the humidity value could be displayed (without displaying temperature) it would be ideal.

pvvx, 2025-04-16:
There is no "%" symbol in the first line of the display.

closed 2025-04-16 by pvvx

elkarto, 2025-04-16:
No problem with the "%" symbol as it will only display humidity, so there will be no confusion. You can just lit the bottom symbol or not lit any symbol at all.

I'm going to look for information on that model you indicate, but judging by the pictures it looks too big, it wouldn't fit inside the boxes.

Thank you for the hint