pvvx/ATC_MiThermometer #642: MHO-C401 (MHO-C401N?) - Screen freeze after using newest BLE firmware

ahochel, 2025-04-23:
I bought one MHO-C401 from Aliexpress ~ two months ago. It was flashed to ZigBee firmware without any problem, and it works in Zigbee2MQTT. Recently, I bought another three from the same seller, and all of them are stuck after flashing to custom firmware.

Here is the log from Telink Flasher v10.1:
```
03:16:51: Searching for devices
03:17:04: Connecting to: MHO_5F28A0
03:17:05: Hardware Revision String: 2022
03:17:05: Software Revision String: V4.2
03:17:05: Firmware Revision String: github.com/pvvx
03:17:05: Detected custom Firmware
03:18:31: Disconnected.
03:21:28: Searching for devices
03:21:46: Connecting to: MHO_5F28A0
03:21:47: Hardware Revision String: 2022
03:21:47: Software Revision String: V4.2
03:21:47: Firmware Revision String: github.com/pvvx
03:21:47: Detected custom Firmware
03:21:47: Hardware Version: MHO-C401(2022) 2022, Software Version: 4.2
```

I've found some old issues where I read some firmware I can try. It works only on two firmwares:
- `Original_OTA_Xiaomi_MHO_C401_v1.0.0_0016.bin`
- `MHO_C401N_v42.bin`

After flashing with those firmwares, the screen goes back to work.

My questions:
1. What strange version of this device did I bought?
2. If it is working with custom `MHO_C401N_v42.bin`, why is it not working with any higher version (screen freeze)?
3. Will it work with any ZigBee firmware?

[image]

[image]

[image]

[image]

pvvx, 2025-04-23:
MHO-C401 were released in two types with different screens and E-Ink controllers.
The old official firmware from Xiaomi does not work on MHO-C401N.
Therefore, there are two firmware versions MHO-C401(old) and MHO-C401N(new).

ahochel, 2025-04-23:
So the box it came with does not contain this star(?) [image] icon, I don't remember the initial state of the screen, how it looked when it came to me, but it has "X" [image] icon in the battery so I'm assuming it's MHO-C401N. Also, only custom firmware that works on this one is `MHO_C401N_v42.bin`.

It's not a problem for me that it's detected wrong. I tried to upload firmware with [Telink OTA flasher](https://pvvx.github.io/ATC_MiThermometer/TelinkOTA.html). I even downloaded Telink Flasher and hardcoded locally in HTML to detect it as MHO-C401N.

@pvvx **My problem is that it only works with version 4.2 of firmware (`MHO_C401N_v42.bin`), any higher version causes the screen to freeze**.

pvvx, 2025-04-23:
MHO-С401N works stably with SW ver 5.1 output.
```
18:03:23: Searching for devices
18:04:28: Connecting to: MHO_614E04
18:04:38: Hardware Revision String: 2022
18:04:38: Software Revision String: V5.1
18:04:38: Firmware Revision String: github.com/pvvx
18:04:38: Detected custom Firmware
18:04:38: Hardware Version: MHO-C401(2022) 2022, Software Version: 5.1
18:04:38: Custom config HEX string: 5507100a065504a9316308aa
```