pvvx/ATC_MiThermometer #648: Battery voltage

andrea271988, 2025-05-14:
I have 4 thermometer with version 4.9 and i can read the battery voltage correctly. now i have a new thermometer with version 5.2 and from home assistant i read 12v. Also on telink flasher i read 12xxx mV. Any solution?

[image]

andrea271988, 2025-05-14:
Just downgraded to 5.1 and now it works, maybe something wrong in the beta firmware

closed 2025-05-14 by andrea271988

satmandu, 2025-05-16:
Maybe keep this open? As this is a regression from 5.1 to the 5.2 beta? (I also see this issue on the same hardware.)

[image]

pvvx, 2025-05-16:
Fixed...