pvvx/ATC_MiThermometer #649: BTHome sensor readings for Count, Opening, and Power?

satmandu, 2025-05-17:
Using the current 5.2 beta firmware with my LYWSD03MMC, with Home Assistant seeing the devices via the BTHome integration, I see `Count`, `Opening`, and `Power` sensor readings in Home Assistant, which I don't understand.

Is that a downstream issue with BTHome? Or are these readings being sent by the firmware (and if so, maybe they shouldn't be for this hardware?)

[image]

(Otherwise everything works great!)

satmandu, 2025-05-17:
I see that this is a partial duplicate of https://github.com/pvvx/ATC_MiThermometer/issues/566

Is there a way to turn off some of these sensor reading advertisements since they all take power to transmit?

pvvx, 2025-05-18:
Set "RS Mode": none
Set "RS report interval": 0

In BTHome format the value "power" will still be transmitted. It is 2 bytes.
In Custom format the value "power" will still be transmitted. It is 1 bit.
Not transmitted in other formats.

closed 2025-05-18 by satmandu

pvvx, 2025-05-18:
Transmission of 2 bytes over 3 channels (PHY 1Mbit) takes 3*2*8/1000000=0.000048 seconds.
These bytes are transmitted in the beacon every other time.
The default beacon interval is 2.5 seconds.
In a month, 30.5*24*60*60/2.5/2=527040 times will be transmitted.

The transmission time of these 2 bytes per month will be 527040*0.000048=25.29792 seconds.
Or 25.29792/60/60=0.0070272 hours per month.
With a transmission consumption of 8 mA, the battery will require 0.0070272*8=0.0562176 mAh per month.
Over the course of a year, this will require 0.0562176*12=0.6746112 mAh from the battery.