pvvx/ATC_MiThermometer #650: LYWSD03MMC-z can't connect to my network when Z2M working on channel 15

PaveLGIL, 2025-05-19:
Hello! When I'm trying to connect to my Z2M on channel 15, I'm getting:

[2025-05-19 12:19:41] error: 	zh:controller: Interview failed for '0xa4c138693a1b7749 with error 'Error: Interview failed because can not get node descriptor ('0xa4c138693a1b7749')'
[2025-05-19 12:19:41] error: 	z2m: Failed to interview '0xa4c138693a1b7749', device has not successfully been paired

When i'm using channel 11 it connect sometime but not everytime. It is importand for me to use channel 15 because channel 11 not stable in my network at all for all devices..

Could you please help me on it?

pvvx, 2025-05-19:
There are no channel restrictions in the firmware.

Reset before pairing. The "reset" and "gnd" contacts must be short-circuited for 15 seconds.

PaveLGIL, 2025-05-22:
Hmmm, it works. onnected, but after OTA update to new version data not send. But I think that it anatoher issue. Thank you)

pvvx, 2025-05-22:
After updating the firmware, it is advisable to perform a new binding.