pvvx/ATC_MiThermometer #660: OTA flashing TH03Z

macmpi, 2025-06-06:
Have a new [TH03Z](https://pvvx.github.io/TH03Z) that I would like to reflash to use it as BLE BTHome with Home Assistant.
I do not have (yet) Zigbee gateway: is it possible to OTA reflash with TelinkMiFlasher.html over bluetooth without Zigbee?
I do not manage to discover the device (Linux/Chrome with #enable-experimental-web-platform-features flag) in the connect dialog (can't tell which device is TH03Z in list). Had no issues with LYWSD03MMC devices...

Thanks for any tip.

pvvx, 2025-06-08:
> is it possible to OTA reflash with TelinkMiFlasher.html over bluetooth without Zigbee?

No. This device only works on Zigbee. A Zigbee gateway is required.

klim123123, 2025-06-22:
But is possible to flash using direct connect to chip,
This device support BLE firmware, including BTHome for Home Assistant.
You need to solder 4 cables and flash device. Look here https://pvvx.github.io/TH03Z/
Web Flasher: https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html
Download firmware TH03Z_v52.bin file from here https://github.com/pvvx/ATC_MiThermometer/tree/master/bin

I just flashed my TH03Z and it's advertising BTHome now.

P.S.
Hold RST to GND before you click flash, clich to flash and release in 5 seconds you have before flashing starts.

macmpi, 2025-06-22:
Thanks for the note and advises.
However I did buy a SLZB device in the meantime (other Zigbee peripherals in the plan), and I'm planning to try OTA via ZHA.

I'm wondering if TH03Z may be able to run both Zigbee and Bluetooth simultaneously (i.e [BZDevice](https://github.com/pvvx/BZdevice) firmware), if such firmware is deprecated (seems older) or up to date, and if such mode may have significant drawbacks (like power consumption or else)...

macmpi, 2025-06-22:
@pvvx I'm a bit confused: my [`TH03Z`](https://pvvx.github.io/TH03Z/) (sold as this and looks like this) seems identified on ZHA as `TS0201` by `_TZ3000_fllyghyj` (sw version:  `0x00000046`): `TS0201` seem quite different physically

Which file should I use then to move to BLE?
Initially thought about `1141-0216-99993001-TH03Z_v52.zigbee`, but it does not detects it...
Should it be one of (which?):
[1141-0211-99993001-TS0201S1_v52.zigbee](https://github.com/pvvx/ATC_MiThermometer/blob/master/zigbee_ota/1141-0211-99993001-TS0201S1_v52.zigbee)
[1141-0211-99993001-TS0201S2_v52.zigbee](https://github.com/pvvx/ATC_MiThermometer/blob/master/zigbee_ota/1141-0211-99993001-TS0201S2_v52.zigbee)
[1141-0211-99993001-TS0201_v52.zigbee](https://github.com/pvvx/ATC_MiThermometer/blob/master/zigbee_ota/1141-0211-99993001-TS0201_v52.zigbee)

Or is it a new reference to be added?

pvvx, 2025-06-23:
> [@pvvx](https://github.com/pvvx) I'm a bit confused: my [`TH03Z`](https://pvvx.github.io/TH03Z/) (sold as this and looks like this) seems identified on ZHA as `TS0201` by `_TZ3000_fllyghyj` (sw version: `0x00000046`): `TS0201` seem quite different physically

First, you need to determine what type of thermometer you have from the pictures of the PCB.
Tuya has many variants with the same circuitry but different firmware. And vice versa - different circuitry, but the names are the same.

macmpi, 2025-06-23:
Thanks for feedback.
> Need "Image Type" number. It is not in your description.

Where/how can I find it from ZHA? According to your site, It should be `0x0216`, but I can not find it on ZHA.

I did add this in `configuration.yaml` [for OTA configuration](https://github.com/zigpy/zigpy/wiki/OTA-Configuration):
```
zha:
  zigpy_config:
    ota:
      enabled: true
      otau_directory: /config/zigpy_ota
      # Download the index over HTTPS
      extra_providers:
        - type: z2m
          url: https://raw.githubusercontent.com/pvvx/ATC_MiThermometer/refs/heads/master/zigbee_ota/index_v52.json

```
But I'm still not offered upgrade option/menu.

macmpi, 2025-06-24:
I'm remote currently and can not crack open it (also a bit nervous to break it up when I'm back).
That's the [item I purchased](https://www.aliexpress.com/item/1005007986989586.html).
Finally [obtained this](https://github.com/user-attachments/files/20890702/TS0201__TZ3000_fllyghyj_c7489d37ba_01_02_scan_results.txt) from ZHA Tookit in case it brings further info.

macmpi, 2025-06-24:
> Try using:

Here is my `configuration.xml` (and also added file in directory): The update is not detected (restarted ZHA integration, version has been checked).
```
zha:
  zigpy_config:
    ota:
      enabled: true
      otau_directory: /config/zigpy_ota
      # Download the index over HTTPS
      extra_providers:
        - type: z2m
          url: https://raw.githubusercontent.com/pvvx/ATC_MiThermometer/refs/heads/master/zigbee_ota/ts0201_tuya2ble.json  
```

pvvx, 2025-06-24:
A full restart of Home Assistant is required to detect the update. And some time for the device to report its IDs.

macmpi, 2025-06-24:
Yes did that too...will let it go for a night long and see if anything pops-up tomorrow
Thanks again for your kind support

macmpi, 2025-06-28:
No Luck after several trials.
Would this picture help?
Thanks!
[image]

macmpi, 2025-06-28:
Success: now on BT telink flasher it says: TS0201 V1.0, Software Version: 5.3
As it is supposed to be TH03Z, would you recommend installing that specific BLE firmware over BT now?
Unsure what to trust in terms of identification: one on the box (TH03Z) or  TS0201

klim123123, 2025-06-28:
> Success: now on BT telink flasher it says: TS0201 V1.0, Software Version: 5.3 As it is supposed to be TH03Z, would you recommend installing that specific BLE firmware over BT now? Unsure what to trust in terms of identification: one on the box (TH03Z) or TS0201

Did you installed BT firmware via OTA using ZHA?

macmpi, 2025-06-28:
> Did you installed BT firmware via OTA using ZHA?

Yes.

macmpi, 2025-07-16:
Are you sure it is exact same device (see [picture](https://github.com/pvvx/ATC_MiThermometer/issues/660#issuecomment-3015238693))?
Which HA version are you using?
I had 2025.5.3 at the time (now 2025.7.1): original firmware version was displayed (0x00000046) along with name/model

I had to do a complete HA restart after applying `configuration.yaml`, and also used the pin to press device button once.
After a while was lucky to get the new update dialog.
Had to do several trials: did not work right-away.

PS: maybe I also had actual [firmware file](https://github.com/pvvx/ATC_MiThermometer/raw/master/zigbee_ota/1141-d3a3-01983001-TS0201BLE_v53.zigbee) within `/config/zigpy_ota` directory too: can't tell if ZHA used local file or it downloaded from `.json`. Tried so many things to get it done!...

klim123123, 2025-07-17:
@macmpi Thank you! I tried everything. Even re-installed Home Assistant. Nothing...
I ended up updating firmware soldering wires.

macmpi, 2025-07-17:
which firmware file did you use? by which model is it identified now?

klim123123, 2025-07-17:
I used TH03Z_v52.bin
Model TH03Z v1.0

macmpi, 2025-07-18:
Thanks.

@pvvx is actual BLE firmware code generic and model files just different  encapsulation of it with particular settings (model name, etc), or is firmware really different per model (to accommodate various sensors, screen support, etc)?

For [this particular device](https://github.com/pvvx/ATC_MiThermometer/issues/660#issuecomment-3015238693), which file is the most appropriate: `TS0201_vNN.bin` or `TH03Z_vNN.bin` ?
What's the potential issue to use one vs the other?

Wouldn't it make sense to add reference of this model (picture, manuf TS0201_TZ3000_fllyghyj, etc) in your models pages to help other potential users to properly identify it?

Thanks for your feedback.

pvvx, 2025-07-18:
Screens, screen driver chips and their protocols are different. Sensors, buttons, LEDs, screen chips, sensor switching and/or display drivers, as well as additional signals, various inputs and outputs have different ports (GPIO) for connection to the main chip.

macmpi, 2025-07-18:
Ok, thanks.
So for our model, which one is recommended `TS0201_vNN.bin` (which I use) or `TH03Z_vNN.bin` (@klim123123 uses)?
How both could equally work?

pvvx, 2025-07-18:
There are already over 40 currently supported modifications with differences in schemes and protocols.
And taking into account the 3 main software versions (BLE, ZigbeeTlc, ZBdevice) there are already more than 120. And if you also take into account the supported firmware versions, then the number of necessary tests requires about a thousand firmwares and tens of thousands of function tests.
For these reasons, debugging and finding errors is shifted to users. I only check a couple of basic options and not all functions.

pvvx, 2025-07-18:
According to the presented picture, the inscription on the sensor is not visible.

The firmware version for Th03z supports the following sensors:
```
#define USE_SENSOR_CHT8305		1
#define USE_SENSOR_CHT8215		0 // off
#define USE_SENSOR_AHT20_30		1
#define USE_SENSOR_SHT4X		1
#define USE_SENSOR_SHTC3		1
#define USE_SENSOR_SHT30		1
```

klim123123, 2025-07-18:
Turns out it’s a little different.

[image]

macmpi, 2025-07-19:
Ok thanks, fair enough.
Was lucky it worked with my model then.
To recap steps for anyone with same model:
- Zigby OTA update with `https://raw.githubusercontent.com/pvvx/ATC_MiThermometer/refs/heads/master/zigbee_ota/ts0201_tuya2ble.json` to transform into BLE (either through` .json` or direct file in directory and wake device after HA full restart is completed)
- Once on BLE, use telinkflasher page to re-flash again with `TH03Z_vNN.bin`

Thanks for all help in this journey.
Closing.

closed 2025-07-19 by macmpi