pvvx/ATC_MiThermometer #664: Display blank after flashing ATC_v52

ghbdtnvbh, 2025-06-12:
Hello! I recently purchased two LYWSD03MMC devices. On the box, the production date is 03.2025.

I first tried flashing `ATC_v52.bin` using the web flasher.  
After flashing, the screen froze. I removed and reinserted the battery — the device rebooted and showed up as LYWSD03MMC in Bluetooth scans.

To be cautious, I then restored the original Xiaomi firmware (as described in the documentation), and after that flashed `ATC_v52.bin` again.

Since then, the screen does not turn on anymore.  
The device is still connectable via Bluetooth and continues broadcasting data, but the display remains completely off regardless of the firmware version (including original and earlier ATC builds).

This behavior is identical on both devices.

Current status in the web flasher:

```
Status: Hardware Version: LYWSD03MMC B1.6, Software Version: 5.2
Vbat: 2939 mV
Temp: 24.39°C
Humi: 52.66%
ID: 54
Switch: Open
Counter: 0
TRG Out: 0
flg: 0x05
```
Could this be due to a  new hardware revision that is incompatible with the display driver used in the current firmware?  
Is there any workaround to restore display functionality?

Thank you!

Mettwurstfan, 2025-06-14:
I've got the exact same problem – the display stays blank after flashing.
Device production date: 03/2025

I flashed ATC_ota_40000.bin via UART with the following log:

```
08:43:11: USB-COM opened.
08:43:14: Reset DTR/RTS (100 ms)
08:43:14: Activate (5 sec)...
08:43:19: Write 82516 bytes in to Flash...
08:43:47: Done (33.073 sec).
08:43:47: Soft Reset MCU
```

After the flash, the display immediately went blank. However, the device is still fully reachable via Telink Flasher.

I tried the following firmware options:

Custom ATC Firmware v5.2

Stock Firmware 1.0.0_0130

Z03MMC Zigbee Firmware

Unfortunately, the screen stays off in all cases. I initially thought I might have damaged the unit while disassembling it – but the exact same issue happens with a second, identical unit, flashed the same way.

Let me know if I can help debugging or testing something. Thanks for your efforts!

pvvx, 2025-06-14:
Maybe this is a new version with a different display controller.
Do the existing original Xiaomi firmwares work with these thermometers?

apppie123, 2025-06-14:
I created a full dump of this firmware. Please find it on below link(I think it expires in 7 days).
https://www.transfernow.net/dl/20250614aP8PpIzR

apppie123, 2025-06-14:
[image]
[image]

[image]

Some more pictures. Pin 18 goes to PSCK. Pin 15 to P7

apppie123, 2025-06-14:
Pulse train on PSCK and PSDA.
In you code I see 700kHz for i2c but this is much slower: 24.46 Hz for one clock pulse.
[image]

pvvx, 2025-06-14:
> I created a full dump of this firmware. Please find it on below link(I think it expires in 7 days). https://www.transfernow.net/dl/20250614aP8PpIzR

Firmware CRC matches SDK.
Cut file for OTA: [new_b16_ota.zip](https://github.com/user-attachments/files/20739844/new_b16_ota.zip)

apppie123, 2025-06-14:
> > Decoding that, there seems to be an issue still(or maybe it is not actual i2c). Start, h55 [ h2A | RD ] NACK, h97 NACK, hF7 NACK, h7B, h5E, h9F NACK, h69, hD5, Error
> 
> This is SPI 8 bit.

That is what I figured indeed. I shall share some trace data later if needed. It is repeating 9 bytes and I haven't see a chip select trace pin. So probably it's not full spi but using a time delay instead of cs pin.

apppie123, 2025-06-15:
Above data was making little sense, my Digital Discovery wasn't correctly decoding the bits..
First message(init): 0x55 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0x00 0xAA
Second message(init): 0x55 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0x00 0xAA

Then every 5 seconds an update is sent to the display:
0x55 0xCB 0x7D 0x6D 0xA5 0x6D 0xF4 0xE7 0xAA
0x55 0xCB 0x7D 0x6D 0xA5 0x6D 0xBF 0xAC 0xAA
0x55 0xCB 0x7D 0x6F 0xA5 0x6D 0xBF 0xAE 0xAA

This seems to match with what you already have in your code @pvvx 
	// B1.6 (UART), lcd_i2c_addr = 0
	// utxb.dma_len = 0;
	// utxb.head = 0;
	utxb.start = 0xAA;
	// utxb.data = {0};
	utxb.end = 0x55;
	memset(display_buff, 0, sizeof(display_buff));
	lcd_send_uart(display_buff);

Start and end byte are matching for B1.6 hardware!

**However**, you are setting PD7 for UART TX:
// GPIO_PD7 set TX UART pin

However, I see the data on testpin P11, which is PB7 and PD7 is used for clock.

pvvx, 2025-06-15:
How to distinguish between UART and SPI versions in the firmware program?

> This looks very similar to what you already have, but instead of uart for B1.6 hardware there is an extra clock and start and end bytes are swapped.

It is quite possible that all the bits are simply inverted.

apppie123, 2025-06-15:
> How to distinguish between UART and SPI versions in the firmware program?

I don't think you can. There is no return line from the display IC and I don't think there are pull-ups internally in the LCD IC to sense if there are two lines connected.

How often can you expect to see the old uart hardware in the wild now? Perhaps a setting on the webpage is needed. I'll see if I can find some other differences on the PCB. 

> > This looks very similar to what you already have, but instead of uart for B1.6 hardware there is an extra clock and start and end bytes are swapped.
> 
> It is quite possible that all the bits are simply inverted.
I don't expect it, see my earlier message.
If I XOR the 6 data bytes I get a checksum byte that matches. But if I invert the data, then the checksum does not match.

Would you be able to write support for this LCD? I'd be happy to flash it and test it. Even just some simple code to see how it behaves. If more information is needed, please let me know.

Probably if you load the bin I shared on one of the other revision hardware you already have you can see the data going out for yourself.

I already downloaded this project, but I am not very familiar with this chip so it will take me a long time to get it going.

apppie123, 2025-06-15:
I found this picture of a B1.6 and the PCB, testpoints and silkscreen is exactly the same..

https://github.com/pvvx/ATC_MiThermometer/issues/125#issuecomment-934708558

pvvx, 2025-06-15:
[test_B16_new_01.zip](https://github.com/user-attachments/files/20746408/test_B16_new_01.zip)

**Minimum SPI Clock** Frequencies (!):
 SYS CLK: 16MHz -  SPI: 62.5 kHz
 SYS CLK: 24MHz -  SPI: 93.75 kHz
 SYS CLK: 32MHz -  SPI: 125 kHz
 SYS CLK: 48MHz -  SPI: 187.5 kHz

[image]

Transfer at startup, after power on:

[image]

Next transmissions:

[image]

apppie123, 2025-06-15:
> [test_B16_new_01.zip](https://github.com/user-attachments/files/20746408/test_B16_new_01.zip)
>
> **Minimum SPI Clock** Frequencies (!): SYS CLK: 16MHz - SPI: 62.5 kHz SYS CLK: 24MHz - SPI: 93.75 kHz [...]

Blank screen unfortunately. I tested two. Verified bluetooth still works.

apppie123, 2025-06-15:
> When you turn on and transmit "FFFFFF..." does the screen display all segments or is it blank?
Yes, for half a second it shows all segments.

pvvx, 2025-06-15:
> But your last init trace is different from the trace I shared before.

No initialization. Always outputs what is accepted. My firmware starts with a cleared display.

pvvx, 2025-06-15:
> I'm not getting any display with your test_B16_new_02 unfortunately.

The next test won't be available for at least an hour. I'm busy with other things right now...

pvvx, 2025-06-15:
Original FW:
Power Up:
[image]

[image]

[image]

CLK 10 kHz:
[image]

Frame transfer is 7.5 ms (!). This is very high. This will consume a lot of battery because a typical thermometer has a wake cycle with advertising transfer in 4.5 ms.

---

[test_B16_new_03.zip](https://github.com/user-attachments/files/20746976/test_B16_new_03.zip)

[image]

apppie123, 2025-06-15:
[image]

pvvx, 2025-06-15:
> Can't we try and up the SCK speed? See where the limit is?

It's easier to throw away such thermometers and forget about them. There are many with better quality display and 2-3 times cheaper.

apppie123, 2025-06-15:
> > Can't we try and up the SCK speed? See where the limit is?
> 
> It's easier to throw away such thermometers and forget about them. There are many with better quality display and 2-3 times cheaper.

This batch of LYWSD03MMC is the latest currently available on aliexpress/alibaba so that would mean it is the end of this sensor(like the non-recommended tuya). I got them super cheap for 3 euro each. It would be a waste.

I will map out the display. Hopefully we can try and improve somehow.

apppie123, 2025-06-15:
60010000000000 = 0.7
60020000000000 = 0.6
60040000000000 = 0.5
60080000000000 = 0.4
60100000000000 = 0.3
60200000000000 = 0.2
60400000000000 = 0.1
60800000000000 = 0.0

60000100000000 = 1.7
60000200000000 = 1.6
60000400000000 = 1.5
60000800000000 = 1.4
60001000000000 = 1.3
60002000000000 = 1.2
60004000000000 = 1.1
60008000000000 = 1.0

60000001000000 = 2.7
60000002000000 = 2.6
60000004000000 = 2.5
60000008000000 = 2.4
60000010000000 = ??
60000020000000 = 2.2
60000040000000 = 2.1
60000080000000 = 2.0

60000000010000 = 3.7
60000000020000 = 3.6
60000000040000 = 3.5
60000000080000 = 3.4
60000000100000 = ??
60000000200000 = 3.2
60000000400000 = 3.1
60000000800000 = 3.0

60000000000100 = 4.7
60000000000200 = 4.6
60000000000400 = 4.5
60000000000800 = 4.4
60000000001000 = 4.3
60000000002000 = 4.2
60000000004000 = 4.1
60000000008000 = 4.0

60000000000001 = 5.7
60000000000002 = 5.6
60000000000004 = 5.5
60000000000008 = 5.4
60000000000010 = 5.3
60000000000020 = 5.2
60000000000040 = 5.1
60000000000080 = 5.0

pvvx, 2025-06-15:
[test_B16_new_04.zip](https://github.com/user-attachments/files/20747252/test_B16_new_04.zip)

apppie123, 2025-06-15:
> [test_B16_new_04.zip](https://github.com/user-attachments/files/20747252/test_B16_new_04.zip)

Yes!!!

pvvx, 2025-06-15:
[test_B16_new_04_clk20kHz.zip](https://github.com/user-attachments/files/20747260/test_B16_new_04_clk20kHz.zip)

apppie123, 2025-06-15:
[image]

apppie123, 2025-06-15:
I don't have a microamp meter here so unfortunately can't help with that.

Would it make sense to try higher transfer rates? See where we can get it.

I am not sure how we could distinguish between old B1.6 and new B1.6 other than at initial flash or a config button in the web UI.

Another thing I noticed is that when you set display refresh to 8 seconds. It takes 8 seconds for the first time to show the info on the LCD(and so you are waiting at a blank screen to see if the battery works). Would it be possible to add a quick first time update right after init?

With your info I want to set the refresh high to save some energy there.

apppie123, 2025-06-15:
It would be great to give some signs. Like -- for temperature and humidity. I have other thermometers without bluetooth who also need some time to boot, but do exactly that. Quickly show -- and after a while you get a reading.

Just so you know that when you set the refresh to 16 seconds or something that you have some feedback that the thermometer is powered and don't sit in doubt if it is working(it always feels longer than 16 seconds :) ).

-- 

Do you want to test higher SPI rate to see if we can shorten the on time?

pvvx, 2025-06-15:
You do not take into account all the conditions.
If the capacitor is included in the power supply, then when the battery is turned on, the capacitor is charged and the voltage does not reach the operating voltage for a long time. In this case, the firmware goes into deep sleep for 3 minutes, while everything possible is disabled in order to achieve a current below 3 μA. This is not possible on all types, since it depends on the installed elements.
In the future, if the power supply has a normal capacitor, this increases the battery life by at least 40%. That is, not 1 year, but one and a half or more. And with an increase in the pause between beacons, the battery will last even longer ...

Therefore, not 16 seconds, but several minutes, for those who soldered the capacitor into the designated seat on the board. :P

pvvx, 2025-06-15:
An example of a battle between a B1.9 thermometer and a CR2032 battery at subzero temperatures in winter. Until 2025, the battery produced a voltage of more than 2.0 V. But then, at low temperatures, it could no longer. And constant shutdowns began. Then it got warmer and it continued to work.

[image]

This is data from a thermometer thrown into an unheated attic. Thrown in the literal sense - through the window. And I don't want to look for it there :)

pvvx, 2025-06-15:
It is impossible to foresee all the options.
Flash tends to give noise of some bits when reading at normal or increased supply voltage, if it was written at minimum. The cell does not have enough charge for the switching level when the supply voltage increases. And at reduced voltage, the switching level of "0" or "1" also decreases and all is well...
Similarly, do not provide for voltage drop during recording. Flash requires voltage higher than the operating voltage of the chip itself.
For example, CR2xxx battery contacts oxidize at currents as low as those of a thermometer. This usually happens within 1 year of operation. And the operating current cannot always break through the oxide film. Sometimes it breaks through, and sometimes the voltage drops sharply... For example, during configuration recording or something else... Then, if there was a reboot, the previous values ​​will be accepted. But if the data was incorrect when reading, then a reset to default will occur.
Solder a capacitor into the power supply - this can eliminate some problems.

pvvx, 2025-06-15:
[image]
[test_B16_new_04_clk36kHz.zip](https://github.com/user-attachments/files/20747698/test_B16_new_04_clk36kHz.zip)

When the frequency is increased and the supply voltage is reduced, the LCD may not work. That is, it works at 3.0V, but when approaching 2.0V - no. It is necessary to check, and also at different temperatures...
Therefore, for some types of LCD controllers, it is necessary to reduce the bus frequencies several times.
Moreover, on some boards, the same Xiaomi, on I2C solders unnecessary protective diodes, which have a huge capacity and do not allow working at standard frequencies for the components used. Apparently, this is another sabotage - they cannot solder a capacitor into the power supply, but bad diodes - easily.

apppie123, 2025-06-16:
Haven't been able to test anything today, sorry. I plan on running a few sensors with the 20kHz and 38kHz firmware on a power supply with 10~50 Ohm resistor in series on 3v to 2v to emulate CR2032 internal resistance. Just need to find some time.

apppie123, 2025-06-17:
Testing with the 20kHz version, I use a Rigol DP832 bench supply and 135 Ohm resistor in series. 8mA limit.

**3.0Volt**
The display always turns on and bluetooth stays connected. Normal expected behaviour.

**At 2.9Volt**
The LYWSD03MMC boots and updates the LCD regularly. But it does not always boot.
When it boots and I connect to it then the display turns off but bluetooth stays connected and responsive.

**2.85Volt**
The LYWSD03MMC never boots.

pvvx, 2025-06-17:
1. Initial power-on causes a longer consumption required for the initial initialization of the entire code and SOC.
2. When starting a BLE connection, it is similarly impossible to normalize the duration and periods of sleep of the chip, since the initial intervals are affected by the connecting adapter. As a result, the BLE connection process causes increased battery consumption. This causes a limitation of the connection capability when the battery is discharged from 40%. At the same time, operation in beacon mode will be maintained until the CR2032 energy is completely depleted.

In general, the CR2032 battery is not designed for normal operation of BLE chips due to premature degradation from current pulses greater than 5 mA.
Only the CR2450 battery is suitable, since the pulse currents available in BLE chips are acceptable for it and do not cause degradation.
TLSR825x at RF TX +0 dBm has a pulse current of 6 mA.

apppie123, 2025-06-17:
Sadly I don't have a microamp meter so I cannot share all details of this new sensor.
I see what you mean and it all makes sense. I'll test some more at 2.9V.

I do hope that you and others have a little better impression of this version.
20kHz appears to be the max(although we didn't test between 20kHz and 36kHz.

Is there already an idea on the still unanswered question: how to implement this in the current firmware?

pudrik, 2025-06-19:
Little side note on these since i managed to brick 3 out of 4 sensors i have, i was wondering what interface did you use to dump the firmware was it Telink official debugger or any generic rx / tx solution with sws ? does it have any kind of read out protection cause i cant get any of the python scripts to read at all and ive tried a ch341 with a resistor and also tl232 but none work, only way to write was having a tx only and that only seem to work from the web page

apppie123, 2025-06-19:
I used the official Telink BDT. I never got those converters to work with RX+TX, only TX worked for me, so I got an official BDT.

pvvx, 2025-06-19:
I've only been using https://github.com/pvvx/TLSRPGM for a few years now.
There's also Telink BDT, but I don't use it.

mnkhbmb, 2025-06-23:
> > [@apppie123](https://github.com/apppie123) can you post the original FW again (to be able to restore at least the display?)
> 
> [#664 (comment)](https://github.com/pvvx/ATC_MiThermometer/issues/664#issuecomment-2973010746)

sir this link is expired

Server4hassio, 2025-06-23:
> > I created a full dump of this firmware. Please find it on below link(I think it expires in 7 days). https://www.transfernow.net/dl/20250614aP8PpIzR
> 
> Firmware CRC matches SDK. Cut file for OTA: [new_b16_ota.zip](https://github.com/user-attachments/files/20739844/new_b16_ota.zip)

Hello. The display got fixed on this version. But this seems to be the original firmware from Mi, right?
So that means there will be a version like 5.3 or something that works with Tasmota, right?

pvvx, 2025-06-23:
> Do you actually need one? The PCB is the same and the connections and pins are the same.

I don't need any Xiaomi LYWSD03MMC at all as they are terrible compared to others.

apppie123, 2025-06-23:
I can't debate that! 👍 
Just trying to help.

vmirage, 2025-06-24:
How about a manual selection by the user for HW B1.6 during firmware flashing with instructions mentioning 2025.3 hardware and possible newer uses the alternate firmware or if experiencing blank screen after flashing, try the alternate firmware.

Server4hassio, 2025-06-24:
> > So that means there will be a version like 5.3 or something that works with Tasmota, right?
> 
> Custom FW: [test_B16_new_04_clk20kHz.zip](https://github.com/user-attachments/files/20747260/test_B16_new_04_clk20kHz.zip)
> 

[image]

The firmware test_B16_new_04_clk20kHz appears as a custom firmware in the Telink flasher, but in Tasmota it doesn't show up as ATCxxxxxx — it appears like in the image instead.

Server4hassio, 2025-06-24:
My Mi device is currently working without a display using Exploit.bin. I just tried flashing the beta v53 directly, but nothing changed — it instantly turned into LYWS03.

To get exploit.bin working:

1.  I had to flash Stock_fw_2.1.1_0159
2.  Flash exploit.bin, and only after that it works (without display). 
  
If I don’t do it in this order, exploit.bin doesn’t work.

So, is there some required firmware that must be present before flashing the beta v53, like in this case?

pvvx, 2025-06-24:
> The firmware test_B16_new_04_clk20kHz appears as a custom firmware in the Telink flasher, but in Tasmota it doesn't show up as ATCxxxxxx — it appears like in the image instead. Not connecting to tasmota

Tasmota doesn't seem to support the open [BTHome](https://bthome.io/) standard...

Wiigian, 2025-06-24:
I bought 10 of these, and tried to flash the first one with ATC_v52.bin, flash seems successful, but screen freezes, and when power cycling the unit, the old firmware is still in use. I tried to flash the stock firmware which made the screen blank, and now it is now showing up in the list of bluetooth devices anymore, guess it is bricked.

Them I came across this thread, and tried the test_B16_new_04_clk20kHz.bin on another LYWSD03MMC. Flash seems successful, but I thought the unit now should have a different firmware than stock? It seems the same.

Connecting to: LYWSD03MMC
19:50:47: MiVer5, DevID: 0x055B-LYWSD03MMC, FnCnt: 15, CtrID: 0x5858 Not bound, Standard certification, MAC: A4C13843E7EA, Data encrypted
19:50:55: NetworkError: Connection Error: Connection attempt failed.
19:50:55: Reconnect 1 from 5
19:51:08: Hardware Revision String: B1.6
19:51:08: Software Revision String: 0159
19:51:08: Firmware Revision String: 2.1.1_0159
19:51:08: Detected Mi device
19:51:08: Connected
19:51:41: Send Login, please wait...
19:51:41: Received device infos are correct
19:51:42: Login successful
19:51:49: File: test_B16_new_04_clk20kHz.bin
19:51:49: File size: 86324 bytes
19:51:49: Add signature file.
19:51:49: Count: 5718
19:51:52: Start DFU
19:53:22: Update done after 88.718 seconds
19:54:34: Searching for devices

19:58:09: Reconnect 1 from 5
19:58:20: Hardware Revision String: B1.6
19:58:20: Software Revision String: 0159
19:58:20: Firmware Revision String: 2.1.1_0159
19:58:20: Detected Mi device
19:58:20: Connected

pvvx, 2025-06-24:
> Update: flashing it from test_B16_new_04_clk20kHz.bin to 1141-020a-01283001-Z03MMC.zigbee also make the screen blank again

There is no Zigbee firmware for the new LYWSD03MMC variant yet.

pvvx, 2025-06-24:
@apppie123 
>  Just trying to help.

UART LCD controller in old HW versions: B1.5/1.6 after receiving new display buffer "AAxxxxxx55" responds with byte 0xAA in RX on GPIO_PB7.
In the SPI version of the LCD, GPIO_PB7 is the SPI data.
If you enable the GPIO_PB7 pull-up to zero, then on the new version with SPI you should get "0" (?)
And on the UART version = "1", since this is the TX UART output of the display controller...

Built-in chip "PullDown" ~100 kOhm. The pull-up time to "0" in the first version 5.3beta is set to 512 μs.
 
* Not have the ability to check how much time is needed to get "0" on GPIO_PB7.
* Starting UART and reading the response "0xAA" for the old version will create an SPI SLK on the new type, and how this will affect it  is unknown. 
* It is unknown what kind of pause is needed or not between block transfers in the SPI version.
* 

Also unknown:
* do older versions of Xiaomi firmware work or not? And which versions?
* what data does the new version report for all UUIDs when connecting? (the flash dump file issued is not complete - not 512 kilobytes)

apppie123, 2025-06-24:
> do older versions of Xiaomi firmware work or not? And which versions?
We tried that in the beginning. No they do not work: https://github.com/pvvx/ATC_MiThermometer/issues/664#issuecomment-2972762207

> the flash dump file issued is not complete - not 512 kilobytes
Which is odd, since I made a 512k dump and used it to recover sensors in the beginnen. But that link has expired. You also confirmed it matched sdk. The smaller version that you made did not work for me to recover sensors.

> Does the SPI version answer something or not? Is the data wire unidirectional or not?
Does not appear to be responding. But I could make some new recordings but we need to compare. Which bytes were not used when you set the right segments? I can focus on those.

pvvx, 2025-06-24:
> If needed I can re-upload but I don't have a server for hosting it.

[image]

The file is attached to the message. Usually in *.zip format.

> We tried that in the beginning. No they do not work: https://github.com/pvvx/ATC_MiThermometer/issues/664#issuecomment-2972762207

These are very old files. I am interested in options with OTA protection using a signature:

If this does not work, then you need a full OTA file with "signed" original version. Otherwise there will be problems with the firmware on the original.
https://github.com/pvvx/ATC_MiThermometer/raw/refs/heads/master/bin/Original/Original_OTA_Xiaomi_LYWSD03MMC_v2.1.1_0159c.bin

Long story short, I need to buy this stupid new version of the thermometer somewhere, gamble by ordering a dozen different ones from different sellers, wait a few months for them to arrive, then throw them in the trash... :)

vmirage, 2025-06-24:
I got mine from here if that helps.

https://a.aliexpress.com/_oEDetqap

pvvx, 2025-06-25:
> I got mine from here if that helps.
> 
> https://a.aliexpress.com/_oEDetqap

Redirects to https://best.aliexpress.com/

pvvx, 2025-06-25:
There are already versions of BLE v5.3b2 and [TelinkMiFlasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) v11.1.
But no one is testing new versions with the SPI display.
There is probably no one needs it.

vmirage, 2025-06-25:
> > I got mine from here if that helps.
> > https://a.aliexpress.com/_oEDetqap
> 
> Redirects to https://best.aliexpress.com/

Try this https://www.aliexpress.com/item/1005005707779050.html

pvvx, 2025-06-26:
General firmware updated to version 5.3b3. Contains new updated version with automatic detection of display controller type (I2C/UART/SPI).

pvvx, 2025-06-26:
Thermometer with SPI display will have HW: B1.1.
This is necessary to select the original firmware.
Due to the fact that the original versions from Xiaomi are different and they do not have auto-recognition of the type of real HW.

[image]

----

Now we need to wait for reviews on the work of the old version B1.6.
Version 5.3b3 is tested on B1.5 with UART LCD, but I do not have B1.6.

pvvx, 2025-06-26:
The sensor type on the new version is also unknown to me. No one reported this :)

Server4hassio, 2025-06-26:
> There are already versions of BLE v5.3b2 and [TelinkMiFlasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) v11.1. But no one is testing new versions with the SPI display. There is probably no one needs it.

Token extractor has stopped working since yesterday. I had switched the previous few to test_B16_new_04_clk20kHz.bin, so now I'm waiting for the token extractor to start working again. 
This is very important for me. 
Also, currently I have to do an OTA update with the stock 2.1.1 firmware first — if it could be made possible to flash directly with v53b2, I’d really appreciate it! 😄

pvvx, 2025-06-26:
The original firmware from Xiaomi with the new version does not provide the ability to determine that this is a new version, incompatible with other types of LYWSD03MMC.
Reading serial number (UUID) is blocked in Chrome Bluetooth API. File a complaint with Chrome :)

[20 more replies]