pvvx/ATC_MiThermometer #670: Support TH05 ?

coliss86, 2025-06-29:
Hello, 

Thanks for your hard work !

I bought a Tuya Smart Wireless Temperature Humidity Sensor from [aliexpress](https://www.aliexpress.com/item/1005006458543396.html) believing it was supported but I may missed something. 

When in pairing mode, it advertised itself as TH05, I was hoping buying a BTH01...

In the OTA flash page, after selecting the sensor, it is not recognized, I think it is expected.

[image]

Here is a screenshot of the NRF app : 

[image]

Here is photo of the PCB

[image]

On the main chip, it is written : 
```
PHY+
PHY62220C
W04IXOPSH
2341EN2794
```

Am I doing something wrong ? Should I buy another sensor ?

Thanks

coliss86, 2025-06-29:
Never mind, I found your github repo : https://github.com/pvvx/THB2 and it says it should not be bought. I should have search carefully before.

closed 2025-06-29 by coliss86

pvvx, 2025-06-29:
If the sensor installed is not СHT8305, then this is a normal version of the thermometer.
It is reflashed via USB-UART(COM) adapter with 3.3V signals.