pvvx/ATC_MiThermometer #674: MJWSD05MMC can't be flash back to original firmware via browser, and mi home can't find it

CaoYK1225, 2025-07-06:
Once I flash custom firmware, if I want to back to original, I have to use usb.
Because it shows "Size firmware is more 131072 bytes!" in browser when I choose the original firmware.

So I flashed my MJWSD05MMC back to original from custom via usb.
After reset, I can't find it in mi home app, but it can be found on laptop running windows via bluetooth. I can even use browser flash cuntom firmware.

I've tried on other phone. On iPhone, it can't be found in mi home app. On Android phone, it can be found on system-bluetooth interface, but I can't see it in mi home app neither.

It do can use custom firmware, everything works fine except connecting mi home.
But I want to use it in mi home, what can I do?

pvvx, 2025-07-07:
> But I want to use it in mi home, what can I do?

The use of a third-party device (fw) in Mi Home is prohibited in the user agreement that opens when connecting devices.

[README](https://github.com/pvvx/ATC_MiThermometer#)
_Custom firmware is not supported in MiHome. But it is possible to work LYWSD03MMC with 'Xiaomi Gateway 3' by first registering in MiHome on the official firmware, flashing to custom firmware with restoring the binding key from MiHome, replacing the device name with "LYWSD03MMC" and setting the advertising type to "MIJIA (MiHome)" marked "encrypted"._

For testing, instead of "LYWSD03MMC" try using "MJWSD05MMC"...

pvvx, 2025-07-07:
> So I flashed my MJWSD05MMC back to original from custom via usb.
> After reset, I can't find it in mi home app, but it can be found on laptop running windows via bluetooth. I can even use browser flash cuntom firmware.

Something was done incorrectly or the firmware was not fully reflashed.

CaoYK1225, 2025-07-07:
> > But I want to use it in mi home, what can I do?但是我想在米家使用它，我该怎么办？
> 
> The use of a third-party device (fw) in Mi Home is prohibited in the user agreement that opens when connecting devices.连接设备时打开的用户协议禁止在米家中使用第三方设备 （fw）。
> 
> [README  自述文件](https://github.com/pvvx/ATC_MiThermometer#) _Custom firmware is not supported in MiHome. But it is possible to work LYWSD03MMC with 'Xiaomi Gateway 3' by first registering in MiHome on the official firmware, flashing to custom firmware with restoring the binding key from MiHome, replacing the device name with "LYWSD03MMC" and setting the advertising type to "MIJIA (MiHome)" marked "encrypted".MiHome 不支持自定义固件。但是可以通过先在米家官方固件上注册米家，刷入自定义固件并从米家恢复绑定密钥，将设备名称替换为“LYWSD03MMC”并将广告类型设置为“米家（米家）”标记为“加密”来 LYWSD03MMC“小米网关 3”。_
> 
> For testing, instead of "LYWSD03MMC" try using "MJWSD05MMC"...对于测试，不要使用 “LYWSD03MMC”，而是尝试使用 “MJWSD05MMC”...

Thanks a lot, it works.

CaoYK1225, 2025-07-07:
> > Because it shows "Size firmware is more 131072 bytes!" in browser when I choose the original firmware.因为当我选择原始固件时，它在浏览器中显示“Size firmware is more 131072 bytes！”。
> 
> Fixed. TelinkMiFlasher.html version updated.固定。TelinkMiFlasher.html 版本已更新。

Thanks, it works.