pvvx/ATC_MiThermometer #675: LYWSD03MMC B1.6 becomes B1.1 after 5.3 upgrade

macmpi, 2025-07-07:
Device was reported B1.6 up until 5.2, but now:
```
6:48:27 PM: Hardware Revision String: B1.1
6:48:27 PM: Software Revision String: V5.3
6:48:27 PM: Firmware Revision String: github.com/pvvx
6:48:27 PM: Detected custom Firmware
6:48:29 PM: Hardware Version: LYWSD03MMC B1.1, Software Version: 5.3
```

macmpi, 2025-07-07:
back to normal after reboot (pull-out battery)

closed 2025-07-07 by macmpi

apppie123, 2025-07-08:
Yes, there seems to be an intermittent issue still. Maybe the 20kHz clock speed is at the top range for some chips and those chips do not always recognise a refresh.

I noticed from the beta's I tried to 5.3 that at boot all characters are displayed(great!) but it takes a varying length of time before this is changed to the right values. This can be seconds but also tens of seconds.

also after sending new settings the sensor appears to most of the times reboot itself. I didn't see this behaviour in V4.9 for example. Also in V4.9 the screen would be blank at boot.

pvvx, 2025-07-09:
> also after sending new settings the sensor appears to **most** of the times reboot itself.

most ?

When rebooting, the BLE connection is necessarily broken.

Automatic reboot at the end of the BLE connection is performed only in the following cases:
1. Turning on, turning off or changing the "pin-code".
2. Switching between the BT5+ or BT4.2 operating formats.

macmpi, 2025-07-09:
ok, thanks for the detailed information: it's really helpful.