pvvx/ATC_MiThermometer #681: lywsd03mmc no change in tempreture or humidity.

kash1982, 2025-07-15:
I have several sensors that all have zigbee firmware from devbis z03mmc which im now trying to go back to  Bluetooth firmware from hear.

I've tried various stock firmware v5.2 and v5.3 and can't seem to get it to work no display but can see a faint display at a angle with all segments on.

Hardware is b1.5 on this one. 

I flashed 
[ATC_ota_400000](https://github.com/devbis/z03mmc/blob/master/assets/ATC_ota_40000.bin)

This got display working so I used the ota page to flash v5.3 and now gets stuck like this.

[image]

After a while leaving it alone it will go back to normal. The ota page will see it when scanning and connect to it. But the intital turn on tempreture and humidity will display without any change at all.

Not sure where to go from hear.

pvvx, 2025-07-16:
Connection log from TelinkMiFlasher.html is required.

Have you tried removing and reinstalling the battery?

pvvx, 2025-07-16:
The problem may be related to adding code to detect the [new chipset version](https://github.com/pvvx/ATC_MiThermometer/issues/664).
I can't solve it in the next few days - I don't have version B1.5.
Use the old version for now: [ATC_v49.zip](https://github.com/user-attachments/files/21265825/ATC_v49.zip)

kash1982, 2025-12-02:
> [@kash1982](https://github.com/kash1982) did you ever find the root cause?

I purchased and used a telink flasher from aliexpress. Flashed all of mine with no issues. Can move any firmware now by using the flasher.