pvvx/ATC_MiThermometer #692: MJWSD06MMC can't get custom firmware on it

rosenrot00, 2025-08-10:
I just received my MJWSD06MMC and wanted to flash it using the USBCOMFlashTx tool.

I connected GND to Vbat-, RTS to Vbat+ and TX to SWS, I see that the device restarts as RTS gets pulled low and the writing process starts, however, it doesn't right any firmware to it as the screen still shows temperature values and the bluetooth signal symbol flashes.

I tried also using TelinkMiFlasher after registering my MJWSD06MMC in Xiaomi Home but when I enter the device id, the token and the ble key, I get "not registered!" from the TelinkMiFlasher.

Could someone tell me what I'm missing?

rosenrot00, 2025-08-10:
It worked out. Thank you.

duffysk, 2026-02-21:
@rosenrot00 I'm hitting exactly the same problem "not registered" with MJWSD06MMC:
9:44:07 AM: Hardware Revision String: V1.3
9:44:07 AM: Software Revision String: 0009
9:44:07 AM: Firmware Revision String: 0009

Registered successfully and obtained the information with token_extractor:
   NAME:     Kúpeľňa 2
   ID:       blt.3.1o5e6a5n84c03
   BLE KEY:  55c9d8f63c24644756c6fe4c75868e91
   MAC:      A4:C1:38:E0:AF:E0
   IP:       20.56.156.45
   TOKEN:    7a8c9e8126b7700cdcf3a0b3
   MODEL:    xiaomi.sensor_ht.mini

How did you managed to flash it when you were hitting Not registered?