pvvx/ATC_MiThermometer #703: Is there any SDK or advice for data transmission directly via BLE(Without Home Assistant Integration)

LebranceBW, 2025-08-31:
I have an MJWSD06MMC device running the custom MJ6_v53.bin firmware (pvvx ATC firmware) and would like to repurpose it to monitor my PC's CPU and GPU temperatures. The "Show on device screen" feature available in [TelinkMiFlasher ](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html)appears suitable for displaying this custom data. However, I want to establish direct BLE communication without involving Home Assistant.

Are there any available libraries (preferably for C or Python) , SDKs, or **advices** that would enable me to develop a host application for sending custom temperature data directly to the device via BLE? Any suggestions would be greatly appreciated. 🌹

LebranceBW, 2025-09-04:
Though I've got no reply, but I still made it.

[ledmonitor](https://github.com/LebranceBW/ledmonitor)

pvvx, 2025-09-04:
**0x22 - Universal function of displaying data on the display.**

* Works with all BLE thermometers with alternative firmware that have a screen.

Supported in [ESPHome](https://esphome.io/components/display/pvvx_mithermometer/) and other applications. 
```
struct {
	uint8_t	id_command; // = 0x22	
	int16_t	big_number; // -995..19995, x0.1
	int16_t	small_number; // -9..99, x1
	uint16_t 	vtime_sec; // validity time, in sec
	struct {
		/* Smiley: 0 = "     " off,
		 * 1 = " ^_^ "
		 * 2 = " -^- "
		 * 3 = " ooo "
		 * 4 = "(   )"
		 * 5 = "(^_^)" happy
		 * 6 = "(-^-)" sad
		 * 7 = "(ooo)" */
		uint8_t smiley			: 3;
		uint8_t percent_on		: 1;
		uint8_t battery		: 1;
		/* 0 = "  ", shr 0x00
		 * 1 = "°Г", shr 0x20
		 * 2 = " -", shr 0x40
		 * 3 = "°F", shr 0x60
		 * 4 = " _", shr 0x80
		 * 5 = "°C", shr 0xa0
		 * 6 = " =", shr 0xc0
		 * 7 = "°E", shr 0xe0 */
		uint8_t temp_symbol	: 3;
	} flg;
} external_data_t;
```

The MJWSD05MMC display has more characters and the data for the display function is different:
```
struct {
	uint8 t	id_command; // = 0x22	
	int32_t	number; // -999.50..19995.50, x0.01
	unt16_t 	vtime_sec; // validity time, in sec
	struct  {
		/* 0 = "     " off,
		 * 1 = " ^-^ "
		 * 2 = " -^- "
		 * 3 = " ooo "
		 * 4 = "(   )"
		 * 5 = "(^-^)" happy
		 * 6 = "(-^-)" sad
		 * 7 = "(ooo)" */
		uint8_t smiley			: 3;
		uint8_t battery			: 1;
		/* 0x00 = "  "
		 * 0x01 = "°г"
		 * 0x02 = " -"
		 * 0x03 = "°c"
		 * 0x04 = " |"
		 * 0x05 = "°Г"
		 * 0x06 = " г"
		 * 0x07 = "°F"
		 * 0x08 = "%" */
		uint8_t temp_symbol		: 4;
	} flg;
```